Koenkk/zigbee2mqtt #1158: Can't pair anymore

gempir, 2019-02-26:
I yesterday all my zigbee devices stopped working correctly on my network and I figured I should do an update of zigbee2mqtt anyway.

After the update i got an JS error when starting the zigbee2mqtt service. I then wanted to remove all devices so I could repair them. I removed from from the configuration, commented them out in my home assistant config, ran some mqtt remove commands. I also then cloned the entire repo again after removing it from /opt

I reflashed my stick with the newest version (I use the recommended Stick from the docs) and then I followed the usual steps like I did a few months ago.

Now None of my devices will pair. I have different devices with the same Problems.
Example:
- Tradfri Bulb which I reset by the manual, but the light never got brighter. After a few secs the log of zigbee2mqtt shows "AM Cannot get the Node Descriptor of the Device: ..." I tried repairing like 5x, restarting the service, restarting the raspberry pi, switching the USB Slot of the device...

- An Aqara Motion Sensor, also same error "AM Cannot get the Node Descriptor of the Device: ..."

I tried rolling back to 1.0.1 of zigbee2mqtt AND reflashed the Stick with a version before group support (from December) but that didn't help either. I can't seem to pair my devices anymore.

Koenkk, 2019-02-26:
Before repairing everything make sure that your old zigbee network is down. Note that the zigbee network is also kept up by e.g. TRADFRI bulbs which are still powered.

gempir, 2019-02-26:
I turned of all bulbs, I have no other powered Zigbee devices, so network should be down. No success though pairing either 1 bulb or a motion sensor.

Koenkk, 2019-02-26:
Are you sure `permit_join: true` is set in `configuration.yaml`?

gempir, 2019-02-26:
Yes. I'm 100% sure.

Koenkk, 2019-02-26:
Can you empty your `data/database.db`?

gempir, 2019-02-26:
@Koenkk Already did that btw. But I updated my previous comment.

closed 2019-02-26 by gempir

Koenkk, 2019-02-26:
Great that you solved the problem :smile:

gempir, 2019-02-26:
@Koenkk could you explain what "Cannot get the Node Descriptor of the Device: ..." means exactly? I'm still having issues pairing more than 1 device and the log shows the error for those devices when trying to pair

Koenkk, 2019-02-26:
Mostly this succeeds after failing a few times (also other existing zigbee networks seems to have a role in this), for the xiaomi sensors you typically have to press the pairing button.