Koenkk/zigbee2mqtt #1159: Report battery state of heiman smoke detector as it's own entity in ha

h4nc, 2019-02-26:
All of my other zigbee2mqtt devices that report a battery state have their own entity (auto discovery).

I got a new heiman smoke detector.
The payload looks like this:

payload '{"smoke":false,"battery_low":false,"linkquality":15}'

So it also does report the battery state (only true or false, instead of a value). 

I think it would only be consistent if that value also gets it's own entity, right? Because it only reports true or false, this should be a binary_sensor too.

Koenkk, 2019-02-26:
Yes, please add to https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L79

```js
    'binary_sensor_battery_low': {
        type: 'binary_sensor',
        object_id: 'battery_low',
        discovery_payload: {
            payload_on: true,
            payload_off: false,
            value_template: '{{ value_json.battery_low}}',
            device_class: 'battery',
        },
    },
```

And then add it to the devices here: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L297.

If this works, please make a PR.

h4nc, 2019-02-26:
@Koenkk Sry I cannot test this :(

I know that should use this:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html

But I already tried this once and I failed at the first point, because I wasn'T able to put that  authorized_keys file in my root directory on my sd card.

However there is no reason why the code above shouldn't work, right?

I would suggest 

  'HS1SA': [configurations.binary_sensor_smoke, configurations.binary_sensor_battery],

also we should add that for the HS3SA (I think it's the same)

Koenkk, 2019-02-26:
No problem, added them for both, thanks!

closed 2019-02-26 by Koenkk

h4nc, 2019-02-26:
So now there will be two entities for that battery state?

This will be in edge in about an hour right?

So than I can test it.

Koenkk, 2019-02-26:
No, 1 for HS1SA, one for HS3SA. 

Will be available in edge in an hour.

h4nc, 2019-02-26:
Of course, I see.

h4nc, 2019-02-26:
> Will be available in edge in an hour.

How does this work? So if you make a change it will be in edge an hour later.
But what if you make a new change in that hour? Does this cancel the first update to edge? Or will there be the new version and some time later the new new version?

Koenkk, 2019-02-27:
Once the build of zigbee2mqtt dev succeeds, a rebuild of hassio edge is triggered (takes +- one hour to complete).

Yes, linkquality is also there for every device.

h4nc, 2019-02-27:
Ok thanks.

Already tried the new battery_low entities. Works, thanks!