Koenkk/zigbee2mqtt #1160: TRADFRI control outlet very difficult to pair

bremy, 2019-02-26:
Hello

I am trying to pair my Tradfri control outlet (French one) but no success. I just go it working once yesterday evening and since no more possible to pair it on a fresh system (I have a template disk image to reload quickly)
I am running on RP2+ with a CC2531 sniffer. Sniffer flashed with the 2 days old DEV firmaware, zigbee2mqtt installed and upadate, and using the most recent https://github.com/Koenkk/zigbee-shepherd-converters.git
I was setting "debug" logs to get the maximum of details

When pairing I can see the device detected and one message "online" received.
```
PM Configuring 0x000d6ffffe9c49ef (0x000d6ffffe9c49ef) ...
...
PM Failed to configure 0x000d6ffffe9c49ef (0x000d6ffffe9c49ef) ('Error: Timed out after 10000 ms'
....
zigbee2mqtt:debug 2/26/2019, 6:19:31 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe9c49ef"'
zigbee2mqtt:debug 2/26/2019, 6:19:31 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe9c49ef"'
zigbee2mqtt:debug 2/26/2019, 6:19:31 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe9c49ef"'
zigbee2mqtt:debug 2/26/2019, 6:19:31 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe9c49ef"'
zigbee2mqtt:debug 2/26/2019, 6:19:31 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe9c49ef"'
zigbee2mqtt:debug 2/26/2019, 6:19:31 PM Received zigbee message of type 'devIncoming' with data '"0x000d6ffffe9c49ef"' of device 'TRADFRI control outlet' (0x000d6ffffe9c49ef)
zigbee2mqtt:info 2/26/2019, 6:19:31 PM Device incoming...
zigbee2mqtt:info 2/26/2019, 6:19:31 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
zigbee2mqtt:debug 2/26/2019, 6:19:31 PM Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI control outlet' (0x000d6ffffe9c49ef)
```
But not topic zigbee2mqtt/0x000d6ffffe9c49ef created!!!! and no other message received from that outlet :-(

Yesterday when it was working, I saw on message with "state" and "linkQuality", but impossible to get this again after hundred's of pair trials

I was also trying to reset the dongle but

- stopping zigbee2mqtt
- unplug Dongle
- plug dongle
- press reset button on the dongle
- restart zigbee2mqtt
- 
again not better
 
any idea?

Thanks

Koenkk, 2019-02-26:
It look like pairing is successful, created message is probably not showing because `0x000d6ffffe9c49ef` is already in `data/configuration.yaml`.

bremy, 2019-02-26:
I can't see the outlet as an object in Jeedom. Probably because the topic zigbee2mqtt/0x000d6ffffe9c49ef is not existing?

Did you notice for that message when I restart zigbee2mqtt when this outlet is supposed to be paired?

``
PM Failed to configure 0x000d6ffffe9c49ef (0x000d6ffffe9c49ef) ('Error: Timed out after 10000 ms
``

bremy, 2019-02-26:
Trying to execute a 'set' command on the outlet, I am getting this error message

`zigbee/0x000d6ffffe9c49ef/set on`

`PM Zigbee publish to device '0x000d6ffffe9c49ef', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
`

bremy, 2019-02-26:
Good news. I started again from scratch and now I can see my remote outlet working in Jeedom
To see it as an new published object in Jeedom, it was necessary to generate that command
`zigbee2mqtt/0x000d6ffffe9c49ef/set off`

and now bingo this is working nice with on and off buttons in Jeedom

Thank you for that wonderfull zigbee2mqtt work

bremy, 2019-02-26:
Next step is now to pair my simple ON/OFF Tradfri remote. But first I am going to save this working config :-)

closed 2019-02-26 by bremy