Koenkk/zigbee2mqtt #1161: GE Link Smart LED Light Bulb, A19

rb2k, 2019-02-26:
I just added the A19 version of the GE Link bulb: https://www.amazon.com/GE-60-Watt-Equivalent-1-Pack-Zigbee/dp/B00NOL16K0

It gets recognized as the BR30 one:

```
  zigbee2mqtt:info 2019-2-26 15:07:12 Device incoming...
  zigbee2mqtt:info 2019-2-26 15:07:12 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2019-2-26 15:07:12 New device with address 0x7ce5240000011c7f connected!
  zigbee2mqtt:info 2019-2-26 15:07:13 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x7ce5240000011c7f"}'
  zigbee2mqtt:info 2019-2-26 15:07:13 MQTT publish: topic 'homeassistant/light/0x7ce5240000011c7f/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/0x7ce5240000011c7f/set","state_topic":"zigbee2mqtt/0x7ce5240000011c7f","name":"0x7ce5240000011c7f_light","unique_id":"0x7ce5240000011c7f_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x7ce5240000011c7f","name":"0x7ce5240000011c7f","sw_version":"Zigbee2mqtt 1.1.1","model":"Link smart LED light bulb, BR30 soft white (2700K) (22670)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

They seem to show up in homeassistant and can be somewhat controlled.

Changing brightness and on/off seems to be a bit flakey:

```
  zigbee2mqtt:error 2019-2-26 15:18:46 Zigbee publish to device '0x7ce5240000011c7f', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 2019-2-26 15:18:47 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":138,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-2-26 15:18:47 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":138,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2019-2-26 15:18:47 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":138,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:error 2019-2-26 15:18:47 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":138,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 2019-2-26 15:18:48 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":1,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2019-2-26 15:18:48 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":1,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 2019-2-26 15:18:49 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-2-26 15:18:49 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":255}'
```

```
{"level":82,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-2-26 15:20:18 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":82,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-2-26 15:20:18 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":255}'
  zigbee2mqtt:error 2019-2-26 15:20:18 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":82,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 2019-2-26 15:20:19 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2019-2-26 15:20:19 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 2019-2-26 15:20:21 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2019-2-26 15:20:22 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
  zigbee2mqtt:info 2019-2-26 15:20:24 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2019-2-26 15:20:25 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
```

Not sure if this is really a side effect of bad coverage or something else.

Koenkk, 2019-02-27:
Can you check with the latest dev firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin (doesn't require re-pairing of devices).

rb2k, 2019-03-02:
Ok, flashed with the latest firmware. I did some experiments close to the zigbee stick: worked fine. I can now turn it off / on and dim it without any issues:

```
  zigbee2mqtt:info 2019-3-1 17:58:45 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":16}'
  zigbee2mqtt:info 2019-3-1 17:58:45 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":255}'
  zigbee2mqtt:info 2019-3-1 17:59:14 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":97,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-3-1 17:59:14 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":255}'
  zigbee2mqtt:info 2019-3-1 17:59:14 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":97}'
  zigbee2mqtt:info 2019-3-1 17:59:15 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":53,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-3-1 17:59:15 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":97}'
  zigbee2mqtt:info 2019-3-1 17:59:15 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":53}'
  zigbee2mqtt:info 2019-3-1 17:59:16 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":9,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-3-1 17:59:16 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":53}'
  zigbee2mqtt:info 2019-3-1 17:59:16 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":9}'
  zigbee2mqtt:info 2019-3-1 17:59:17 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":1,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-3-1 17:59:17 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":9}'
  zigbee2mqtt:info 2019-3-1 17:59:17 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":1}'
  zigbee2mqtt:info 2019-3-1 17:59:19 Zigbee publish to device '0x7ce5240000011c7f', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-3-1 17:59:19 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":1}'
  zigbee2mqtt:info 2019-3-1 17:59:19 MQTT publish: topic 'zigbee2mqtt/0x7ce5240000011c7f', payload '{"state":"ON","brightness":255}'
```

Well, I guess the A19 bulb works, just sharing its name with the BR30 at the moment.

rb2k, 2019-03-02:
Moving the bulbs a few rooms over causes issues again. I think the original problem might just have been a coverage issue.

Koenkk, 2019-03-02:
I guess so, can this be closed?

rb2k, 2019-03-03:
Should this be added to https://www.zigbee2mqtt.io/information/supported_devices.html ?
It's a different bulb (A19 vs BR30), but by now they share the same name.

Koenkk, 2019-03-04:
Yes, for that, you need to make a PR to update the description here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L1543

rb2k, 2019-03-06:
I opened one over here: https://github.com/Koenkk/zigbee-shepherd-converters/pull/320

Time to close this one :)

closed 2019-03-06 by rb2k