Koenkk/zigbee2mqtt #1163: have issue with CC2530+ttl2USB+zigbee2mqtt

msadeghz, 2019-02-27:
Hi friends,
I want to create a coordinator with CC2530 and ttl USB adapter. I did all of the steps but I got error below:
[image]

It's said that P2.0 of CC2530 is used for uart interface. P0.2 and P0.3 are rx/tx pins of uart. what's the functionality of P2.0.  I have disabled the rtscts in configuration.yaml. I use default zigbee2mqtt firmware.

msadeghz, 2019-02-27:
I've checked the document 
[https://www.silabs.com/content/usergenerated/asi/cloud/attachments/siliconlabs/en/community/wireless/zigbee-and-thread/forum/jcr:content/content/primary/qna/etrx357-lr_cvs_texas-TmdR/Z-Stack%20ZNP%20Interface%20Specification.pdf](url)
it has suggested to connect cts and rts pin of CC2530 to ttl2USB equivalent pins.

CC2530                      FT232 module
--------                       ---------------
RX                                       TX
TX                                       RX
CTS                                    RTS
RTS                                    CTS

I did that and flash the CC2530 chip with default firmware (Home 1.2.2a.44539) and make "rtscts" true in configuration file. the issue is solved. but after afew seconds the ZNP doesn't send incoming message to uart. but it works fine in Zigbee side. but it just doesn't send the data to zigbee2mqtt.

JLFN, 2019-02-28:
hey @msadeghz  iam looking for that firmware can you please share it?
https://github.com/Koenkk/Z-Stack-firmware/issues/45

msadeghz, 2019-02-28:
> hey @msadeghz iam looking for that firmware can you please share it? I would like to flash it with Smartrf.
> [Koenkk/Z-Stack-firmware#45](https://github.com/Koenkk/Z-Stack-firmware/issues/45)
> @Koenkk is it possible to get a alternative firmware like this ?

I use IAR workbench, [default z-stack source code (Z-STACK-HOME)](http://www.ti.com/tool/z-stack-archive) and cc-debugger. to flash with SmartRF you need hex file. how can I help you?

JLFN, 2019-02-28:
> I use IAR workbench, [default z-stack source code (Z-STACK-HOME)](http://www.ti.com/tool/z-stack-archive) and cc-debugger. to flash with SmartRF you need hex file. how can I help you?

Hey yes i have cc-debugger and SmartRF also. yes please the .hex can you make a pull request to Koenkk/Z-Stack-firmware also ?

msadeghz, 2019-02-28:
please check this hex file is helpful. the hex file format is "intel extended".

[CC2530.zip](https://github.com/Koenkk/zigbee2mqtt/files/2914427/CC2530.zip)

JLFN, 2019-02-28:
> I've checked the document
> [https://www.silabs.com/content/usergenerated/asi/cloud/attachments/siliconlabs/en/community/wireless/zigbee-and-thread/forum/jcr:content/content/primary/qna/etrx357-lr_cvs_texas-TmdR/Z-Stack%20ZNP%20Interface%20Specification.pdf](url)
> it has suggested to connect cts and rts pin of CC2530 to ttl2USB equivalent pins.
> 
> CC2530 -- FT232 module
> RX <--------> TX
> TX <--------> RX
> (P0.5) CTS <--------> RTS
> (P0.4) RTS <--------> CTS
> 
> I did that and flash the CC2530 chip with default firmware (Home 1.2.2a.44539) and make "rtscts" true in configuration file. the issue is solved. but after afew seconds the ZNP doesn't send incoming message to uart. but it works fine in Zigbee side. but it just doesn't send the data to zigbee2mqtt.

@msadeghz maybe its the quality of your chip iam going to try with an FTDI UART TTL to USB that should be better quality.

msadeghz, 2019-03-04:
@JLFN  yes, you are correct.
I remove the ttl2USB module and connect CC2530 directly to processor through UART. the issue is solved.
the ft232 driver in raspbian is not stable.

closed 2019-03-04 by msadeghz