Koenkk/zigbee2mqtt #1164: Sniffing Xiaomi - cannot decrypt packets

timdonovanuk, 2019-02-27:
Hi,

Has anyone had success sniffing this traffic? I can sniff the packets, but they do not decrypt using the standard Trust Center link key of 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39 (ZigBeeAlliance09). I have tried with two separate sniffing applications now and two different sniffing firmwares on the CC2531.

Stuck right now.

Cheers.

Koenkk, 2019-02-27:
You need to sniff the (second) encryption key first (as explained in step 3 of http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

timdonovanuk, 2019-02-27:
The **second** encryption key **first**?

The instructions say "Add the Trust Center link key" first - are you saying I need to ignore this and add the Transport Key first?

I have added the Trust Center link key, but it doesn't work, hence the ticket. I cannot decrpyt the second key (Transport Key) if the first Trust Center link key doesn't work, because the Transport Key will be encrypted still....

timdonovanuk, 2019-02-27:
FINALLY. The instructions you made assume that the Zigbee devices you want to sniff are on channel 11. This is not always the case, for example Xiaomi devices are often on channel 20. 

If using Zboss Sniffer change the right hand dropdown to your desired channel:

[image]

dzungpv, 2019-02-27:
Use trial version of ubiqua sniffer and then it will automatically detect the key, after that copy to wireshark. You can try all channel from 11 to 26 one by one until it get the packet

Koenkk, 2019-02-27:
@timdonovanuk so where did the packets come from which you were sniffing on channel 11?

closed 2019-02-27 by Koenkk

timdonovanuk, 2019-02-27:
Its possible to have more than one zigbee gateway..? Hue/Tradfri/SmartThings/Hive etc. all have zigbee hubs.