Koenkk/zigbee2mqtt #1165: upgrade from 1.0.1 to 1.1.1

yaniv1111, 2019-02-27:
Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

yaniv1111, 2019-02-27:
Hi,

I running zigbee2mqtt 1.0.1 
what need to be done to upgrade it to 1.1.1?

Thanks

Koenkk, 2019-03-03:
See step 6 here http://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

closed 2019-03-03 by Koenkk