Koenkk/zigbee2mqtt #1167: Lights dropping connection

GForce2010, 2019-02-28:
My issue isn't with Zigbee2MQTT itself but I am hoping somebody can give me some advise.

I had all my lights connected via a Philips Hue Bridge for some time but I noticed that lights had started dropping their connection and had to be power cycled to get them to reconnect.

Thinking I had a problem with my bridge I Installed Zigbee2MQTT and transferred all my lights over to that. Unfortunately I can still see that lights are dropping connection, on the positive side, so far, they have all reconnected by themselves without needing a power cycle.

All the bulbs are being used as routers so I can'r see single strength being an issue, can having multiple Zigbee bulbs close to each other cause problems? I have 4 Spot Bulbs in the living room that are within 30-40 cm apart, 2 of them are fine but the other 2 tend to drop off from time to time.

Back when I was using the Hue Bridge I tried different channels and that made no difference.

GForce2010, 2019-03-06:
After mush troubleshooting it turned out to be interference from my WiFi. 

In case anybody else has similar problems here is what I did.

Set WiFi to channel 11 and Zigbee to channel 11 (Although the channel number is the same this is about as far as you can separate them by frequency.)
Moved everything that can support it over to 5G WiFi.
Re-Paired some items so they connected to closer routers, the network map that Zigbe2MQTT can provide made easy.

I now only have 1 problem bulb that seems to drop it connection from time to time. I plan to install a standalone router nearby in an attempt to resolve that.

closed 2019-03-06 by GForce2010

Koenkk, 2019-03-06:
@GForce2010 thanks for reporting your findings.

ciB89, 2019-05-25:
Do I need to repair devices when I move them around? I used to bring everything close to the coordinator when pairing, and then moved my devices to its intented location - was that a mistake? One of my bubls is losing its connection all the time and I was wondering why...

Koenkk, 2019-05-25:
After some time the device should find a new parent, but this completly depends on the device