Koenkk/zigbee2mqtt #1168: Problems starting zigbee2mqtt

doppiej, 2019-02-28:
Hello,

I am having problems with starting zigbee2mqtt.
After running npm start I am receiving this. What am I doing wrong?
`pi@RP-Domoticz:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.1.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2019-2-28 13:23:47 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-02-28.13-23-46'
  zigbee2mqtt:debug 2019-2-28 13:23:47 Removing old log directory '/opt/zigbee2mqtt/data/log/2019-02-28.13-22-43'
  zigbee2mqtt:debug 2019-2-28 13:23:48 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
/opt/zigbee2mqtt/lib/extension/devicePublish.js:13
    zigbeeShepherdConverters.toZigbeeConverters.on_off,
                                                ^

TypeError: Cannot read property 'on_off' of undefined
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/extension/devicePublish.js:13:49)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/controller.js:13:32)
    at Module._compile (module.js:653:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-02-28T12_23_48_578Z-debug.log
`

This is my config file:
`# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyUSB_zb2mq

# Optional: advanced settings
advanced:
  # Optional: Logging level, options: debug, info, warn, error
  log_level: debug`

Koenkk, 2019-02-28:
Did you make any changes to `zigbee-shepherd-converters`?

doppiej, 2019-02-28:
No, I am still trying to set zigbee2mqtt up! However I can't get it to work unfortunately.

Koenkk, 2019-02-28:
What instructions did you follow?

doppiej, 2019-03-01:
Yours, on https://www.zigbee2mqtt.io/

Koenkk, 2019-03-01:
Can you do:

```bash
rm -rf node_modules
npm install
```

doppiej, 2019-03-06:
I tried. Same issue.
This is the npm log, maybe something you could use: [https://pastebin.com/Cc89r5Yh](https://pastebin.com/Cc89r5Yh)

Koenkk, 2019-03-06:
What is the output of 

`cat node_modules/zigbee-shepherd-converters/index.js` 

and 

`cat node_modules/zigbee-shepherd-converters/converters/toZigbee.js`

doppiej, 2019-03-06:
Both are shown here: [https://pastebin.com/aAUqHDpQ](https://pastebin.com/aAUqHDpQ)

Koenkk, 2019-03-06:
What is the output of `npm ls zigbee-shepherd-converters` (your `index.js` is from an old version).

doppiej, 2019-03-06:
Shown here: [https://pastebin.com/bQ6du4jh](https://pastebin.com/bQ6du4jh)
Do I need to make an update?

Koenkk, 2019-03-06:
What is the output of:

```bash
rm -rf node_modules
npm install
```

doppiej, 2019-03-06:
Shown here: [https://pastebin.com/zySU5t73](https://pastebin.com/zySU5t73)

Koenkk, 2019-03-06:
What is the output of `git status` and `git diff`?

doppiej, 2019-03-06:
Shown here: [https://pastebin.com/k0SCvbMK](https://pastebin.com/k0SCvbMK)

doppiej, 2019-03-06:
I guess I copied not all from `git diff`: [https://pastebin.com/QSamiVbz](https://pastebin.com/QSamiVbz)

Koenkk, 2019-03-06:
Can you update npm to the latest version and try again?

```bash
sudo npm install -g npm
rm -rf node_modules
npm install
npm start
```

doppiej, 2019-03-06:
This is the response: [https://pastebin.com/YUjqZn2T](https://pastebin.com/YUjqZn2T)
However the 3rd command wasn't in the original reply I guess, so I missed it. Should I still perform it now?

Koenkk, 2019-03-06:
Looks good! Seem that your original problem has been solved now. 

Some other process is locking your sniffer, verify that the path to the sniffer in `configuration.yaml` is correct and try rebooting the system.

RobertTheLink, 2019-03-06:
my responses of 
`rm -rf node_modules`
`npm install`
`npm start`
reboot of RasPi
`npm start`. ( did not change anything)
 are here: https://pastebin.com/RP1kTwKz

unfortunately this did not yet help

Koenkk, 2019-03-07:
`Error: Connection refused: Not authorized`

Did you set-up a MQTT server with a password? You need to add that to `configuration.yaml` (http://www.zigbee2mqtt.io/configuration/configuration.html)

doppiej, 2019-03-07:
> Looks good! Seem that your original problem has been solved now.
> 
> Some other process is locking your sniffer, verify that the path to the sniffer in `configuration.yaml` is correct and try rebooting the system.

I tried the reboot and it looks promising.
However, when performing the showing of the status I am getting this:
pi@RP-Domoticz:~$ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled)
   Active: active (running) since Wed 2019-03-06 20:00:58 CET; 15h ago
 Main PID: 606 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─606 npm
           ├─954 sh -c node index.js
           └─956 node index.js

Instead of what you show in the documentation:
pi@raspberry:/opt/zigbee2mqtt $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-06-07 20:27:22 BST; 3s ago
 Main PID: 665 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─665 npm
           ├─678 sh -c node index.js
           └─679 node index.js

Jun 07 20:27:22 raspberry systemd[1]: Started zigbee2mqtt.
Jun 07 20:27:23 raspberry npm[665]: > zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
Jun 07 20:27:23 raspberry npm[665]: > node index.js
Jun 07 20:27:24 raspberry npm[665]: 2018-6-7 20:27:24 INFO Starting zigbee-shepherd
Jun 07 20:27:25 raspberry npm[665]: 2018-6-7 20:27:25 INFO zigbee-shepherd started

Koenkk, 2019-03-07:
@doppiej did zigbee2mqtt start? You can find the logging in `data/log`.

doppiej, 2019-03-07:
Hmm, I guess not. It looks like it only shows logs of when I performed the `npm start` commands...
However, I also performed all the steps mentioned in "Running as a daemon with systemctl" and when I show the status it says that it is active and running.
So what am I doing wrong?

RobertTheLink, 2019-03-07:
Thx Koenkk,
it's working for me now. Mosquitto was starting manually, but failing to start as service, and therefore was off after reboot.

closed 2019-05-14 by stale[bot]