Koenkk/zigbee2mqtt #1170: Add support for Home Assistant API

TheZoker, 2019-02-28:
I'm not sure if this makes sense or is even possible, but:
What I really like about [ESPHome ](https://esphome.io/components/api) is, that apart from having the ability to use MQTT you can also integrate ESPHome devices via the HA api. And that makes things easier, faster and you don't need a mqtt broker. I'm not suggesting to drop mqtt in any way, but maybe consider adding support for the native API :)

Koenkk, 2019-03-01:
I'm interested on why integrating via the HA api would be faster than integrating via MQTT. Could you explain more about this?

nldroid, 2019-03-01:
Here are the reasons why ESPHome is supporting the Home Assistant Api now:

https://esphome.io/changelog/v1.10.0

Koenkk, 2019-03-01:
@nldroid thanks for linking this! However, I dont see any big benefits from adding this yet, not having to install a MQTT broker is one, but this isn't something that is difficult to do. Memory isn't a limitation on systems that zigbee2mqtt is running on (especially not at the byte level).

Something that should also be taken into account is the extra maintance of such a feature.

nldroid, 2019-03-01:
Yeah, i know that it will be a pain to maintain two interfaces. Mqtt is not really a problem for me.

The way esphome works is really nice. I moved from espeasy to esphome because the mqtt messages from espeasy do not support auto discovery but that isn't a problem with zigbee2mqtt.

Besides that, you'll still need something to send messages to zigbee2mqtt for managing the system (to put on pairing for example)

TheZoker, 2019-03-01:
Agreed, was just wondering if it worth the maintenance

closed 2019-03-02 by Koenkk

HA-TB303, 2020-04-10:
I have been using esphome from the start, at first with mqtt, and based on my own experiance the home assistant api integration has been Faster and more robust.

I vote in favor of the api support :)