Koenkk/zigbee2mqtt #1172: Device Default State On/Off

soloam, 2019-03-01:
Hello, I have some xiaomi plugs that I what always on as default, If I unplug them and then plug them, they stay off as default. Can I force ON?

Thank You

Koenkk, 2019-03-01:
Is this a setting that can be set when using the Xiaomi gateway?

soloam, 2019-03-01:
I don't think so! I don't use the gateway for a long time, but I don't
recall seeing this option in the gateway!

I was thinking on something like.... If the device becomes online, set
state to on! For example! I can do this in the code, but in the core
zigbee2mqtt would make it more safe!

Thank you

soloam, 2019-03-01:
I checked... in the MiHome App there is a option to restore the state to the previews one! That would work!

ryanbeaton, 2019-03-01:
https://github.com/Koenkk/zigbee2mqtt/issues/297#issuecomment-415929791

Koenkk, 2019-03-03:
Can you sniff the zigbee traffic while setting this option? http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

soloam, 2019-03-03:
Yes I'll do that tomorrow! Thank you

soloam, 2019-03-04:
Hello, here is the packet capture 

[export.zip](https://github.com/Koenkk/zigbee2mqtt/files/2926389/export.zip)

From what I can see he changes the 3 atributs in a row

Koenkk, 2019-03-04:
@soloam I don't have your network keys, either give me the network key, or you can post screenshots with all info here.

soloam, 2019-03-04:
@Koenkk how can I send you a private message?

Koenkk, 2019-03-04:
`@Koenkk` on telegram.

Koenkk, 2019-03-04:
Please try with the following changes: https://github.com/Koenkk/zigbee-shepherd-converters/commit/49a728aca474337610d55a9fe3ac402fe691b2e9

In the capture, it sends 4 different values to the same cluster/attribute ID. We need to figure out the correct one.
Send to `zigbee2mqtt/[DEVICE_ID]/set` payload `{"power_on_behavior": VALUE}` with the following VALUES:
- `aa:80:03:d3:07:41:03`
- `aa:80:03:d3:07:40:02`
- `aa:80:03:d3:07:3f:01`
- `aa:80:05:d1:47:3e:01:10:01`

After send each value, check if it uses the expected power on behavior.

soloam, 2019-03-05:
Hello @Koenkk 

I tried the chages, but it did not work. I publish the commands and I get this on the log

`2019-3-5 09:09:47 - info: Zigbee publish to device '0x00158d00028a5428', genBasic - write - [{"attrId":65520,"dataType":65,"attrData":"aa:80:03:d3:07:3f:01"}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null`

No changes in the power on

Thank you

Koenkk, 2019-03-06:
When sniffing the command being send by zigbee2mqtt, Wireshark showed that the packet was malformed. It seems that zigbee-shepherd doesn't correctly from the octet string. Need to dive deeper here.

wimpie007, 2019-03-08:
a default state would also be nice for the tradfri lamps...
They always are fysically ON after power on/off, and i get "off" as default state in HA.
(even last state is not always correct: after an "unavailable" it should always report "off".

Koenkk, 2019-03-09:
@wimpie007 unfortunately TRADFRI lamps don't support this, it's somethings that has to be solved at the firmware level.

wimpie007, 2019-03-09:
coudnt we have a default state 'on' option?
now it is default 'off'. This would solve it for devices who doesnt remember its state after power on/off...

Koenkk, 2019-03-09:
@wimpie007 as you are using device availability, this should be fetched once this device comes back online. Please try with zigbee2mqtt 1.2.0, if it's not working please report in #775

wimpie007, 2019-03-12:
@koenkk: indeed! works in 1.2.1!
THANKS

closed 2019-05-12 by Koenkk

Thulinma, 2020-01-05:
> When sniffing the command being send by zigbee2mqtt, Wireshark showed that the packet was malformed. It seems that zigbee-shepherd doesn't correctly from the octet string. Need to dive deeper here.

I would like to pick this back up, as I also have a pair of ZNCZ02LM's and would like to be able to set the power-on behavior similarly.
@Koenkk I tried to update your code to the current format (I think...?) as follows:

```
    ZNCZ02LM_power_on_behavior: {
        key: ['power_on_behavior'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('genBasic', {0xFFF0: {value: value, type: 0x41}}, options.xiaomi);
        },
    },
```

But that fails with the message `zigbee2mqtt:error 2020-01-05T16:48:27: Publish 'set' 'power_on_behavior' to '[DEVICE_ID]' failed: 'Error: Write for 'OCTETSTR' not available'`, and I'm not sure how to change that part...

Any hints? :smiley:

Koenkk, 2020-01-09:
@Thulinma implemented, will be available in the latest dev branch in a few hours, documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/ZNCZ02LM.md#power-outage-memory

Thulinma, 2020-01-09:
Works like a charm, thanks! You just made my day :+1:

sirs2k, 2020-03-25:
I'm on latest Dev and I don't know why I'm getting `No converter available for 'power_outage_memory' (true)`
By the way, does this restore the color settings too?

It's very annoying to have to change the color back to daylight whenever the kids turn the lights off from the switch

Electronlibre2012, 2020-12-04:
> 
> 
> Works like a charm, thanks! You just made my day 👍

Hello @Thulinma ,

i have folow the instructions but it doesnt work  : 
mqtt.publish
  topic: zigbee2mqtt/Prise Salon tv/set
  payload: "power_outage_memory: true"

then push the switch on the plug from OFF to ON,

then unplug / wait 5 secondes / plug again and the plug stay off.

I have not the DEV version of zigbee2mqtt, does i need to install the "edge" version to make it works please?

Thanks for your reply...i am stuck.

Thulinma, 2020-12-04:
> i have folow the instructions but it doesnt work :
> mqtt.publish
> topic: zigbee2mqtt/Prise Salon tv/set
> payload: "power_outage_memory: true"
Make sure you use the payload `{"power_outage_memory": true}`, exactly like that. I'm not sure what application/syntax that is, but it looks like it doesn't match the required payload format...? I could be wrong about that, of course.

Electronlibre2012, 2020-12-04:
thanks for your help but : 

mqtt.publish
topic: zigbee2mqtt/Prise Salon tv/set
payload: {"power_outage_memory": true}

return this error : "voluptuous.error.MultipleInvalid: value should be a string for dictionary value @ data['payload']"

so that why i use "power_outage_memory: true".

I have MQTT Explorer install on win10 and i can see on the zigbee2mqqt/Prise Salon tv/set : power_outage_memory: true

How do you do to publish {"power_outage_memory": true} for make it works for you please?

I publish in "service" from "development tools" menu  in hassio.

Thanks

jusxon, 2022-10-26:
@Koenkk Is possible to get this supported on the US version of the plug (ZNCZ12LM)? It supports the same features as the EU plug, but when using the same settings as the EU plug in a custom converter on the US plug, it gives the error `Publish 'get' 'power_outage_memory' to 'Plug' failed: 'Error: Not supported'` as well as the error;

`Error: Not supported at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:2235:23) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (node:events:539:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27) at WebSocket.emit (node:events:527:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1178:20) at Receiver.emit (node:events:527:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:528:14)`

Koenkk, 2022-10-26:
@jusxon done!

jusxon, 2022-10-28:
Thank you very much!

thekev, 2023-01-21:
I have a ZNCZ12LM showing firmware build date 12-24-2021. By date alone it appears to be a match for latest in zigbee-OTA - no other version info is exposed. On what firmware is this tested @jusxon ? It does not work for me. In attempting to toggle the value in `Exposes`, I receive the error:
```
Publish 'set' 'power_outage_memory' to 'plug1' failed: 'Error: Write 0x00158d0008ceee97/1 aqaraOpple({"513":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```
I also receive a similar error for `overload_protection`, `led_disabled_night` and `auto_off`.

junosuarez, 2023-02-24:
I'm seeing the same as @thekev with `UNSUPPORTED_ATTRIBUTE` errors on read and write to `overload_protection`, `led_disabled_night` and `auto_off`. I purchased the device recently (February 2023):

- Zigbee model: lumi.plug.maus01
- Zigbee manufacturer: LUMI
- Firmware: 12-24-2021

Example logs:

```
Debug 2023-02-24 09:31:47Received MQTT message on 'zigbee2mqtt/c-1007-aqara-power-plug/set' with data '{"led_disabled_night":true}'
Debug 2023-02-24 09:31:47Publishing 'set' 'led_disabled_night' to 'c-1007-aqara-power-plug'
Info 2023-02-24 09:31:47MQTT publish: topic 'zigbee2mqtt/c-1007-aqara-power-plug', payload '{"consumption":0.74,"device_temperature":29,"energy":0.74,"last_seen":"2023-02-24T09:31:47-08:00","led_disabled_night":false,"linkquality":168,"overload_protection":null,"power":47.51,"power_outage_count":8,"power_outage_memory":false,"state":"ON","update":{"installed_version":17,"latest_version":17,"state":"idle"}}'
Error 2023-02-24 09:31:47Publish 'set' 'led_disabled_night' to 'c-1007-aqara-power-plug' failed: 'Error: Write 0x00158d0008e1b9b6/1 aqaraOpple({"515":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2023-02-24 09:31:47Error: Write 0x00158d0008e1b9b6/1 aqaraOpple({"515":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:2286:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
```