Koenkk/zigbee2mqtt #1174: How to switch to DEV branch?

salopette, 2019-03-01:
Hello, I keep forgetting How to switch to DEV branch?

Is it possible to document this somewhere?

So it does not work for me ??

`
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt
cp -R data data-backup
git checkout data/configuration.yaml
git checkout dev
git pull
rm -rf node_modules
npm install
cp -R data-backup/* data
rm -rf data-backup
sudo systemctl start zigbee2mqtt`

Thank you

Koenkk, 2019-03-02:
Added to the documentation: http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

closed 2019-03-02 by Koenkk

JacekKac, 2022-09-24:
i have the same isssue right now - did it many times, and now git 
$ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/master

flight4287, 2022-11-07:
I got it this way
$ git checkout -b dev
$  git branch -a
* dev
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/master

npm start

> zigbee2mqtt@1.28.2 start
> node index.js

Building Zigbee2MQTT... (hash changed), finished
Zigbee2MQTT:info  2022-11-07 13:23:20: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-11-07.13-23-20' filename: log.txt
Zigbee2MQTT:info  2022-11-07 13:23:20: Starting Zigbee2MQTT version 1.28.2 (**commit #e66fe85**)

kirovilya, 2022-11-07:
@flight4287 You need call `npm ci` after changing branch

OlsiIgitt, 2023-02-16:
Hi Koen!

1) How can I switch to the dev branch under Windows?

2) When will the changes in dev branch usually be transferred to the master branch?

Thanks in advance...

henkiejan1, 2023-05-02:
I have tried anything to switch to the DEV branch but didn't get it working. It say always "up-to-date" but that's with the regular version. What i have done a many times... (also complete installed zigbee2mqtt again)

`orangepi5:zigbee2mqtt:# git fetch origin dev                                                                                                                         <dev ✗>
From https://github.com/Koenkk/zigbee2mqtt
 * branch            dev        -> FETCH_HEAD
orangepi5:zigbee2mqtt:# git checkout dev                                                                                                                             <dev ✗>
M	data/configuration.yaml
Already on 'dev'
orangepi5:zigbee2mqtt:# git pull                                                                                                                                     <dev ✗>
Already up to date.
orangepi5:zigbee2mqtt:# npm ci                                                                                                                                       <dev ✗>

added 761 packages, and audited 762 packages in 11s

73 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
orangepi5:zigbee2mqtt:# npm start                                                                                                                                    <dev ✗>

> zigbee2mqtt@1.30.4 start
> node index.js

Zigbee2MQTT:debug 2023-05-02 22:54:54: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2023-05-02 22:54:54: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-05-02.22-54-54' filename: log.txt
Zigbee2MQTT:debug 2023-05-02 22:54:54: Removing old log directory '/opt/zigbee2mqtt/data/log/2023-04-15.12-16-53'
Zigbee2MQTT:info  2023-05-02 22:54:54: Starting Zigbee2MQTT version 1.30.4 (commit #b2dd21e)
Zigbee2MQTT:info  2023-05-02 22:54:54: Starting zigbee-herdsman (0.14.111)
`

`>git branch -a

* dev
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/master
`

Anyone a idea????

xX-Nexus-Xx, 2023-08-28:
same issue here ... but   * dev is missing in my list at all

SteveBrackenbury, 2024-01-02:
> I got it this way $ git checkout -b dev $ git branch -a
>
> * dev
>   master
>   remotes/origin/HEAD -> origin/master
>   remotes/origin/master
>
> npm start
>
>
> Building Zigbee2MQTT... (hash changed), finished [...]

This is *NOT* a solution as you've simply created a new "local" git branch called "dev" seeding it from origin/master.

SteveBrackenbury, 2024-01-02:
Run the following command to view your git fetch config:

`git config --get remote.origin.fetch`

The ouput of the command when run on my workstation revealed that fetch is configured to a specific remote origin branch: master.  The effect being that all other remote branches are effectively "hidden":

`+refs/heads/master:refs/remotes/origin/master`

To make other remote branches visible locally, we need to modify the git fetch config by running the following command which substitues "master" with a wildcard "*" so that **ALL** remote branches will be fetched:

`git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"`

After executing the above git command, run the following command to fetch all remote branches:

`git fetch --all`

You will then see the remote dev branch (and other remote branches) successfully fetched.

You should thenbe able to successfully checkout the dev branch:

```
git checkout dev

M       data/configuration.yaml
branch 'dev' set up to track 'origin/dev'.
Switched to a new branch 'dev'
```

diamand2k22, 2024-12-12:
i have a problem with dev branche. change to dev branche works but run npm ci aborted with following error:

`pi4b@pi4b:/opt/zigbee2mqtt$ git checkout dev
Branch 'dev' folgt nun 'origin/dev'.
Zu neuem Branch 'dev' gewechselt
pi4b@pi4b:/opt/zigbee2mqtt$ npm ci
npm error code EUSAGE
npm error
npm error The `npm ci` command can only install with an existing package-lock.js                                                                                                                                                             on or
npm error npm-shrinkwrap.json with lockfileVersion >= 1. Run an install with npm                                                                                                                                                             @5 or
npm error later to generate a package-lock.json file, then try again.
npm error
npm error Clean install a project
npm error
npm error Usage:
npm error npm ci
npm error
npm error Options:
npm error [--install-strategy <hoisted|nested|shallow|linked>] [--legacy-bundlin                                                                                                                                                             g]
npm error [--global-style] [--omit <dev|optional|peer> [--omit <dev|optional|pee                                                                                                                                                             r> ...]]
npm error [--include <prod|dev|optional|peer> [--include <prod|dev|optional|peer                                                                                                                                                             > ...]]
npm error [--strict-peer-deps] [--foreground-scripts] [--ignore-scripts] [--no-a                                                                                                                                                             udit]
npm error [--no-bin-links] [--no-fund] [--dry-run]
npm error [-w|--workspace <workspace-name> [-w|--workspace <workspace-name> ...]                                                                                                                                                             ]
npm error [-ws|--workspaces] [--include-workspace-root] [--install-links]
npm error
npm error aliases: clean-install, ic, install-clean, isntall-clean
npm error
npm error Run "npm help ci" for more info
npm error A complete log of this run can be found in: /home/pi4b/.npm/_logs/2024                                                                                                                                                             -12-12T17_58_40_098Z-debug-0.log
`

anyone have an idea what cause this problem?

floxigen, 2024-12-16:
> i have a problem with dev branche. change to dev branche works but run npm ci aborted with following error:
> 
> ![image](https://private-user-images.githubusercontent.com/119601644/395277571-c6fcbda0-8f85-4e28-b646-8daffb620222.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzNDU5OTEsIm5iZiI6MTczNDM0NTY5MSwicGF0aCI6Ii8xMTk2MDE2NDQvMzk1Mjc3NTcxLWM2ZmNiZGEwLThmODUtNGUyOC1iNjQ2LThkYWZmYjYyMDIyMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNlQxMDQxMzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YTBkZWNlMDA2YTEyNjlmYjNhZWJmMDlmM2RlMjA0NDQ5YWJjMDYzMjg5YWNhMzFkNWZlNTg5MDBkM2Q3YmMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jLHqm6WltXmWpTLduYk0dOnUoT9sneAF0bVDCofycPY)
> 
> anyone have an idea what cause this problem?

run `npm install`

D3rPaPaH0d3n, 2025-01-18:
No matter what I try, after "git checkout dev" I always get the message "data/configuration.yaml: needs merge
error: you need to resolve your current index first"

Has anything changed in the steps? Or am I just too stupid?

THX in advanced!