Koenkk/zigbee2mqtt #1176: Switching Tradfri outlet On reports wrong state

bizziebis, 2019-03-02:
**Zigbee2mqtt version 1.1.1 (commit #4ea55fe)
Coordinator firmware CC2531ZNP-Prod_20190223**

When I switch ON a Tradfri outlet, the returned status is OFF. This will prevent me from switching OFF this device from Domoticz (because it shows already OFF state)

```
2019-3-2 11:59:50 - debug: Received MQTT message on 'zigbee2mqtt/Lamp Aquarium/set' with data '{"state": "ON"}'
2019-3-2 11:59:51 - info: Zigbee publish to device '0x000d6ffffe9c3f6e', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-3-2 11:59:51 - info: MQTT publish: topic 'zigbee2mqtt/Lamp Aquarium', payload '{"state":"ON","linkquality":47}'
2019-3-2 11:59:51 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0x000d6ffffe9c3f6e)
2019-3-2 11:59:51 - info: MQTT publish: topic 'zigbee2mqtt/Lamp Aquarium', payload '{"state":"ON","linkquality":47}'
2019-3-2 11:59:51 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0x000d6ffffe9c3f6e)
2019-3-2 11:59:51 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"16387":255,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}}' of device 'TRADFRI control outlet' (0x000d6ffffe9c3f6e)
2019-3-2 11:59:51 - info: MQTT publish: topic 'zigbee2mqtt/Lamp Aquarium', payload '{"state":"ON","linkquality":47}'
2019-3-2 11:59:51 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"65533":1}}' of device 'TRADFRI control outlet' (0x000d6ffffe9c3f6e)
2019-3-2 11:59:51 - info: MQTT publish: topic 'zigbee2mqtt/Lamp Aquarium', payload '{"state":"OFF","linkquality":47}'
2019-3-2 11:59:51 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0x000d6ffffe9c3f6e)
```

peterjc-123, 2019-03-02:
Nice

Koenkk, 2019-03-02:
Should be fixed in the dev branch, thanks for reporting this.

bizziebis, 2019-03-02:
Thanks! Looks good so far.

salopette, 2019-03-02:
Coordinator firmware CC2531ZNP-Prod_20190223

where are these?

Koenkk, 2019-03-02:
@salopette https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin

@bizziebis great, thanks!

closed 2019-03-02 by Koenkk