Koenkk/zigbee2mqtt #1177: Hue dimmer switch : PM Failed to configure

Bastiencc, 2019-03-02:
Hey.

My hue dimmer switch are paired and works great.
But when I restart zigbee2mqtt I get this (it's not always the same switch that have error) : 

`
Mar 02 16:19:40 raspberrypi npm[5510]:   zigbee2mqtt:error 3/2/2019, 4:19:40 PM Failed to configure hue_switch_2 (0x0017880104af5115) ('undefined')
Mar 02 16:19:40 raspberrypi npm[5510]:   zigbee2mqtt:error 3/2/2019, 4:19:40 PM Failed to configure hue_switch_3 (0x001788010408bb63) ('undefined')
Mar 02 16:19:40 raspberrypi npm[5510]:   zigbee2mqtt:error 3/2/2019, 4:19:40 PM Failed to configure hue_switch_1 (0x001788010408cda7) ('undefined')
`

What do I need to do ?

Koenkk, 2019-03-02:
This fails because the device is sleeping, not a problem as it only has to succeeds once (it has if it is working).

Logging has already been adapted in the dev branch to prevent confusion

closed 2019-03-02 by Koenkk

Bastiencc, 2019-03-02:
Ok great, thanks for your answer ! And also, thanks for your great work !!!