Koenkk/zigbee2mqtt #1178: flash cc2530

yaniv1111, 2019-03-03:
Hi

whats should be the steps flashing cc2530 with this board

link:
https://www.aliexpress.com/item/ZigBee-super-far-wireless-Communication-over-1500-meters-evaluation-kit-CC2530-Eval-Kit2/2030752764.html?spm=a2g0s.9042311.0.0.77be4c4dbJV2Jh

thanks

hwijers, 2019-03-03:
Just as described here: [Flashing the firmware on the CC2531 USB stick](https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html)
without the converter circuit board.
Then use this image: [image](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2530_CC2591/bin)
This should be compatible with your cc2530 + rfx2401 module.

In your zigbee2mqtt configuration.yaml you have to set rtscts to false.

yaniv1111, 2019-03-03:
thank @hwijers 
how do i pair it with my network ?

hfellner, 2019-03-08:
I just noticed, I have a couple of such boards laying around somewhere. Just out of curiosity, if not using the WaveShare "motherboard" and its USB to serial converter, but the serial pins of the module and connecting them to the Raspberry PI serial pins, and also the [RTS/CTS pins on the P5 Header](https://ethertubes.com/raspberry-pi-rts-cts-flow-control/), it should be able to do hardware flow control, shouldn't it?
I would imagine this solution would be preferable to the CC2531 for its greater stability (no USB) and much greater range.

closed 2019-05-14 by stale[bot]