Koenkk/zigbee2mqtt #1179: Nue Smart ZigBee Light Dimmer Switch (HGZB-02A)

Eeeeeediot, 2019-03-03:
Hi,

I am on hass.io using the edge addon. I have the Nue Smart Dimmer (HGZB-02A) dimmer which is not supported. Its brother (HGZB-01A) is already supported.

https://www.amazon.co.uk/gp/product/B07GSK1HVP/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

Can you add support please? The database log is below.

Thanks

{"id":8,"type":"Router","ieeeAddr":"0x00158d00025c04e5","nwkAddr":10637,"manufId":4478,"manufName":"FeiBit","powerSource":"Mains (single phase)","modelId":"FNB56-ZSC01LX1.2","epList":[1],"status":"online","joinTime":1551604468,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"RtCmymnDqy3UpM7r"}

Koenkk, 2019-03-03:
Can you try to add to `devices.js` (http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html):

```js
{
        zigbeeModel: ['FNB56-ZSC01LX1.2'],
        model: 'HGZB-01A',
        vendor: 'Nue / 3A',
        description: 'ZigBee smart light controller',
        extend: generic.light_onoff_brightness,
    },
```

Eeeeeediot, 2019-03-03:
HI @Koenkk . Thanks for your speedy reply.

I think i did it correctly but i got this log.

```
[Info] Searching for custom devices file in zigbee2mqtt data path...
2019-03-03T21:28:19: PM2 log: Launching in no daemon mode
2019-03-03T21:28:19: PM2 log: App [npm:0] starting in -fork mode-
2019-03-03T21:28:19: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /app
> node index.js
  zigbee2mqtt:info 3/3/2019, 9:28:24 PM Logging to directory: '/share/zigbee2mqtt/log/2019-03-03.21-28-23'
/app/node_modules/zigbee-shepherd-converters/devices.js:3
        model: 'HGZB-01A',
             ^
SyntaxError: Unexpected token :
    at new Script (vm.js:79:7)
    at createScript (vm.js:251:10)
    at Object.runInThisContext (vm.js:303:10)
    at Module._compile (internal/modules/cjs/loader.js:657:28)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-03T21_28_25_283Z-debug.log
```

Eeeeeediot, 2019-03-03:
@koenkk

Should the model number not be HGZB-02A?

I didnt know how to add to a devices.js file so I created a new one in the correct path with only this entry. I hope that is what I am supposed to do.

Thanks

Koenkk, 2019-03-04:
Yes, that should indeed be the model number, this is not the reason for the crash. Can you post your devices.js?

Eeeeeediot, 2019-03-04:
Here you go. I have pasted a photo of the devices.js when i open it with VSC. Hopefully you can spot the bug.

```
{
    zigbeeModel: ['FNB56-ZSC01LX1.2'],
    model: 'HGZB-01A',
    vendor: 'Nue / 3A',
    description: 'ZigBee smart light controller',
    extend: generic.light_onoff_brightness,
},
```

[image]

Koenkk, 2019-03-04:
Please try with

```js
const devices = [
{
    zigbeeModel: ['FNB56-ZSC01LX1.2'],
    model: 'HGZB-01A',
    vendor: 'Nue / 3A',
    description: 'ZigBee smart light controller',
    extend: generic.light_onoff_brightness,
},
];

module.exports = devices.map((device) =>
    device.extend ? Object.assign({}, device.extend, device) : device
);
```

Eeeeeediot, 2019-03-04:
@Koenkk 

It crashed with the following error

```2019-03-04T18:32:20: PM2 log: Launching in no daemon mode
2019-03-04T18:32:20: PM2 log: App [npm:0] starting in -fork mode-
2019-03-04T18:32:20: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /app
> node index.js
  zigbee2mqtt:info 3/4/2019, 6:32:24 PM Logging to directory: '/share/zigbee2mqtt/log/2019-03-04.18-32-24'
/app/node_modules/zigbee-shepherd-converters/devices.js:8
      fromZigbee: [fz.brightness, fz.state_change, fz.state, fz.brightness_report, fz.ignore_genGroups_devChange],
                   ^
ReferenceError: fz is not defined
    at Object.<anonymous> (/app/node_modules/zigbee-shepherd-converters/devices.js:8:20)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/app/node_modules/zigbee-shepherd-converters/index.js:3:17)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
npm
 ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-04T18_32_26_211Z-debug.log
```

Koenkk, 2019-03-04:
Please try with

```js
'use strict';

const debug = require('debug')('zigbee-shepherd-converters:devices');
const fz = require('./converters/fromZigbee');
const tz = require('./converters/toZigbee');

const devices = [
  {
    zigbeeModel: ['FNB56-ZSC01LX1.2'],
    model: 'HGZB-01A',
    vendor: 'Nue / 3A',
    description: 'ZigBee smart light controller',
    supports: 'on/off, brightness',
    fromZigbee: [fz.brightness, fz.state_change, fz.state, fz.brightness_report, fz.ignore_genGroups_devChange],
    toZigbee: [tz.light_onoff_brightness, tz.ignore_transition, tz.light_alert],
  },
];

module.exports = devices.map((device) =>
    device.extend ? Object.assign({}, device.extend, device) : device
);
```

Eeeeeediot, 2019-03-04:
@Koenkk 

Yes! It works (kind of). The light toggles off and has a brightness slider when you click into it. The lightbulb icon changes colour depending on brightness but there is no difference in the bulb itself. I can confirm that i have dimmable bulbs.

any ideas? Also, should I correct the model number?

here is the log incase you need it
```
zigbee2mqtt:info 3/4/2019, 6:46:46 PM MQTT publish: topic 'homeassistant/light/0x00158d00025c04e5/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/0x00158d00025c04e5/set","state_topic":"zigbee2mqtt/0x00158d00025c04e5","name":"0x00158d00025c04e5_light","unique_id":"0x00158d00025c04e5_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00025c04e5","name":"0x00158d00025c04e5","sw_version":"Zigbee2mqtt 1.1.1","model":"ZigBee smart light controller (HGZB-01A)","manufacturer":"Nue / 3A"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/4/2019, 6:46:46 PM MQTT publish: topic 'homeassistant/sensor/0x00158d00025c04e5/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d00025c04e5","json_attributes_topic":"zigbee2mqtt/0x00158d00025c04e5","name":"0x00158d00025c04e5_linkquality","unique_id":"0x00158d00025c04e5_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00025c04e5","name":"0x00158d00025c04e5","sw_version":"Zigbee2mqtt 1.1.1","model":"ZigBee smart light controller (HGZB-01A)","manufacturer":"Nue / 3A"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

Koenkk, 2019-03-04:
Can you provide the debug log when dimming? (http://www.zigbee2mqtt.io/configuration/configuration.html)

Eeeeeediot, 2019-03-04:
The bulb kind of double blinks when it comes on as well. It doesn't do this when the button on the device is pressed manually or when power is cut and reconnected to the device/bulb.

I don't remember getting this before
```
zigbee2mqtt:warn 3/4/2019, 7:00:04 PM Device with modelID 'lumi.ctrl_neutral2' is not supported.
```
log
```zigbee2mqtt:warn 3/4/2019, 6:54:27 PM Device with modelID 'lumi.ctrl_neutral2' is not supported.
  zigbee2mqtt:warn 3/4/2019, 6:57:09 PM No converter available for 'HGZB-02A' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"manufacturerName":"FeiBit","modelId":"FNB56-ZSC01LX1.2","powerSource":1}}'
  zigbee2mqtt:warn 3/4/2019, 6:57:09 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 6:57:09 PM No converter available for 'HGZB-02A' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"FeiBit","modelId":"FNB56-ZSC01LX1.2","powerSource":1}}'
  zigbee2mqtt:warn 3/4/2019, 6:57:09 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 3/4/2019, 6:57:09 PM Device incoming...```

```  zigbee2mqtt:info 3/4/2019, 6:59:32 PM Zigbee publish to device '0x00158d00025c04e5', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 3/4/2019, 6:59:32 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":128,"brightness":255}'
  zigbee2mqtt:info 3/4/2019, 6:59:33 PM Zigbee publish to device '0x00158d00025c04e5', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 3/4/2019, 6:59:33 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":254}'
  zigbee2mqtt:info 3/4/2019, 6:59:33 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":254}'
  zigbee2mqtt:info 3/4/2019, 6:59:33 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":254}'
  zigbee2mqtt:info 3/4/2019, 6:59:33 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":123,"brightness":254}  zigbee2mqtt:warn 3/4/2019, 7:00:04 PM Device with modelID 'lumi.ctrl_neutral2' is not supported.```

Eeeeeediot, 2019-03-04:
@Koenkk 

Sorry - i figured out what you meant after my latest comment.

log
```
3/4/2019, 7:04:37 PM - info: Logging to directory: '/share/zigbee2mqtt/log/2019-03-04.19-04-37'
3/4/2019, 7:04:39 PM - info: Starting zigbee2mqtt version 1.1.1 (commit #unknown)
3/4/2019, 7:04:39 PM - info: Starting zigbee-shepherd
3/4/2019, 7:04:40 PM - info: zigbee-shepherd started
3/4/2019, 7:04:40 PM - info: Coordinator firmware version: '20180815'
3/4/2019, 7:04:40 PM - info: Currently 7 devices are joined:
3/4/2019, 7:04:40 PM - info: 0x00158d0001d0c6f6 (0x00158d0001d0c6f6): unkown - undefined unknown (Router)
3/4/2019, 7:04:40 PM - info: 0x00158d0002f42d79 (0x00158d0002f42d79): unkown - undefined unknown (Router)
3/4/2019, 7:04:40 PM - info: 0x00158d0002b43011 (0x00158d0002b43011): unkown - undefined unknown (EndDevice)
3/4/2019, 7:04:40 PM - info: 0x00158d0001e7fec9 (0x00158d0001e7fec9): unkown - undefined unknown (EndDevice)
3/4/2019, 7:04:40 PM - info: 0x00158d0001e80911 (0x00158d0001e80911): unkown - undefined unknown (EndDevice)
3/4/2019, 7:04:40 PM - info: 0x00158d00026df558 (0x00158d00026df558): unkown - undefined unknown (Router)
3/4/2019, 7:04:40 PM - info: 0x00158d00025c04e5 (0x00158d00025c04e5): HGZB-02A - Nue / 3A ZigBee smart light controller (Router)
3/4/2019, 7:04:40 PM - info: Zigbee: disabling joining new devices.
3/4/2019, 7:04:40 PM - info: Connecting to MQTT server at mqtt://192.168.0.29:1883
3/4/2019, 7:04:40 PM - info: zigbee-shepherd ready
3/4/2019, 7:04:40 PM - info: Connected to MQTT server
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001d0c6f6', payload '{"state":"OFF","brightness":254,"linkquality":7}'
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f42d79', payload '{"state":"OFF","brightness":254,"linkquality":44}'
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002b43011', payload '{"illuminance":12,"linkquality":118,"occupancy":false,"battery":91,"voltage":2985}'
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e7fec9', payload '{"battery":100,"voltage":3062,"linkquality":68}'
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e80911', payload '{"battery":100,"voltage":3052,"linkquality":68}'
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00026df558', payload '{"state_left":"OFF","linkquality":39,"state_right":"OFF"}'
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":128,"brightness":31}'
3/4/2019, 7:04:40 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":false}'
3/4/2019, 7:04:49 PM - info: Zigbee publish to device '0x00158d00025c04e5', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:04:49 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"OFF","linkquality":128,"brightness":31}'
3/4/2019, 7:04:51 PM - info: Zigbee publish to device '0x00158d00025c04e5', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:04:51 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":128,"brightness":31}'
3/4/2019, 7:04:52 PM - info: Zigbee publish to device '0x00158d00025c04e5', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:04:52 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"OFF","linkquality":128,"brightness":31}'
3/4/2019, 7:04:53 PM - info: Zigbee publish to device '0x00158d00025c04e5', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:04:53 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":128,"brightness":31}'
3/4/2019, 7:04:56 PM - info: Zigbee publish to device '0x00158d00025c04e5', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:04:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":128,"brightness":255}'
3/4/2019, 7:04:56 PM - info: Zigbee publish to device '0x00158d00025c04e5', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:04:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":254}'
3/4/2019, 7:04:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":254}'
3/4/2019, 7:04:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":254}'
3/4/2019, 7:04:58 PM - info: Zigbee publish to device '0x00158d00025c04e5', genLevelCtrl - moveToLevelWithOnOff - {"level":177,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:04:58 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":177}'
3/4/2019, 7:04:58 PM - info: Zigbee publish to device '0x00158d00025c04e5', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:04:58 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":177}'
[... cut ...]
3/4/2019, 7:05:31 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":120,"brightness":254}'
3/4/2019, 7:05:31 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":120,"brightness":254}'
3/4/2019, 7:05:33 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":120,"brightness":254}'
3/4/2019, 7:05:33 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":120,"brightness":254}'
```

Koenkk, 2019-03-04:
I don't see any debug level (it's the `log_level` setting, https://github.com/danielwelch/hassio-zigbee2mqtt)

Eeeeeediot, 2019-03-04:
My bad

```
3/4/2019, 7:12:53 PM - info: Logging to directory: '/share/zigbee2mqtt/log/2019-03-04.19-12-53'
3/4/2019, 7:12:53 PM - debug: Removing old log directory '/share/zigbee2mqtt/log/2019-03-04.18-45-48'
3/4/2019, 7:12:54 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
3/4/2019, 7:12:54 PM - debug: Loaded state from file /share/zigbee2mqtt/state.json
3/4/2019, 7:12:54 PM - debug: Saving state to file /share/zigbee2mqtt/state.json
3/4/2019, 7:12:54 PM - info: Starting zigbee2mqtt version 1.1.1 (commit #unknown)
3/4/2019, 7:12:54 PM - info: Starting zigbee-shepherd
3/4/2019, 7:12:55 PM - info: zigbee-shepherd started
3/4/2019, 7:12:55 PM - info: Coordinator firmware version: '20180815'
3/4/2019, 7:12:55 PM - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ecde65","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1551726775,"joinTimeLeft":0}
3/4/2019, 7:12:55 PM - info: Currently 7 devices are joined:
3/4/2019, 7:12:55 PM - info: 0x00158d0001d0c6f6 (0x00158d0001d0c6f6): unkown - undefined unknown (Router)
3/4/2019, 7:12:55 PM - info: 0x00158d0002f42d79 (0x00158d0002f42d79): unkown - undefined unknown (Router)
3/4/2019, 7:12:55 PM - info: 0x00158d0002b43011 (0x00158d0002b43011): unkown - undefined unknown (EndDevice)
3/4/2019, 7:12:55 PM - info: 0x00158d0001e7fec9 (0x00158d0001e7fec9): unkown - undefined unknown (EndDevice)
3/4/2019, 7:12:55 PM - info: 0x00158d0001e80911 (0x00158d0001e80911): unkown - undefined unknown (EndDevice)
3/4/2019, 7:12:55 PM - info: 0x00158d00026df558 (0x00158d00026df558): unkown - undefined unknown (Router)
3/4/2019, 7:12:55 PM - info: 0x00158d00025c04e5 (0x00158d00025c04e5): HGZB-02A - Nue / 3A ZigBee smart light controller (Router)
3/4/2019, 7:12:55 PM - info: Zigbee: disabling joining new devices.
3/4/2019, 7:12:55 PM - info: Connecting to MQTT server at mqtt://192.168.0.29:1883
3/4/2019, 7:12:56 PM - info: zigbee-shepherd ready
3/4/2019, 7:12:56 PM - debug: Mounted the cieApp (epId 11)
3/4/2019, 7:12:56 PM - info: Connected to MQTT server
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001d0c6f6', payload '{"state":"OFF","brightness":254,"linkquality":7}'
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f42d79', payload '{"state":"OFF","brightness":254,"linkquality":44}'
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002b43011', payload '{"illuminance":12,"linkquality":118,"occupancy":false,"battery":91,"voltage":2985}'
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e7fec9', payload '{"battery":100,"voltage":3062,"linkquality":68}'
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e80911', payload '{"battery":100,"voltage":3052,"linkquality":68}'
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00026df558', payload '{"state_left":"OFF","linkquality":39,"state_right":"OFF"}'
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":128,"brightness":31}'
3/4/2019, 7:12:56 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":false}'
3/4/2019, 7:13:09 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d00025c04e5/set' with data '{"state": "OFF"}'
3/4/2019, 7:13:09 PM - info: Zigbee publish to device '0x00158d00025c04e5', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:13:09 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"OFF","linkquality":128,"brightness":31}'
3/4/2019, 7:13:12 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d00025c04e5/set' with data '{"state": "ON"}'
3/4/2019, 7:13:12 PM - info: Zigbee publish to device '0x00158d00025c04e5', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:13:12 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":128,"brightness":31}'
3/4/2019, 7:13:15 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d00025c04e5/set' with data '{"state": "OFF"}'
3/4/2019, 7:13:15 PM - info: Zigbee publish to device '0x00158d00025c04e5', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:13:15 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"OFF","linkquality":128,"brightness":31}'
3/4/2019, 7:13:17 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d00025c04e5/set' with data '{"state": "ON"}'
3/4/2019, 7:13:17 PM - info: Zigbee publish to device '0x00158d00025c04e5', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/4/2019, 7:13:17 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":128,"brightness":31}'
3/4/2019, 7:13:26 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d00025c04e5/set' with data '{"state": "ON", "brightness": 255}'
[... cut ...]
3/4/2019, 7:13:49 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":254}'
3/4/2019, 7:13:49 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'FNB56-ZSC01LX1.2' (0x00158d00025c04e5)
3/4/2019, 7:13:49 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00025c04e5', payload '{"state":"ON","linkquality":126,"brightness":254}'
3/4/2019, 7:13:56 PM - debug: Ping 0x00158d00026df558
3/4/2019, 7:14:06 PM - error: Failed to ping 0x00158d00026df558
```

Eeeeeediot, 2019-03-04:
hey @Koenkk 

Its working! My GF swapped out the bulb to a non dimmable one

There doesn't seem to be a blink either.

Thank you very much!

Eeeeeediot, 2019-03-04:
On a side note,

Is there anyway to change the way a device's domain is registered on Home Assistant?

I have a "Aqara double key wired wall switch (QBKG03LM) by Xiaomi" but it presents as a switch rather than a light. I can use the "light switch" component in HA to alter this but its a workaround rather than a fix.

Is there anyway to change its domain?

Also, when is it likely that support to the "HGZB-02A - Nue / 3A ZigBee smart light controller" will eb avaiable on the edge addon?

Thanks for all your hard work!!!

Koenkk, 2019-03-05:
Changing the domain is not possible (without zigbee2mqtt code changes).

Will be supported in the edge addon tomorrow.

closed 2019-03-05 by Koenkk

Nosstrom, 2019-09-26:
Does the Livolo zigbee dimmer supported ?