Koenkk/zigbee2mqtt #1182: Philips dimmer switch now working anymore

JumpmanJunior, 2019-03-03:
With the newest dev version (probably because of the update of the zigbee-shepherd-converters) the Philips dimmer switch is not working anymore.

> 2019-3-3 21:46:50 - debug: Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'RWL021' (0x0017880104e680f9)
> 2019-3-3 21:46:50 - warn: No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
> 2019-3-3 21:46:50 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
> 2019-3-3 21:50:35 - debug: Received zigbee message of type 'cmdOffWithEffect' with data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}' of device 'RWL021' (0x0017880104e680f9)
> 2019-3-3 21:50:35 - warn: No converter available for '324131092621' with cid 'genOnOff', type 'cmdOffWithEffect' and data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}'
> 2019-3-3 21:50:35 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

richredleg, 2019-03-03:
have the same to update after the last dev branch. If I install Master, the hue dimmer switch (324131092621) will work again.

Notification when pairing in dev...: No converter available for '324131092621' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}'

Koenkk, 2019-03-04:
@ugrug could you help with this?

ugrug, 2019-03-04:
I replaced the genOnOff and genLevelCtrl cluster processing to Philips manufacturer specific cluster as it indicates the same event. 
I think as the device already binded to these cluster can cause these messages.
I will add an option for switching between genOnOff and genLevel / Philips manufacturer specific cluster processing.

Koenkk, 2019-03-04:
@urug I dont think we should do that, existing users just have to factory reset and re-pair the device.

hfellner, 2019-03-04:
I ran into the same problems yesterday, after switching to the dev version with two dimmers already paired. However, when I tried to remove one of them and re-pair it, this would consistently fail, producing lots of error messages about unsupported profiles and missing converters, then even failing to even identify the modelId (reporting as 'undefined'). In about 50 attempts to re-pair, there wasn't a successful attempt.

Switched back to master branch (along with corresponding dongle), started it, and re-pairing the dimmer worked on the first attempt.

So after switching back to the dev branch, I just added the missing converters in my working copy and now both dimmers are working again, but I wanted to let you know that re-pairing doesn't seem to be possible with the dev branch.

ugrug, 2019-03-04:
Now I tried from clean start with `git clone -b dev https://github.com/Koenkk/zigbee2mqtt.git /usr/lib/zigbee2mqtt` and `npm install`.
Having the same problem, on first start.
Finally I checked the `cluster_defs.json` files and `manuSpecificPhilips` cluster missing from `/usr/lib/zigbee2mqtt/node_modules/zcl-id/definitions/cluster_defs.json` and `/usr/lib/zigbee2mqtt/node_modules/zcl-packet/node_modules/zcl-id/definitions` the other 2 `cluster_defs.json` file content right.
The same for `common.json`, missing `manuSpecificPhilips` cluster from
`/usr/lib/zigbee2mqtt/node_modules/zcl-packet/node_modules/zcl-id/definitions/common.json` and `/usr/lib/zigbee2mqtt/node_modules/zcl-id/definitions/common.json`.

Koenkk, 2019-03-05:
@ugrug my bad, forgot to update something, can you check if this has been fixed?

ugrug, 2019-03-05:
@Koenkk 
Cluster definitions right now, but unfortunatelly I also error got messages.

```
Mar  5 21:52:56 cubie npm[16979]: 2019-03-05T20:52:56.336Z zigbee-shepherd-converters:devices Configured '(cb) => ep2.bind('manuSpecificPhilips', coordinator, cb)' with result 'OK'
Mar  5 21:52:56 cubie npm[16979]: 2019-03-05T20:52:56.641Z zigbee-shepherd:request REQ --> ZDO:bindReq
Mar  5 21:52:56 cubie npm[16979]: 2019-03-05T20:52:56.915Z zigbee-shepherd:msgHdlr IND <-- ZDO:bindRsp
Mar  5 21:52:56 cubie npm[16979]: 2019-03-05T20:52:56.917Z zigbee-shepherd-converters:devices Configured '(cb) => ep2.bind('genPowerCfg', coordinator, cb)' with result 'OK'
Mar  5 21:52:57 cubie npm[16979]: 2019-03-05T20:52:57.236Z zigbee-shepherd-converters:devices Configured '(cb) => ep2.report('genPowerCfg', 'batteryPercentageRemaining', 0, 1000, 0, cb)' with result 'Error: Profile: 49246 is not supported.'
Mar  5 21:52:57 cubie npm[16979]: #033[33m  zigbee2mqtt:warn#033[39m 2019-3-5 21:52:57 Failed to configure 0x00178801021874fe (0x00178801021874fe) ('Error: Profile: 49246 is not supported.') (attempt #1)
```

I try to find out what is the difference with the master version, what was working for me.

ugrug, 2019-03-05:
Now I just put back the old converters with empty message, this way seem to be working, but shuld be find better solution.

https://github.com/Koenkk/zigbee-shepherd-converters/commit/3619d917127ebbc98315e621a4b13ae4f2289624
https://github.com/Koenkk/zigbee-shepherd-converters/commit/df4c94b570311acf5d9ec2587398d8cf49360c5b

Koenkk, 2019-03-06:
@ugrug could you make a PR? I think these should be `Ignore converters` (which can be found at the bottom of the `fromZigbee.js` file.

ugrug, 2019-03-07:
@Koenkk 
(https://github.com/Koenkk/zigbee-shepherd-converters/pull/322)

Still looking for the best solution as described in (https://github.com/Koenkk/zigbee2mqtt/issues/585#issuecomment-470432478)

ugrug, 2019-03-08:
this issue hopefully solved now.
https://github.com/Koenkk/zigbee2mqtt/issues/585

Koenkk, 2019-03-08:
Should be fixed in the dev branch, can somebody verify this?

hfellner, 2019-03-08:
Can confirm the error messages about unsupported profiles, undefined modelIds, or missing converters, with the Hue dimmers, are gone in the dev branch now.

But while every button on a dimmer press shows up in the zigbee2mqtt debug log, nothing at all gets published to the MQTT server, so the dimmer is still unusable. Are there any extra steps required, to get them working again? (Did the procedure like switching to dev branch, stopped zigbee2mqtt, backed up data, deleted node_modules, git pull, npm install, restored data backup, restarted zigbee2mqtt.)

JumpmanJunior, 2019-03-09:
For me the remote is now working again. MQTT publish is also working. There is even a new "counter" attribute so I am now able to see how many times a button was pressed. In the log there are also additional zigbee messages like the following one:

> 2019-3-9 00:52:14 - debug: Received zigbee message of type 'cmdHueNotification' with data '{"cid":"manuSpecificPhilips","data":{"button":1,"unknown1":3145728,"type":2,"unknown2":0,"time":1}}' of device 'RWL021' (0x0017880104e680f9)

A further change is that I am now also receiving messages from the remote if it is bound to a bulb. In the past zigbee2mqtt stopped receiving these messages until the remote was unbound. But on the other side with the bound remote now I am not able to control the brightness any more, only on and off are working.

hfellner, 2019-03-09:
I had to re-pair the dimmers, now they are working for me, also.

@Koenkk 
Is it normal that there are error messages about modelId="undefined" and unsupported devices during the pairing process?
> zigbee2mqtt:error 3/9/2019, 7:51:06 AM Cannot get the Node Descriptor of the Device: 0x0017880104e5686f (Error: Timed out after 10000 ms)
zigbee2mqtt:debug 3/9/2019, 7:51:09 AM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Philips","modelId":"RWL021","powerSource":3}}' of device 'undefined' (0x0017880104e5686f)
zigbee2mqtt:info 3/9/2019, 7:51:10 AM New device 'undefined' with address 0x0017880104e5686f connected!
zigbee2mqtt:info 3/9/2019, 7:51:10 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x0017880104e5686f","meta":{}}'
zigbee2mqtt:warn 3/9/2019, 7:51:10 AM Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn 3/9/2019, 7:51:10 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:debug 3/9/2019, 7:51:10 AM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}' of device 'RWL021' (0x0017880104e5686f)
zigbee2mqtt:debug 3/9/2019, 7:51:11 AM Received zigbee message of type 'endDeviceAnnce' with data '"0x0017880104b559df"' of device 'SML001' (0x0017880104b559df)
zigbee2mqtt:debug 3/9/2019, 7:51:11 AM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"RWL021","dateCode":"20160302","powerSource":3,"swBuildId":"5.45.1.17846"}}' of device 'RWL021' (0x0017880104e5686f)
zigbee2mqtt:debug 3/9/2019, 7:51:11 AM Received zigbee message of type 'devInterview' with data '"0x0017880104e5686f"'

After the error messages the dimmers still paired successfully and work, though.

The counter function seems a bit flakey. If I repeatedly press a button 3 times, always in the same speed, more often than not I get either 3 messages with counter:1 or 2 messages, one with counter:1 and one with counter:2 than actually getting a message with counter:3 (did happen two times out of 20, though). The counter works more reliably with values >3.

This is just an observation, though, I don't really use repeated button presses.

ugrug, 2019-03-09:
For repeated press, you can try multiple_press_timeout option. By default it's only 250ms.

Koenkk, 2019-03-09:
Great that this is working now.

@hfellner if it's working its no problem.

closed 2019-03-09 by Koenkk