Koenkk/zigbee2mqtt #1183: Added bulbs don't dim 100%

hgelpke, 2019-03-04:
I've recently added 6" commercial electric downlights to my config. They paired fine (though shown as 4") and can turn on/off with no problem.

The only issue I have is that even with the slider at zero in HA, the bulbs are still lit to about 25%

How can I override this either in zigbee2mqtt or HA?

Koenkk, 2019-03-04:
Have you seen these bulbs dimming more with a different gateway (perhaps these just can dim more?)

hgelpke, 2019-03-04:
I have not. They are wink bulbs but i don't have a wink so i paired them
directly with zigbee2mqtt

Koenkk, 2019-03-04:
Just to make sure, could you post your zigbee2mqtt debug log (http://www.zigbee2mqtt.io/configuration/configuration.html). In this way we can verify that zigbee2mqtt is sending the correct data.

hgelpke, 2019-03-05:
> zigbee2mqtt@1.1.1 start /zigbee2mqtt-1.1.1
> node index.js
  zigbee2mqtt:info 2019-3-4 21:46:43 Logging to directory: 'data/log/2019-03-04.21-46-43'
  zigbee2mqtt:debug 2019-3-4 21:46:44 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2019-3-4 21:46:44 Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 2019-3-4 21:46:44 Starting zigbee2mqtt version 1.1.1 (commit #unknown)
  zigbee2mqtt:info 2019-3-4 21:46:44 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-3-4 21:46:44 zigbee-shepherd started
  zigbee2mqtt:info 2019-3-4 21:46:44 Coordinator firmware version: '20190109'
  zigbee2mqtt:debug 2019-3-4 21:46:45 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b001936dc87","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190109},"startTime":1551764804,"joinTimeLeft":0}
  zigbee2mqtt:info 2019-3-4 21:46:45 Currently 14 devices are joined:
  zigbee2mqtt:info 2019-3-4 21:46:45 Kitchen2 (0x000d6f000aa9d1de): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 Kitchen4 (0x000d6f000aa82762): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 Kitchen3 (0x000d6f000aa85d11): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x000d6f00057d7034): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x000d6f000aa9d24f): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x00178801045a1b8c): 9290011370 - Philips Hue white A60 bulb E27 (Router)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x00158d0002b5803f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x00158d0002ebfc4f): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0xd0cf5efffedceca9): E1603 - IKEA TRADFRI control outlet (Router)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x000b57fffe915916): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x000d6ffffe8e95c2): unkown - undefined unknown (Router)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x000d6f00057aa6e4): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 unknown (0x000d6f000b3be4b3): unkown - undefined unknown (EndDevice)
  zigbee2mqtt:info 2019-3-4 21:46:45 Kitchen1 (0x000d6f000aa9a3fe): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
  zigbee2mqtt:warn 2019-3-4 21:46:45 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-3-4 21:46:45 Allowing new devices to join.
  zigbee2mqtt:warn 2019-3-4 21:46:45 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-3-4 21:46:45 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-3-4 21:46:45 Connecting to MQTT server at mqtt://192.168.0.205:1883
  zigbee2mqtt:debug 2019-3-4 21:46:45 Configuring unknown (0xd0cf5efffedceca9) ...
  zigbee2mqtt:debug 2019-3-4 21:46:45 Configuring unknown (0x000b57fffe915916) ...
  zigbee2mqtt:info 2019-3-4 21:46:45 zigbee-shepherd ready
  zigbee2mqtt:info 2019-3-4 21:46:45 Connected to MQTT server
  zigbee2mqtt:info 2019-3-4 21:46:45 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-3-4 21:46:45 MQTT publish: topic 'zigbee2mqtt/Kitchen2', payload '{"state":"ON","color_temp":372,"linkquality":0,"color_mode":2,"color":{"x":0.381,"y":0.377},"brightness":121}'
  zigbee2mqtt:info 2019-3-4 21:46:45 MQTT publish: topic 'zigbee2mqtt/Kitchen4', payload '{"state":"ON","color_temp":372,"brightness":121,"linkquality":28,"color_mode":2,"color":{"x":0.381,"y":0.377}}'
  zigbee2mqtt:info 2019-3-4 21:46:45 MQTT publish: topic 'zigbee2mqtt/Kitchen3', payload '{"state":"ON","color_temp":372,"linkquality":0,"color_mode":2,"color":{"x":0.381,"y":0.377},"brightness":121}'
  zigbee2mqtt:info 2019-3-4 21:46:45 MQTT publish: topic 'zigbee2mqtt/0x000d6f00057d7034', payload '{"state":"ON","linkquality":31,"brightness":159,"color_temp":372,"color_mode":2,"color":{"x":0.381,"y":0.377}}'
  zigbee2mqtt:info 2019-3-4 21:46:45 MQTT publish: topic 'zigbee2mqtt/0x000d6f000aa9d24f', payload '{"state":"ON","linkquality":47,"color_temp":372,"color_mode":2,"color":{"x":0.381,"y":0.377},"brightness":159}'
  zigbee2mqtt:info 2019-3-4 21:46:45 MQTT publish: topic 'zigbee2mqtt/0x00178801045a1b8c', payload '{"state":"OFF"}'
  zigbee2mqtt:info 2019-3-4 21:46:45 MQTT publish: topic 'zigbee2mqtt/0x00158d0002b5803f', payload '{"temperature":22.51,"linkquality":123,"humidity":51.84,"pressure":982,"battery":100,"voltage":3005}'
[... cut ...] :"Kitchen1_light","unique_id":"0x000d6f000aa9a3fe_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f000aa9a3fe","name":"Kitchen1","sw_version":"Zigbee2mqtt 1.1.1","model":"Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (53170161)","manufacturer":"Commercial Electric"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:debug 2019-3-4 21:46:45 Mounted the cieApp (epId 11)
  zigbee2mqtt:info 2019-3-4 21:46:46 Succesfully configured unknown (0xd0cf5efffedceca9)

hgelpke, 2019-03-05:
I am starting to wonder if the bulb only dims so far and it's independent of it's pairing. I had a spare smartthings hub and I paired it to that with a similar result. 

I emailed Wink and they said it dims to 1% but even that 1% seems extremely bright.  Compared to my standard overheads on a caseta (separate circuit/switch) these don't dim nearly as much.

Koenkk, 2019-03-05:
Then it's indeed a hardware limitation of the bulb.

closed 2019-05-04 by Koenkk

jshethbright, 2020-06-02:
> I've recently added 6" commercial electric downlights to my config. They paired fine (though shown as 4") and can turn on/off with no problem.
>
> The only issue I have is that even with the slider at [...]

This is a side question, but how were you able to pair these? I have zigbee2mqtt set up properly with my CC2531 and no matter what I try I cannot even get these downlights to show up in the zigbee2mqtt log

sibero80, 2020-10-18:
Hello all, I'm about to pull the trigger on a similar model, but is listed as 53170162 instead of 53170161.
In your eperience, do a lot of things change on iterations of the same model? could this break support?
If anyone is interested, they are heavily discounted right now. https://www.amazon.com/Commercial-Electric-Matte-Recessed-Downlight/dp/B01LE3757M