Koenkk/zigbee2mqtt #1184: Connecting devices to router

KoumaMisna, 2019-03-04:
I flashed another CC-stick with router firmware and was able to pair it with the coordinator. Also I deleted coordinators database so it has no devices connected.

Now if I try to pair for example Xiaomi Aqara temperature sensors they all pair to the coordinator even though the router is 5cm from the device and coordinator is on another floor.
And how can I tell this, well the end device reports link quality between 7 and 0.

So my question is that how am I able to select to which device I want to pair the end devices?

Koenkk, 2019-03-04:
I wouldn't worry to much about the linkquality and networkmap (it's not complete accurate).

Is the device working properly?

KoumaMisna, 2019-03-04:
No. The devices which have linkquality under 5 are not really working. That's why I ordered another stick to the floor which has most of the devices as router. Is it so that I can't put only the router to accept new connections?

Or is the problem with the Xiaomi devices? Do they somehow remember where they were paired earlier and now try to connect to the same device?

Koenkk, 2019-03-04:
What is the linkquality between the router and the coordinator?

KoumaMisna, 2019-03-04:
I have no idea where to see it.

Here is part of the log from home assistant:
`zigbee2mqtt:info 2019-3-4 19:36:10 MQTT publish: topic 'homeassistant/sensor/0x00124b00193618e8/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00124b00193618e8","json_attributes_topic":"zigbee2mqtt/0x00124b00193618e8","name":"0x00124b00193618e8_linkquality","unique_id":"0x00124b00193618e8_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00124b00193618e8","name":"0x00124b00193618e8","sw_version":"Zigbee2mqtt 1.1.1","model":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)","manufacturer":"Custom devices (DiY)"},"availability_topic":"zigbee2mqtt/bridge/state"}'`

Koenkk, 2019-03-04:
What you could try is re-pairing the device not directly close to the router, but at a little distance, where it's still in range of the router, but not in range of the coordinator.

Landrash, 2019-03-04:
I've had success with the approach @Koenkk suggested.

Move the router a room or two away and place the item I want to pair in close proximity to the router I want pair a device with. 
When doing this I've resorted to removing the device with the remove command before trying to repair the item.

Using a MQTT client for a phone makes this rather easy.

KoumaMisna, 2019-03-04:
@Landrash I tried to go a floor down but still no success. How do I remove the device? Is there some instructions?

Landrash, 2019-03-05:
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html

Remove is a bit down the page.

DerElch87, 2019-03-08:
Hi there,
to be honest, I also don't understand how to pair devices through a router instead of directly via the coordinator.
I tried the solutions like described above, so move the end device far away from the coordinator with a router in between etc.
But nothing helps as you can see in the webgraph below.
All devices are directly connected to the coordinator and I'm scared that I will get problems because of too much devices soon.
The firmware I'm using is the Prod version: 20190109
Currently paired are some of each:
 - Müller Licht Tint LED bulb (5x)
 - Xiaomi Aqara bulb (1x)
 - Aqara motion sensor (4x)
 - Aqara double key (2x)
 - Aqara cube (2x)
I want to add some more lights as also sensors to fully automate my home.
How can I asure that I don't get the "famous" failure messages and avoid reflashing my CC Stick?
Thanks in advance.

BTW, my webgraph looks like this:
`digraph G {
node[shape=record];
  "0x00124b0012013c7c" [style="bold", label="{0x00124b0012013c7c|Coordinator|No model information available|online}"];
  "0x00158d0003410305" [style="rounded", label="{0x00158d0003410305|Router|Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, opal white (404006/404008/404004)|offline}"];
  "0x00158d0003410305" -> "0x00124b0012013c7c" [style="dashed", label="0"]
  "0x00158d0002fdfe84" [style="rounded", label="{0x00158d0002fdfe84|Router|Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, opal white (404006/404008/404004)|offline}"];
  "0x00158d0002fdfe84" -> "0x00124b0012013c7c" [label="82"]
  "0x00158d000340f465" [style="rounded", label="{0x00158d000340f465|Router|Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, color, opal white (404000/404005/404012)|offline}"];
  "0x00158d000340f465" -> "0x00124b0012013c7c" [label="37"]
  "0x00158d000327ac95" [style="rounded", label="{0x00158d000327ac95|Router|Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, opal white (404006/404008/404004)|offline}"];
  "0x00158d000327ac95" -> "0x00124b0012013c7c" [label="1"]
  "0x00158d0002afdc05" [style="rounded, dashed", label="{0x00158d0002afdc05|EndDevice|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|online}"];
  "0x00158d0002ec7d64" [style="rounded, dashed", label="{0x00158d0002ec7d64|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d0002ec7d64" -> "0x00124b0012013c7c" [label="170"]
  "0x00158d0002b6e25f" [style="rounded, dashed", label="{0x00158d0002b6e25f|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d0002b6e25f" -> "0x00124b0012013c7c" [label="170"]
  "0x00158d00031d1446" [style="rounded", label="{0x00158d00031d1446|Router|Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, opal white (404006/404008/404004)|offline}"];
  "0x00158d00031d1446" -> "0x00124b0012013c7c" [label="1"]
  "0x00158d0002c5ce7f" [style="rounded, dashed", label="{0x00158d0002c5ce7f|EndDevice|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|online}"];
  "0x00158d0002c5ce7f" -> "0x00124b0012013c7c" [label="98"]
  "0x00158d0002b76583" [style="rounded, dashed", label="{0x00158d0002b76583|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d00027de282" [style="rounded, dashed", label="{0x00158d00027de282|EndDevice|Xiaomi Mi/Aqara smart home cube (MFKZQ01LM)|online}"];
  "0x00158d00027de282" -> "0x00124b0012013c7c" [label="20"]
  "0x00158d00027de329" [style="rounded, dashed", label="{0x00158d00027de329|EndDevice|Xiaomi Mi/Aqara smart home cube (MFKZQ01LM)|online}"];
  "0x00158d00027de329" -> "0x00124b0012013c7c" [label="126"]
  "0x00158d0002d6f4c0" [style="rounded", label="{0x00158d0002d6f4c0|Router|Xiaomi Aqara smart LED bulb (ZNLDP12LM)|online}"];
  "0x00158d0002d6f4c0" -> "0x00124b0012013c7c" [label="49"]
  "0x00158d0002c58943" [style="rounded, dashed", label="{0x00158d0002c58943|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
}`

hansblafoo, 2019-03-16:
I'm also having trouble connecting Aqara devices over a router to the network. I tried to re-pair the Aqara Waterleak Sensor with a Innr SP120 as a router, but the Sensor seems to always connect to the coordinator. Is there a way to reduce the range of the coordinator so that I increase the likelihood of getting connected to a router?

kariofillis, 2019-04-30:
is there a way to distinguish if the messages are routed through a router, or if the connection is directly to the coordinator? i recently added a router and would like to know if i have to repair the remote devices to make them talk to the new router.

closed 2019-07-06 by stale[bot]

Krystm, 2021-12-14:
Im noticing issues like this as well, however it may be more related to the routers not doing joins and only the coordinator which is another on going issue.