Koenkk/zigbee2mqtt #1185: OSRAM Smart+ motion recent new problem

dreimer1986, 2019-03-04:
Hi, just wanted to let you know that an update to 2a3670d caused a new line in my list of problems with the motion sensor from Osram to show up:

```
  zigbee2mqtt:warn 3/4/2019, 9:03:06 AM Failed to configure Motion (0x000d6f000e059aae) ('Error: request unsuccess: 112') (attempt #3)
  zigbee2mqtt:warn 3/4/2019, 9:03:06 AM This can be ignored if the device is working properly
  zigbee2mqtt:warn 3/4/2019, 9:03:08 AM No converter available for 'AC01353010G' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":13,"zoneStatus":32,"iasCieAddr":"0x00124b0014d9d7b5","zoneId":255}}'
  zigbee2mqtt:warn 3/4/2019, 9:03:08 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

First two lines can be ignored as motion did not work for a while now and I just had no interest in repairing it yet because it's a pain to do that on this device. The missing converter ist what I wonder about.

Koenkk, 2019-03-04:
Does it report occupancy?

dreimer1986, 2019-03-04:
Right now it does not. I tried to re-pair it and this is the result then:

```
  zigbee2mqtt:error 3/4/2019, 8:13:16 PM Cannot get the Node Descriptor of the Device: 0x000d6f000e059aae (Error: Timed out after 10000 ms)
  zigbee2mqtt:warn 3/4/2019, 8:13:19 PM Message without device!
  zigbee2mqtt:error 3/4/2019, 8:13:36 PM Cannot get the Node Descriptor of the Device: 0x000d6f000e059aae (Error: Timed out after 10000 ms)
  zigbee2mqtt:warn 3/4/2019, 8:14:02 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 3/4/2019, 8:14:02 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 3/4/2019, 8:14:05 PM No converter available for 'AC01353010G' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"zclVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"CentraLite","modelId":"Motion Sensor-A"}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:05 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 8:14:06 PM No converter available for 'AC01353010G' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"dateCode":"����������������","powerSource":3,"locationDesc":"N/A","physicalEnv":0}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:06 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 8:14:09 PM No converter available for 'AC01353010G' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryVoltage":27,"batterySize":2,"batteryQuantity":1,"batteryVoltMinThres":21,"batteryVoltThres1":22}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:09 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 8:14:10 PM No converter available for 'AC01353010G' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryVoltThres2":23,"batteryVoltThres3":24,"batteryAlarmState":0}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:10 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 8:14:10 PM No converter available for 'AC01353010G' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryVoltThres2":23,"batteryVoltThres3":24,"batteryAlarmState":0}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:10 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 8:14:11 PM No converter available for 'AC01353010G' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"zclVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"CentraLite","modelId":"Motion Sensor-A"}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:11 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 8:14:12 PM No converter available for 'AC01353010G' with cid 'genIdentify', type 'attReport' and data '{"cid":"genIdentify","data":{"identifyTime":0}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:12 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 8:14:16 PM No converter available for 'AC01353010G' with cid 'genPollCtrl', type 'attReport' and data '{"cid":"genPollCtrl","data":{"checkinInterval":14400,"longPollInterval":24,"shortPollInterval":4,"fastPollTimeout":40,"checkinIntervalMin":240}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:16 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/4/2019, 8:14:17 PM No converter available for 'AC01353010G' with cid 'genPollCtrl', type 'attReport' and data '{"cid":"genPollCtrl","data":{"longPollIntervalMin":23,"fastPollTimeoutMax":120}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:17 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 3/4/2019, 8:14:19 PM MQTT publish: topic 'zigbee2mqtt/Motion', payload '{"temperature":19.47,"linkquality":49,"occupancy":false,"tamper":true,"battery_low":false}'
  zigbee2mqtt:warn 3/4/2019, 8:14:21 PM No converter available for 'AC01353010G' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":13,"zoneStatus":38,"iasCieAddr":"0xffffffffffffffff","zoneId":255}}'
  zigbee2mqtt:warn 3/4/2019, 8:14:21 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Koenkk, 2019-03-05:
You can ignore the no converter available messages (it's something I have to fix).

The real error here is: `  zigbee2mqtt:error 3/4/2019, 8:13:36 PM Cannot get the Node Descriptor of the Device: 0x000d6f000e059aae (Error: Timed out after 10000 ms)`.

What you need to do is
- stop zigbee2mqtt
- remove all entries of  `0x000d6f000e059aae` from database.db
- start zigbee2mqtt
- re-pair

dreimer1986, 2019-03-05:
Did so and the result is pasted at the end. Accoring to some other reports I follow all the how_to_support_new_devices is not only my problem. Ah and btw, thx for the great hint with the .db I never would have thought to open it up and wipe stuff off it because I thought It's binary like a database... Result. Still a pain to pair this thing, but it's doable aaaaand works fine again! ^^ Now only the how_to_support_new_devices spamming is left ^^

```
  zigbee2mqtt:error 3/5/2019, 8:35:57 PM Cannot get the Node Descriptor of the Device: 0x000d6f000e059aae (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 3/5/2019, 8:36:17 PM Cannot get the Node Descriptor of the Device: 0x000d6f000e059aae (Error: Timed out after 10000 ms)
  zigbee2mqtt:info 3/5/2019, 8:36:27 PM New device 'undefined' with address 0x000d6f000e059aae connected!
  zigbee2mqtt:info 3/5/2019, 8:36:27 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f000e059aae","meta":{}}'
  zigbee2mqtt:warn 3/5/2019, 8:36:27 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 3/5/2019, 8:36:27 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 3/5/2019, 8:36:30 PM No converter available for 'AC01353010G' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"zclVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"CentraLite","modelId":"Motion Sensor-A"}}'
  zigbee2mqtt:warn 3/5/2019, 8:36:30 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 3/5/2019, 8:36:30 PM MQTT publish: topic 'homeassistant/binary_sensor/0x000d6f000e059aae/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x000d6f000e059aae","json_attributes_topic":"zigbee2mqtt/0x000d6f000e059aae","name":"0x000d6f000e059aae_occupancy","unique_id":"0x000d6f000e059aae_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f000e059aae","name":"0x000d6f000e059aae","sw_version":"Zigbee2mqtt 1.1.1","model":"SMART+ Motion Sensor (AC01353010G)","manufacturer":"OSRAM"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/5/2019, 8:36:30 PM MQTT publish: topic 'homeassistant/sensor/0x000d6f000e059aae/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x000d6f000e059aae","json_attributes_topic":"zigbee2mqtt/0x000d6f000e059aae","name":"0x000d6f000e059aae_temperature","unique_id":"0x000d6f000e059aae_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f000e059aae","name":"0x000d6f000e059aae","sw_version":"Zigbee2mqtt 1.1.1","model":"SMART+ Motion Sensor (AC01353010G)","manufacturer":"OSRAM"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/5/2019, 8:36:30 PM MQTT publish: topic 'homeassistant/sensor/0x000d6f000e059aae/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f000e059aae","json_attributes_topic":"zigbee2mqtt/0x000d6f000e059aae","name":"0x000d6f000e059aae_linkquality","unique_id":"0x000d6f000e059aae_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f000e059aae","name":"0x000d6f000e059aae","sw_version":"Zigbee2mqtt 1.1.1","model":"SMART+ Motion Sensor (AC01353010G)","manufacturer":"OSRAM"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:warn 3/5/2019, 8:36:31 PM No converter available for 'AC01353010G' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"dateCode":"����������������","powerSource":3,"locationDesc":"N/A","physicalEnv":0}}'
  zigbee2mqtt:warn 3/5/2019, 8:36:31 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/5/2019, 8:36:34 PM No converter available for 'AC01353010G' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":13,"zoneStatus":32,"iasCieAddr":"0x00124b0014d9d7b5","zoneId":255}}'
  zigbee2mqtt:warn 3/5/2019, 8:36:34 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/5/2019, 8:36:34 PM No converter available for 'AC01353010G' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryVoltage":27,"batterySize":2,"batteryQuantity":1,"batteryVoltMinThres":21,"batteryVoltThres1":22}}'
  zigbee2mqtt:warn 3/5/2019, 8:36:34 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/5/2019, 8:36:35 PM No converter available for 'AC01353010G' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryVoltThres2":23,"batteryVoltThres3":24,"batteryAlarmState":0}}'
  zigbee2mqtt:warn 3/5/2019, 8:36:35 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/5/2019, 8:36:37 PM No converter available for 'AC01353010G' with cid 'genIdentify', type 'attReport' and data '{"cid":"genIdentify","data":{"identifyTime":0}}'
  zigbee2mqtt:warn 3/5/2019, 8:36:37 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
[... cut ...] ":1}}'
  zigbee2mqtt:warn 3/5/2019, 8:37:11 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/5/2019, 8:37:12 PM No converter available for 'AC01353010G' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneType":13,"zoneStatus":36,"iasCieAddr":"0x00124b0014d9d7b5","zoneId":23}}'
  zigbee2mqtt:warn 3/5/2019, 8:37:12 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
´´´

Koenkk, 2019-03-06:
@dreimer1986 that should be fixed in the latest dev branch, can you confirm?

dreimer1986, 2019-03-06:
Not at home right now, thus I only can tell what happens on a reboot. Such reboot caused the first log snippet I posted all the time before the fixes in HEAD rev. Now it's all silent again. So I guess things are all fine now. And thx for helping with the occupancy fix ^^

thissiac, 2019-05-09:
New problems is exactly the right term.

Aafter i couldn't pair the old motion sensor anymore, i got myself a new one. Now the situation is different again.

This was the first message that came after activation:

Mai 09 08:01:21 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:21 MQTT publish: topic 'z2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Mai 09 08:01:24 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:24 Connecting with device...
Mai 09 08:01:24 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:24 MQTT publish: topic 'z2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Mai 09 08:01:25 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:25 Connecting with device...
Mai 09 08:01:25 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:25 MQTT publish: topic 'z2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Mai 09 08:01:25 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:25 Device incoming...
Mai 09 08:01:25 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:25 MQTT publish: topic 'z2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
Mai 09 08:01:25 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:25 New device with address 0x000d6f000f83f7f8 connected!
Mai 09 08:01:25 openHABianPi npm[24526]:   zigbee2mqtt:info 2019-5-9 08:01:25 MQTT publish: topic 'z2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f000f83f7f8"}'
Mai 09 08:01:26 openHABianPi npm[24526]:   zigbee2mqtt:error 2019-5-9 08:01:26 Failed to configure 0x000d6f000f83f7f8 (0x000d6f000f83f7f8) ('Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.')

After stopping z2mqtt and restarting it, the problem looks like this:

2019-5-9 09:35:14 MQTT publish: topic 'z2mqtt/0x000d6f000f83f7f8', payload '{"occupancy":false,"tamper":false,"battery_low":false}'
zigbee2mqtt:error 2019-5-9 09:35:27 Failed to configure 0x000d6f000f83f7f8 (0x000d6f000f83f7f8) ('unsupAttribute')
2019-5-9 09:40:53 MQTT publish: topic 'z2mqtt/0x000d6f000f83f7f8', payload '{"occupancy":false,"tamper":false,"battery_low":false,"temperature":21.73,"linkquality":21}'

From now on I get regular entries in the log file, but except for the values for temperature nothing changes. And even the temperatures change like in slow motion.

After stopping z2mqtt and restarting 2. time, the problem looks like this:

2019-5-9 11:28:23 Failed to configure 0x000d6f000f83f7f8 (0x000d6f000f83f7f8) ('Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.')

No further entries visible in the log.

Every time I start with "npm start" I see an entry which looks fine.

2019-5-9 11:44:10 MQTT publish: topic 'z2mqtt/0x000d6f000f83f7f8', payload '{"occupancy":false,"tamper":false,"battery_low":false,"temperature":22.97,"linkquality":15}'

Immidently the next entry

2019-5-9 11:44:11 Failed to configure 0x000d6f000f83f7f8 (0x000d6f000f83f7f8) ('Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.')

What next?

Maybe the Smart+ Motion Sensor is not the right product for zigbee2mqtt?

dreimer1986, 2019-06-10:
Sorry forgot to comfirm... Comfirmed ^^

closed 2019-06-10 by dreimer1986