Koenkk/zigbee2mqtt #1186: Add support for Orvibo RL804QZB (3 relay device)

zen2, 2019-03-04:
I'm trying to use a multi-relay device Orvibo RL804QZB.

I have paired it and actually I got:

Log:

```
2019-3-4 10:41:33 - info: Connecting with device...
2019-3-4 10:41:33 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
2019-3-4 10:41:33 - info: Device incoming...
2019-3-4 10:41:33 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2019-3-4 10:41:33 - info: New device with address 0x00124b001b5f5dba connected!
2019-3-4 10:41:33 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00124b001b5f5dba"}'
2019-3-4 10:41:33 - warn: Device with modelID '82c167c95ed746cdbd21d6817f72c593' is not supported.
2019-3-4 10:41:33 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-3-4 10:41:33 - warn: Device with modelID '82c167c95ed746cdbd21d6817f72c593' is not supported.
2019-3-4 10:41:33 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

Database extract:

```
{"id":51,"type":"Router","ieeeAddr":"0x00124b001b5f5dba","nwkAddr":57133,"manufId":0,"manufName":"ORVIBO","powerSource":"Mains (single phase)","modelId":"82c167c95ed746cdbd21d6817f72c593","epList":[1,2,3],"status":"online","joinTime":1551692491,"endpoints":{"1":{"profId":260,"epId":1,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}}},"_id":"zrjs9LQuQZZ59jfg"}
{"id":51,"type":"Router","ieeeAddr":"0x00124b001b5f5dba","nwkAddr":57133,"manufId":0,"manufName":"ORVIBO","powerSource":"Mains (single phase)","modelId":"82c167c95ed746cdbd21d6817f72c593","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}}},"_id":"zrjs9LQuQZZ59jfg"}
{"id":51,"type":"Router","ieeeAddr":"0x00124b001b5f5dba","nwkAddr":57133,"manufId":0,"manufName":"ORVIBO","powerSource":"Mains (single phase)","modelId":"82c167c95ed746cdbd21d6817f72c593","epList":[1,2,3],"status":"online","joinTime":1551693305,"endpoints":{"1":{"profId":260,"epId":1,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}}},"_id":"zrjs9LQuQZZ59jfg"}
{"id":51,"type":"Router","ieeeAddr":"0x00124b001b5f5dba","nwkAddr":57133,"manufId":0,"manufName":"ORVIBO","powerSource":"Mains (single phase)","modelId":"82c167c95ed746cdbd21d6817f72c593","epList":[1,2,3],"status":"online","joinTime":1551693305,"endpoints":{"1":{"profId":260,"epId":1,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":11,"inClusterList":[0,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}}}}},"_id":"zrjs9LQuQZZ59jfg"}
```

I'm not sure what I need to use to add the device and converter in order to use it.
I'm yet glad that the pairing is working :)
This device can be used for many things and I'm sure many people will be use it if we got a support for it.

Additional informations can be found here:
https://community.smartthings.com/t/orvibo-zigbee-relay-rl804qzb/49463/17

Koenkk, 2019-03-04:
Add to `devices.js`:

```js
{
        zigbeeModel: ['82c167c95ed746cdbd21d6817f72c593'],
        model: 'RL804QZB',
        vendor: 'Orvibo',
        description: 'Multi-functional 3 gang relay',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [tz.on_off],
        ep: (device) => {
            return {'1': 1, '2': 2, '3': 3};
        },
    },
```

Then send to `zigbee2mqtt/0x00124b001b5f5dba/1/set`, `zigbee2mqtt/0x00124b001b5f5dba/2/set`, `zigbee2mqtt/0x00124b001b5f5dba/3/set` payload `ON` or `OFF`

closed 2019-05-10 by stale[bot]

aivs, 2020-02-20:
ok, now we can control each relays, but how to get reports if button clicked ?