Koenkk/zigbee2mqtt #1188: Ikea control outlet E1603 not working

Wamor, 2019-03-04:
Hello,

Please delete this post if my post is not bug-related.
I have installed Domoticz on a Raspberry Pi and have an Ikea Led Bulb and a Ikea Control Outlet (E1603) connected.
The light-bulb I can switch on/off without any problems but the Control Outlet is not working.
I'm a little but puzzled and don't know what to do. Are there any suggestions how I can debug this issue? For example when switching on/off the light-bulb I see the following log-message:
Received heartbeat message from device "MyHardware - Light". 
The control outlet sends no reply at all.

Regards,

Wamor

Koenkk, 2019-03-05:
Can you post your zigbee2mqtt debug log? (`log_level` from http://www.zigbee2mqtt.io/configuration/configuration.html)

Wamor, 2019-03-06:
Hello Koenkk,

Thank you for your reply.
After I posted my message I did also some additional testing/experimenting. I have deleted the devices in the devices-section of the configuration.yaml (located in /opt/zigbee2mqtt/data) in the assumption that the devices could be detected over new.
This was not happening because the devices were still showing up in the devices-tab of Domoticz.
I then connected the Control outlet to my Ikea hub to check if it was working ok. This was the case.
After that I restored the devices-section in the configuration.yaml and reset the Control outlet to make it redetect by CC-dongle. I also pressed both buttons in the CC-dongle in order to reset it? Now I have the E1603 outlet working but now my Lightbulb LED1623G12 is not working anymore...
They are still both detected by Domoticz. Is there a way to completely erase all the detected devices? I can then reset the devices and start to detected them over new with domoticz.

Regards,

Wamor

For you reference I have attached my latest log-file from domoticz.

[Log2019-03-06_14-17.txt](https://github.com/Koenkk/zigbee2mqtt/files/2936449/Log2019-03-06_14-17.txt)

Wamor, 2019-03-06:
Everything is working now. After pairing the light-bulb with an Ikea hub and resetting it by switching on for 6 times the light is detected by Domoticz with CC2531-dongle.
Dont't know if this sequence is always working but for now my problem is solved. It looks like pairing is not always reliable.

closed 2019-03-06 by Wamor