Koenkk/zigbee2mqtt #1189: npm start exits with error

RobertTheLink, 2019-03-04:
Following the instructions from https://www.zigbee2mqtt.io/
I prepared the CC2531 on a Windows machine and running z2m on a Raspi.
I get always and error when doing "npm start" in /opt/zigbee2mqtt
logs in  [pastebin] (https://pastebin.com/rmC6gM7H)
I also tried hassbian-config scripts as mentioned in related issue reports but that did not help here.
Any idea?

Koenkk, 2019-03-05:
What is the output of `ls /opt/zigbee2mqtt`?

RobertTheLink, 2019-03-05:
`ls /opt/zigbee2mqtt`
yields
`data
docker
docs
hassbian-scripts_0.11.1_all.deb
images
index.js
lib
LICENSE
node_modules
npm-shrinkwrap.json
package.json
README.md
scripts
test
update.sh`

Koenkk, 2019-03-06:
Could you provide the whole output when running `ls /opt/zigbee2mqtt && rm -rf node_modules && npm install && npm start`

Koenkk, 2019-03-06:
You probably have the same errors as in #1168, solution: https://github.com/Koenkk/zigbee2mqtt/issues/1168#issuecomment-470105468

closed 2019-05-05 by Koenkk