Koenkk/zigbee2mqtt #1190: Another "Error while starting zigbee-shepherd, attemping to fix"

timdonovanuk, 2019-03-04:
Running a _very_ standard setup of:

- Pi 3 Model B+
- Latest Raspbian 
- Latest cc2531 CC2531ZNP-Prod.hex firmware 
- Latest zigbee2mqtt docker image (koenkk/zigbee2mqtt:arm32v6) using the following docker-compose entry:

```
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt:arm32v6
    container_name: zigbee2mqtt
    restart: always
    network_mode: host
    volumes:
      - /opt/zigbee2mqtt:/app/data
    devices:
      - "/dev/ttyACM0"
```

- I have tried "Pressing the reset button on the stick.."...and starting the docker container again
- My config has log_level debug
- There are no other containers also accessing serial ports
- I have rebooted the Pi
- My CC2531 is in the right place:

```
ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Mar  4 21:49 /dev/ttyACM0
```
Log is short:

```
2019-3-4 22:27:54 - info: Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
2019-3-4 22:27:54 - info: Starting zigbee-shepherd
2019-3-4 22:27:57 - info: Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2019-3-4 22:28:57 - info: Starting zigbee-shepherd
2019-3-4 22:29:00 - error: Error while starting zigbee-shepherd!
2019-3-4 22:29:00 - error: Press the reset button on the stick (the one closest to the USB) and start again
2019-3-4 22:29:00 - error: Failed to start Error: request timeout
    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)

```

timdonovanuk, 2019-03-04:
Am trying a re-flash of the stick, so will report back tomorrow morning.

mkroman, 2019-03-05:
Had the same issue on the CC2531 with the latest coordinator from the `dev` branch. It suddenly stopped working and would no longer start.

I had to switch back to the one from the `master` branch, i.e. `CC2531ZNP-Prod_20190109`

timdonovanuk, 2019-03-05:
Re-flashed and all is well! Closing.

closed 2019-03-05 by timdonovanuk