Koenkk/zigbee2mqtt #1191: Feature request: ICTC-G-1(round Ikea dimmer) Add Action to the MQTT message

WagnerBeccard, 2019-03-05:
For better integrating into HomeAssistant i would realy appriciate a `action` in the payload. 
like ` rotate_left ` ,` rotate_right ` ,` quick_rotate_left ` ,` quick_rotate_right `  that we can better work with templates and set lights more dynamic.

JLFN, 2019-03-05:
I use like this and it works perfect.
```yaml
automation:

- alias: Ikea dimmer
  initial_state: 'off'
  trigger:
    platform: mqtt
    topic: 'zigbee2mqtt_4/0x000b57fffe2f8d88'
  action:
    - service: light.turn_on
      data:
        entity_id: light.bed_1
    - service: light.turn_on
      data_template:
        entity_id: light.bed_1
        brightness: '{{ states.sensor.ikea_dimmer.state }}'
    - condition: template
      value_template: "{{ states.sensor.ikea_dimmer.state == '0' }}"
    - service: light.turn_off
      data:
        entity_id: light.bed_1
```

WagnerBeccard, 2019-03-05:
Therefore the request - I would change the brightness of an room, But with diffrent levels on each light.
Would open the possibility of using it as "button". 

like in 
https://github.com/Koenkk/zigbee2mqtt/issues/542
 `movemode":0` for left and `movemode":1` for right.

just found
https://github.com/Koenkk/zigbee2mqtt/issues/690
is this tecnical is an duplicate?

Koenkk, 2019-03-06:
@WagnerBeccard would you manage to update the code here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L84 (should be the only thing that requires changing).

Koenkk, 2019-03-06:
Implemented in the dev branch.

Possible actions are: 
- `rotate_left`
- `rotate_right`
- `rotate_stop` (published after `rotate_left` or `rotate_right`)
- `rotate_left_quick`
- `rotate_right_quick`

closed 2019-03-06 by Koenkk