Koenkk/zigbee2mqtt #1192: zigbee2mqtt config not paying attention to my friendly names

roygbiv856, 2019-03-05:
Back when I was on hassio on a pi, I could change the friendly names in the z2m/config and see the new names in the logs. I recently moved to a NUC, still running hassio like on the pi, but my new friendly names are not reflected in the logs. After a restart of hassio, my friendly names completely reset to the default friendly names of a long string of numbers. Where exactly should I go to change the friendly names and have them reflected in the logs?

Koenkk, 2019-03-05:
Can you post your zigbee2mqtt logs?

roygbiv856, 2019-03-06:
@Koenkk Here you are, buddy

https://pastebin.com/SWd0Ppp1

Koenkk, 2019-03-06:
Can you share you `configuration.yaml` (don't forget to mask sensitive information like passwords).

roygbiv856, 2019-03-06:
@Koenkk The friendly names often get reset back to their default long number names, so here I only have 1 changed:

https://pastebin.com/g8nstyGE

The friendly name never appears in the logs

Koenkk, 2019-03-06:
Could you provide your startup log after changing 1 friendly name? I guess the real issue here is that the friendly names get reset automatically, could you find out how to reproduce this?

Andreln, 2019-03-12:
I just paired a IKEA outlet and I get the same error. Homeassistant doesn't register the device with friendly name instead it takes the id and appends _switch (switch.0x000d6ffffef5119a_switch). 

After pairing I added a friendly name and restarted both zigbee2mqtt and homeassistant.

Koenkk, 2019-03-13:
@Andreln changing this after the device got discovered by HA is not possible (limitation of HA). http://www.zigbee2mqtt.io/integration/home_assistant.html#im-confused-about-the-different-device-ids-names-and-friendly-names

closed 2019-05-12 by Koenkk