Koenkk/zigbee2mqtt #1194: What happens to messages when an end device can't be reached for several hours?

will-emmerson, 2019-03-05:
I am interested in the project but just wondered what happens when you send a message to a device which is unreachable for an extended period of time.

I realise that Zigbee itself has some retry mechanisms built in but these seem to be short term and it appears it gives up after a few minutes, but I could be wrong about this. The documentation says that the coordinator and routers store messages, for end devices, is this reliable?

Does the mqtt client only give an ACK to the mqtt server when the end device has successfully received the message? (If mqtt QOS is used at all)

I realise that mqtt has a queue but this doesn't help as the bridge device itself *should* have a reliable connection.

Koenkk, 2019-03-05:
It retries a few times, in a time frame of a few seconds (AFIAK). 

Normally you don't send messages to end-devices (because they are sleeping), what's your use-case/device here?

will-emmerson, 2019-03-08:
The use case is a custom device which can lose power or connectivity on a regular basis, for several hours.

Koenkk, 2019-03-09:
I think there is some retry mechanism, but I don't know for how long the messages are kept. The best way to find out is to try.

closed 2019-03-14 by will-emmerson