Koenkk/zigbee2mqtt #1195: Feature Request for all motion sensors, "No motion since missing"

DuckY1987, 2019-03-05:
Dear all,

i wanted to transfer all my zigbee devices to zigbee2mqtt, first thank you all for that great mqtt integration.

Now i hit the point, that i miss a vital feature what home-assistant's xiaomi integration had before, what is missing for the occupancy sensors.

No motion since

I used it for ALL light automation, there is no similar value in this integration.

Would it be possible to add it?

The workaround would be timestamps, but that overcomplicates a lot things.

Koenkk, 2019-03-05:
What time format do you prefer for this? Seconds since epoch? [ISO 8601](https://nl.wikipedia.org/wiki/ISO_8601)?

DuckY1987, 2019-03-05:
Xiaomi uses seconds since last movement triggered, max is 1800 that allows a maximum of 30 minutes, what is a lot to deal with, to reduce noise on the mosquitto, i would do it like xiaomi is doing the updates in steps of 120, 180,300,600,1200,1800 (2,3,5,10,20,30 minutes) i would suggest to add 7,15,25 minutes (420,900,1500)

Perfect for every motion based automation.

Thats how it looks like:
No motion since: 1800

I use this wait_template in hass:

```yaml
  - wait_template: >-
      {{
         state_attr('binary_sensor.motion_sensor_xxx','No motion since')|int >= (states('input_number.light_global_timer')|int * 60)
      }}
```

Koenkk, 2019-03-06:
So to verify, this message should just be send after a certain time of no motion? E.g. after 1800 seconds of no occupancy zigbee2mqtt should send `{"no_occupancy_since": 1800}`

DuckY1987, 2019-03-07:
Precisely! Each step one message :-)

{"no_occupancy_since": 120}
{"no_occupancy_since": 180}
and so on
till
{"no_occupancy_since": 1800}

 if you decide to make it more dynamic and configurable, you could use a

{"update_interval": 30} (each 30 seconds a message)
and
{"max_update_interval": 1800} (no more messages after this)

additionally.

When motion true, reset to 0 as long as movement is detected:

{"no_occupancy_since": 0}

and both parameters you might be able to change with a mqtt payload.

Afaik i saw that you implemented kind of a time_out already which might have some simlar function.

Then its even better than what xiaomi implemented ;-).

Koenkk, 2019-03-15:
Implemented in the dev branch (http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). Documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/configuration/device_specific_configuration.md#rtcgq01lm-rtcgq11lm

Bastiencc, 2019-03-16:
This new state is very interesting. I'm trying to use it in HA : 

 ```
- id: capteur_couloir_not_empty
  alias: Capteur couloir not empty
  initial_state: 'on'
  trigger:
    - platform: mqtt
      topic: 'zigbee2mqtt/couloir_capteur'
  condition:
    - condition: template
      value_template: '{{ true == trigger.payload_json.occupancy }}'
  action:
    - service: light.turn_on
      entity_id: light.couloir_plafond
    - wait_template: "{{ trigger.payload_json.no_occupancy_since|int >= states.input_number.light_global_timer.state|int }}"
    - service: light.turn_off
      entity_id: light.couloir_plafond
 ```

But I get this error , because `no_occupancy_since` is not alway sent
 ```
2019-03-16 11:13:45 INFO (MainThread) [homeassistant.components.automation] Executing Capteur couloir not empty
2019-03-16 11:13:45 INFO (MainThread) [homeassistant.helpers.script] Script Capteur couloir not empty: Running script
2019-03-16 11:13:45 INFO (MainThread) [homeassistant.helpers.script] Script Capteur couloir not empty: Executing step call service
2019-03-16 11:13:45 INFO (MainThread) [homeassistant.helpers.script] Script Capteur couloir not empty: Executing step wait template
2019-03-16 11:13:45 ERROR (MainThread) [homeassistant.helpers.condition] **Error during template condition: UndefinedError: 'dict object' has no attribute 'no_occupancy_since'**
 ```

zigbee2mqtt config : 
 ```
  '0x00158d0003664f59':
    friendly_name: couloir_capteur
    retain: false
    occupancy_timeout: 20
    no_occupancy_since: [10, 20, 30, 40 , 50 , 60, 70]
 ```

zigbee2mqtt log: 
 ```
Mar 16 11:23:26 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:26 AM MQTT publish: topic 'zigbee2mqtt/couloir_capteur', payload '{"illuminance":306,"linkquality":76,"occupancy":false,"battery":100,"voltage":3005}'
Mar 16 11:23:26 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:26 AM MQTT publish: topic 'zigbee2mqtt/couloir_capteur', payload '{"illuminance":306,"linkquality":73,"occupancy":true,"battery":100,"voltage":3005}'
Mar 16 11:23:27 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:27 AM Zigbee publish to device '0x00158d0002d39c01', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 16 11:23:27 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:27 AM MQTT publish: topic 'zigbee2mqtt/couloir_plafond', payload '{"state":"ON","linkquality":15,"color":{"x":0.442,"y":0.41},"brightness":255,"color_temp":338}'
Mar 16 11:23:27 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:27 AM Zigbee publish to device '0x00158d0002d39c01', lightingColorCtrl - moveToColorTemp - {"colortemp":338,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 16 11:23:27 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:27 AM MQTT publish: topic 'zigbee2mqtt/couloir_plafond', payload '{"state":"ON","linkquality":15,"color":{"x":0.442,"y":0.41},"brightness":40,"color_temp":338}'
Mar 16 11:23:27 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:27 AM MQTT publish: topic 'zigbee2mqtt/couloir_plafond', payload '{"state":"ON","linkquality":15,"color":{"x":0.442,"y":0.41},"brightness":40,"color_temp":338}'
Mar 16 11:23:28 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:28 AM Zigbee publish to device '0x00158d0002d39c01', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 16 11:23:28 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:28 AM MQTT publish: topic 'zigbee2mqtt/couloir_plafond', payload '{"state":"ON","linkquality":15,"color":{"x":0.442,"y":0.41},"brightness":255,"color_temp":338}'
Mar 16 11:23:28 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:28 AM MQTT publish: topic 'zigbee2mqtt/couloir_plafond', payload '{"state":"ON","linkquality":15,"color":{"x":0.442,"y":0.41},"brightness":255,"color_temp":338}'
Mar 16 11:23:28 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:28 AM Zigbee publish to device '0x00158d0002d39c01', lightingColorCtrl - read - [{"attrId":7}] - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 16 11:23:36 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:36 AM MQTT publish: topic 'zigbee2mqtt/couloir_capteur', payload '{"illuminance":306,"linkquality":73,"occupancy":true,"battery":100,"voltage":3005,"no_occupancy_since":10}'
Mar 16 11:23:41 raspberrypi npm[16535]:   zigbee2mqtt:error 3/16/2019, 11:23:41 AM Failed to ping 0x00158d0002d6f569
Mar 16 11:23:42 raspberrypi npm[16535]:   zigbee2mqtt:error 3/16/2019, 11:23:42 AM Failed to ping 0x00158d0002d39c01
Mar 16 11:23:42 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:42 AM Zigbee publish to device '0x00158d0002d39c01', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 16 11:23:42 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:42 AM MQTT publish: topic 'zigbee2mqtt/couloir_plafond', payload '{"state":"OFF","linkquality":15,"color":{"x":0.442,"y":0.41},"brightness":255,"color_temp":338}'
Mar 16 11:23:46 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:46 AM MQTT publish: topic 'zigbee2mqtt/couloir_capteur', payload '{"illuminance":306,"linkquality":73,"occupancy":false,"battery":100,"voltage":3005}'
Mar 16 11:23:46 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:46 AM MQTT publish: topic 'zigbee2mqtt/couloir_capteur', payload '{"illuminance":306,"linkquality":73,"occupancy":false,"battery":100,"voltage":3005,"no_occupancy_since":20}'
Mar 16 11:23:56 raspberrypi npm[16535]:   zigbee2mqtt:info 3/16/2019, 11:23:56 AM MQTT publish: topic 'zigbee2mqtt/couloir_capteur', payload '{"illuminance":306,"linkquality":73,"occupancy":false,"battery":100,"voltage":3005,"no_occupancy_since":30}'
 ```

Bastiencc, 2019-03-16:
I try and it works great , thanks @Koenkk !
Juste a question, how to use `occupancy_timeout` to force this sensor to update more often, it seems to be hardcoded 1min (or hardware ?)

DuckY1987, 2019-03-16:
I'm totally excited @Koenkk thanks a lot for this implementation, i will try it out when i'm back home after this weekend.

@Bastiencc 
According to the documentation, koeenk implemented it according to my request, so you set the update frequency in the array like this:

no_occupancy_since: [10, 20, 40, 60]

So basically you receive an update after 10, 20, 40 and 60 seconds, and then each 60 seconds till the timeout is reached.

Koenkk, 2019-03-16:
@Bastiencc the sensor only send the value every minute (but note that right after pairing the sensor it's in 'test' mode it will send it's value every 5 seconds). It's possible to keep it test mode by applying a hardware mod: https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806

closed 2019-03-16 by Koenkk

DuckY1987, 2019-03-18:
@Koenkk 

no_occupancy_since: 

Disappears as it isn't included in every message.

When movement is detected (occupancy: true )

the attribute disappears.

it should have then no_occupancy_since: 0

Instead of deleting the attribute the value should get reseted to 0

 :-)

Koenkk, 2019-03-19:
Should be fixed with the latest dev branch.

stoinov, 2019-03-23:
So this introduced a new "bug" - I have not set this setting for my devices, but I am now receiving `no_occupancy_since: 0` along with every `occupancy: 1`.

rengaboy, 2019-12-14:
Hello. Thank you for awesome integration.
I have a problem with that option. I add it to my sensor "no_occupancy_since: 120", but it did not work. Here is error in log
`error 2019-12-14T16:55:29: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: options.no_occupancy_since.forEach is not a function
    at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:736:44)
    at converters.forEach (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:144:41)
    at Array.forEach (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:142:20)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:228:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:57:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:415:22))
`

rengaboy, 2019-12-30:
@Koenkk
I solved my previous problem by added [ ] to value. Sorry for my inattentiveness. But i run into one more problem. It is not working like @DuckY1987 said. In my automation, i check that "no_occupancy_since" is more than 1 hour, and then, when entrance door sensor activates, my automation should switch on entrance light, but nothing happens, because randomly, maybe on battery or linkstate updates, "no_occupancy_since" attribute disappears, and then reappears only after "occupancy: true" message. Is it possible to make the value increase every minute, fi i have "no_occupancy_since: [60]" in config, like @DuckY1987 asked?