Koenkk/zigbee2mqtt #1197: Feature request: "expire_after: 1" for switches / buttons

Erelen-Laiquendi, 2019-03-06:
With 1 second expiration, it is possible to write buttons click automation more simple.

Current:
```
    trigger:
      platform: mqtt
      topic: 'zigbee2mqtt/<FRIENDLY_NAME>'
    condition:
      condition: template
      value_template: '{{ "single" == trigger.payload_json.click }}'
```

With expiration, just:
```
    trigger:
      platform: state
      entity_id: sensor.kitchen_corner_switch
      to: 'single'
```
This way, I use only sensor, and don't need to go deeper - to MQTT.
And with expiration, we have correct meaning of "state": "single" (or other click) - it's just a moment (ok, 1 second :) but it current scheme, what is state meaning after 1 minute? Last clicked button? Last clicked - it's not a _current_ state.

One more real example from my kitchen. I want to toggle light by left button on "enterance switch" and by "corner switch".
Current automation:
```
- alias: Kitchen 3 Lamps
  trigger:
  - platform: mqtt
    topic: 'zigbee2mqtt/Kitchen Entrance Switch'
  - platform: mqtt
    topic: 'zigbee2mqtt/Kitchen Corner Switch'
  condition:
    condition: template
    value_template: '{{ trigger.payload_json.click == "left" or trigger.payload_json.click == "single" }}'
```
Very, very complex and strange. And more accurate code must be more complex: ((left on enterance switch) or (single on corner switch)).

With expiration, simple intuituve code:
```
    trigger:
    - platform: state
      entity_id: sensor.kitchen_entrance_switch
      to: 'left'
    - platform: state
      entity_id: sensor.kitchen_corner_switch
      to: 'single'
```

Koenkk, 2019-03-06:
This has already been implemented: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L133 and https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/integration/home_assistant.md#via-home-assistant-entity

Erelen-Laiquendi, 2019-03-06:
Thanks!!!
Waiting for release :)

closed 2019-03-06 by Koenkk