Koenkk/zigbee2mqtt #1198: Philips Iris pairing

treksn, 2019-03-06:
Hi. I am trying to connect my Philips Iris (Gen3, Remote with clickwheel, LLC010) with my hassio home server via a CC2531 and of course zigbee2mqtt.
But a pairing is not possible. Nothing happens. Is this device supported (I think the documentation says yes)?
With the zboss sniffer I can see the zigbee packages, when I turn on/off the lamp after pairing the remote with the lamp.
I don't have a hue bridge, I just use the Iris with the remote and CC2531. 
Can anybody give me some advice how to get my Iris working with zigbee2mqtt?

My log:
```
Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-03-06T02:14:18: PM2 log: Launching in no daemon mode
2019-03-06T02:14:18: PM2 log: App [npm:0] starting in -fork mode-
2019-03-06T02:14:18: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /zigbee2mqtt-1.1.1
> node index.js
  zigbee2mqtt:info 3/6/2019, 2:14:22 AM Logging to directory: '/share/zigbee2mqtt/log/2019-03-06.02-14-22'
  zigbee2mqtt:debug 3/6/2019, 2:14:22 AM Removing old log directory '/share/zigbee2mqtt/log/2019-03-02.09-30-27'
  zigbee2mqtt:debug 3/6/2019, 2:14:22 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 3/6/2019, 2:14:23 AM Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 3/6/2019, 2:14:23 AM Starting zigbee2mqtt version 1.1.1 (commit #unknown)
  zigbee2mqtt:info 3/6/2019, 2:14:23 AM Starting zigbee-shepherd
  zigbee2mqtt:info 3/6/2019, 2:14:27 AM zigbee-shepherd started
  zigbee2mqtt:info 3/6/2019, 2:14:27 AM Coordinator firmware version: '20190109'
  zigbee2mqtt:debug 3/6/2019, 2:14:27 AM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b001936aba3","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190109},"startTime":1551834867,"joinTimeLeft":0}
  zigbee2mqtt:info 3/6/2019, 2:14:27 AM Currently 0 devices are joined:
  zigbee2mqtt:warn 3/6/2019, 2:14:27 AM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 3/6/2019, 2:14:27 AM Allowing new devices to join.
  zigbee2mqtt:warn 3/6/2019, 2:14:27 AM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 3/6/2019, 2:14:27 AM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 3/6/2019, 2:14:27 AM Connecting to MQTT server at mqtt://192.168.188.23:1883
  zigbee2mqtt:info 3/6/2019, 2:14:28 AM zigbee-shepherd ready
  zigbee2mqtt:info 3/6/2019, 2:14:28 AM Connected to MQTT server
  zigbee2mqtt:info 3/6/2019, 2:14:28 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:debug 3/6/2019, 2:14:28 AM Mounted the cieApp (epId 11)](url)
```
... nothing happens ...

My conf:
```
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": true,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://192.168.188.23:1883",
  "serial_port": "/dev/ttyACM0",
  "devices": [],
  "network_key": [],
  "log_level": "debug",
  "include_device_information": true,
  "overwrite": true
}
```

My database.db
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001936aba3","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1551832446,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[0,1281,3],"outClusterList":[1280,1282,3],"clusters":{"genBasic":{"dir":{"value":1},"acls":{"zclVersion":"R","hwVersion":"R","manufacturerName":"R","modelId":"R","dateCode":"R","powerSource":"R","locationDesc":"RW","physicalEnv":"RW","deviceEnabled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"VebGSvGKRcDq2Jlo"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001936aba3","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1551834867,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"VebGSvGKRcDq2Jlo"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001936aba3","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1551834867,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[0,1281,3],"outClusterList":[1280,1282,3],"clusters":{"genBasic":{"dir":{"value":1},"acls":{"zclVersion":"R","hwVersion":"R","manufacturerName":"R","modelId":"R","dateCode":"R","powerSource":"R","locationDesc":"RW","physicalEnv":"RW","deviceEnabled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"VebGSvGKRcDq2Jlo"}
```

Sniffing seems to work?:
[image]

Koenkk, 2019-03-06:
When putting the remote into pairing mode, do you see the pairing packets in the sniffing log?

treksn, 2019-03-06:
These are the packages I see, when I pair the remote with my lamp:
[image]

Koenkk, 2019-03-07:
If I understand it correctly, you want both devices to be paired with zigbee2mqtt, right?

If this is the case, you don't need to pair the remote directly to the lamp, but to zigbee2mqtt. Pairing it to the lamp will create a separate zigbee network.

treksn, 2019-03-07:
Ah ok. But how do I pair my lamp with zigbee2mqtt? Thant's what I mainly want.

treksn, 2019-03-08:
Okay. It also didn't work to pair the remote with z2m (zigbee2mqtt). Nothing happens in the log.
To make clear what I want: I just want to pair the Living Colors Lamp with z2m. Ideally I can control my lamp via z2m and with the remote but that't not so important. 
In the documentation it sais: "To Pair hold Hold ON and Bottom Left Key (Favorite 1) in Front of the LED until the Light Blinks and turns Orange. If connection was succesfull the Light will turn Green." I tried this a lot of times, but my LED never had another color then green. The LED of the CC2531 is meant there, right? Can I do something to debug, what is happening? 
Oh, I am using a raspberry pi with hassio btw and the official hassio-addon from danielwelch. Could this be the problem? I don't figured out how you both (and the plugins) are connected to each other.

Koenkk, 2019-03-08:
I didn't write those instructions, but the LED of the CC2531 never does this, so the LED on the remote must be meant here. It's seems you are having troubles putting the remote into pairing mode.

treksn, 2019-03-08:
Ok. Then I think the colour of the lamp is meant, because the remote doesn't have any colored leds. I paired (and resetted) the remote with the lamp a thousand times, so I think that's not the case. When I pair the remote with the Iris, I press the ON-Button for a few seconds. Then the lamp and the remote are blinking and are paired. When I do this in front of the 2531 stick, nothing happens. I tried it with a resetted remote and with a remote which is already connected to the lamp.

Koenkk, 2019-03-08:
@pixeldoc2000 do you have this device? could you help?

treksn, 2019-03-09:
My issue is probably related to this one: #375 
Maybe @tb-killa can also help? 
Thank you so much, guys!

pixeldoc2000, 2019-03-10:
@Koenkk yes, i have a Philips LivingColours Iris (Gen 3) and Philips LivingColours Bloom (Gen 3).

@treksn did you follow the pairing instruction? https://www.zigbee2mqtt.io/getting_started/pairing_devices.html#philips-living-colors-iris-friends-of-hue
I belive you did, because you quoted it.

> To Pair hold Hold ON and Bottom Left Key (Favorite 1) in Front of the LED until the Light Blinks and turns Orange. If connection was succesfull the Light will turn Green.

To clarify the "color" in this context. The Philips LivingColours Device should change color depenting on it's state.

Here are some tips:
- You may need to try to reset the device a few time.
- Also the device need to be close to your zigbee coordinator, the same as with the remote
- It also depends on which channel your zigbee network is running.

**A word of warning:** Use on of the ZLL Zigbee Channels, else you may not able to reset your device with the Philips Gen 3 remote. Thats where i'm currently stuck.

pixeldoc2000, 2019-03-10:
**database.db** entry of my Philips LC Iris:
`{"id":5,"type":"Router","ieeeAddr":"0x0017880101228c34","nwkAddr":36650,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LLC010","epList":[11,242],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":512,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":3,"manufacturerName":"Philips","modelId":"LLC010","dateCode":"20140923","powerSource":1,"swBuildId":"5.23.1.13452"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":0,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":19,"remainingTime":0}},"genOta":{"dir":{"value":2},"attrs":{"upgradeServerId":"0xffffffffffffffff","fileOffset":4294967295,"currentFileVersion":1107309708,"currentZigbeeStackVersion":2,"downloadedFileVersion":4294967295,"imageUpgradeStatus":0,"manufacturerId":4107,"imageTypeId":65535,"minimumBlockReqDelay":0}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"currentHue":0,"currentSaturation":0,"remainingTime":0,"currentX":28508,"currentY":26542,"colorMode":1,"numPrimaries":3,"primary1X":0,"primary1Y":0,"primary1Intensity":134,"primary2X":0,"primary2Y":0,"primary2Intensity":206,"primary3X":0,"primary3Y":0,"primary3Intensity":37,"whitePointX":28508,"whitePointY":26542,"colorPointRX":0,"colorPointRY":0,"colorPointRIntensity":134,"colorPointGX":0,"colorPointGY":0,"colorPointGIntensity":206,"colorPointBX":0,"colorPointBY":0,"colorPointBIntensity":37,"enhancedCurrentHue":0,"enhancedColorMode":1,"colorLoopActive":0,"colorLoopDirection":0,"colorLoopTime":25,"colorLoopStartEnhancedHue":8960,"colorLoopStoredEnhancedHue":56545,"colorCapabilities":15}},"lightLink":{"dir":{"value":1},"attrs":{}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{"genGreenPowerProxy":{"dir":{"value":3},"attrs":{}}}}},"_id":"SGO2ibPJD1skLVLW"}
`

treksn, 2019-03-10:
Thank you for your answer. I will try the channel switch on the remote. But what is still confusing me a bit is, that you are saying: To pair the remote hold ON and bottom left key. But isn‘t it just holding the ON key? In the philips doc you cant find any functionality which is triggered by ON + bottom left.

treksn, 2019-03-10:
Okay nice!! Finally it worked! Thank you!! :-)

closed 2019-03-10 by Koenkk

streilu, 2019-04-14:
@trwksn how did you manage to connect the Living Colors to z2m?
I spent some hours trying to pair my 2 old Auras (remote with clickwheel), but didn't get it working.
Can you describe how you suceeded?

treksn, 2019-04-14:
I managed to pair my newer Iris (with click wheel) the older one with the round touch wheel didn‘t work. 
1. reset your remote
2. pair the remote with the lamp (pushing on button for a long time until the phillips lamp is blinking and turns orange/green)
3. enable pairing mode in z2m (permit_join)
4. UNpair your remote with the phillips lamp by pushing on and bottom left button (scene 1) on the remote until the lamp blinks and then turns red/green. Your remote should be somewhere inbetween your lamp and the sniffer device. You should then see something in your z2m logs. Set log level to debug in z2m, if you want to see all logs.

Hope that helps.

streilu, 2019-04-14:
Thanks, will try it later today.
Sounds logical, I think I made the mistake to always pair the remote to the lamp again. 😂

uweng, 2020-03-13:
There are two Philips LivingColors Iris:

1. Philips LivingColors Iris 7199960PH, that may be supported
2. Philips LivingColors Iris 7099960PH, this is NOT supported
I've got the second one with no success to link to CC2531 and zigbee2mqtt.
Hardware linking does not work and "touch link" as well. I got "Failed to factory reset device through Touchlink".
Has anybody the same problem?   Any hints?

pixeldoc2000, 2020-03-29:
@uweng 

I you have a Philips Hue Dimmer Switch you can try to (factory) reset your Philips LivingColors with it and than try to pair it with z2m.

jfu334, 2022-11-30:
After lots of trying I'm happy to add something to the issue with the 7099960PH. 

The 7099960PH **is** supported. It does talk zigbee, it can be joined through zigbee2mqtt and everything I've tested so far (on/off, brightness, color) does work fine.

Initially I had the same problems as stated here. I was able to join two LLC012 without any problems (use remote, hold pressed I and 1 until the light flashes etc.) but this did not work with the 7099960PH. 

What did work was to join the device through Touchlink in zigbee2mqtt. The device was recognized as a Touchlink device on the first try, but it required 2..3 resets until it was added to the network. 

Also the device is recognized as 7199960PH but the sticker on the wire clearly states 7099960PH. Maybe some functionality was added through the latest firmware update, mine reports 67.93.11 - so it seems to be important to  update the firmware using a hue bridge before trying zigbee2mqtt.

martindinicolino, 2023-01-10:
> They are not recognized through touchlink and at least the steps above don't work with them. However they can be paired to a hue bridge through touchlink and are perfectly working there.

I have LivingColors Aura, LLC007, 7099830PH, it also shows as „Living Colors Gen3 Bloom“ in the HueEssentials app, SW 6.0.8274. And the same you are describing for your lamps is true for mine as well.

It‘s too bad, because I was able to get everything else in Z2M and i would not like to keep the Hue Bridge around for this one light, when I got into Z2M mainly to consolidate all my zigbee devices into one network.

Sadly I was unable to find any further reading materials, documentations or other paths forward on this issue. I‘m very new to this platform, so i was wondering if there is a way to ask for a specific device to be added to Z2M. If not, I would be happy to sink some time into trying to get this light working with Z2M, but would need some instructions. Does anyone have some further suggestions?

jfu334, 2023-01-10:
Apparently the protocol used by the lamps is quite a mess. Some use SmartLink and some use LightLink and neither the product code nor the product ID does allow to distinguish those. I have four 7099960PH, three use SmartLink and one uses LightLink. 

However the sticker on the cable seems to give good advice. If the printed SN has 3 digits you have a SmartLink lamp, if it has 6 digits its a LightLink one. Only the latter can work with Zigbee2Mqtt as SmartLink is a completely different protocol with (to my knowledge) no public documentation which only uses the same physical layer as Zigbee Light Link. 

The Hue Bridge supports both of them, that's why the lights work with them. 

The one LightLink 7099960PH seems to have some compatibility issues as well. It does not work as a router and also doesn't connect to other routers which means it has to be close to the coordinator. Its connections also do not appear in the network map of Zigbee2Mqtt. 

The later Philips Hue Lights seem to work better, I have two 7299760PH which do work as router, even for non-ZLL devices, and connect to other routers without problems. Their connections also appear in the map. 

In the end I will keep the hue bridge for the Iris lights.

martialc, 2023-12-08:
> The later Philips Hue Lights seem to work better, I have two 7299760PH which do work as router, even for non-ZLL devices, and connect to other routers without problems. Their connections also appear in the map.

I'm curious, how did you manage to get those in zigbee2mqtt? I have one, but it just completely refuses to pair with the hue bridge method.

Sorry for necroposting, I tried reaching out to you directly, but couldn't.

jfu334, 2023-12-08:
I just added them the usual way. Use the remote to do a factory reset, then the lamp is recognized by zigbee2mqtt. Touchlink through zigbee2mqtt also works, but the lamps have to be really close. 

Mine do have a 6 digit S/N code on their sticker, so they are regular zigbee devices. If yours have 3 digits they are not zigbee but the older SmartLink, they won't work.