Koenkk/zigbee2mqtt #1200: Lamp state is send with wrong payload

sehraf, 2019-03-06:
Two of my three Hue GU10 lamps can't be controlled by HA anymore after new installation. Usually the payload is `{"state":"ON"}'` but for two is is now `'{"state_wohnzimmer.light.wall.left":"ON"}'` (see log below). This is not recognized by HA and so it thinks that the lamp is off.

I also see that the zigbee response from the bulb is not received. 
Any suggestions? I tried to repair them once without success.

---
```
zigbee2mqtt:debug 6.3.2019, 17:40:10 Received MQTT message on 'zigbee2mqtt/wohnzimmer.light.wall.middle/set' with data '{"state": "OFF"}'
zigbee2mqtt:info 6.3.2019, 17:40:10 Zigbee publish to device '0x00178801045450e8', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:info 6.3.2019, 17:40:10 MQTT publish: topic 'zigbee2mqtt/wohnzimmer.light.wall.middle', payload '{"state":"OFF"}'

zigbee2mqtt:debug 6.3.2019, 17:40:10 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'LCT003' (0x00178801045450e8)
zigbee2mqtt:info 6.3.2019, 17:40:10 MQTT publish: topic 'zigbee2mqtt/wohnzimmer.light.wall.middle', payload '{"state":"OFF"}'

zigbee2mqtt:debug 6.3.2019, 17:40:11 Received MQTT message on 'zigbee2mqtt/wohnzimmer.light.wall.middle/set' with data '{"state": "ON"}'
zigbee2mqtt:info 6.3.2019, 17:40:11 Zigbee publish to device '0x00178801045450e8', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:info 6.3.2019, 17:40:11 MQTT publish: topic 'zigbee2mqtt/wohnzimmer.light.wall.middle', payload '{"state":"ON"}'

zigbee2mqtt:debug 6.3.2019, 17:40:12 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'LCT003' (0x00178801045450e8)
zigbee2mqtt:info 6.3.2019, 17:40:12 MQTT publish: topic 'zigbee2mqtt/wohnzimmer.light.wall.middle', payload '{"state":"ON"}'

zigbee2mqtt:debug 6.3.2019, 17:40:13 Received MQTT message on 'zigbee2mqtt/wohnzimmer.light.wall.right/set' with data '{"state": "ON"}'
zigbee2mqtt:info 6.3.2019, 17:40:13 Zigbee publish to device '0x0017880104545693', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:info 6.3.2019, 17:40:13 MQTT publish: topic 'zigbee2mqtt/wohnzimmer.light.wall.right', payload '{"state_wohnzimmer.light.wall.right":"ON","color_temp":230,"color_mode":1,"color":{"x":0.368,"y":0.269,"saturation":46.1,"hue":295.9}}'

zigbee2mqtt:debug 6.3.2019, 17:40:14 Received MQTT message on 'zigbee2mqtt/wohnzimmer.light.wall.left/set' with data '{"state": "ON"}'
zigbee2mqtt:info 6.3.2019, 17:40:14 Zigbee publish to device '0x00178801045453fb', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:info 6.3.2019, 17:40:14 MQTT publish: topic 'zigbee2mqtt/wohnzimmer.light.wall.left', payload '{"state_wohnzimmer.light.wall.left":"ON","brightness":143}'
```

Koenkk, 2019-03-06:
What is the model of your hue bulbs?

sehraf, 2019-03-06:
They are all `8718696485880 - Philips Hue white and color ambiance GU10 (Router)`. 
(They were working before)

Koenkk, 2019-03-06:
This was indeed a nasty bug, fixed it (also added some testcases to prevent it from happening again!). Fixed in the dev branch.

closed 2019-03-06 by Koenkk