Koenkk/zigbee2mqtt #1201: Jeedom integration and Ikea bulbs

z4rn0x, 2019-03-06:
Hello Koenkk
I'm using zigbee2mqtt successfully since about 6 months with Xiaomi devices like the switch and temp/pressure sensors. It is working great and I'm very thankful for that! It's integrated in Jeedom.
Recently I tried to add a new device (Ikea LED1545G12). I see in the zigbee2mqtt log that it's paired successfully, but nothing more, I don't see in my plugin... After a hundred of pairing procedures, a thousands of tests and trials, I might have found the root cause of the problem...
Even if I'm not using Home Assistant, I set the homeassistant value to true in the configuration.yaml file. And then I got a lot of equipment automatically created in Jeedom... So the communication is obviously working, but the interpretation must be the problem. Looking at the equipment created, I identified some / and \ that are probably in excess... see the pic attached...
[image]
Is it something you would be able to fix in one of the coming release? I'd love to help you on this, but I'm not competent at all to do that...
I understand that there are many home automation software and you won't be able to solve everything, but this issue looks like a minor one (according to me)
Let me know if I can do any other trial or provide some sort of help.
Thanks again for what you've done so far!

Koenkk, 2019-03-06:
Can you provide the log where you see those \/. I want to isolate that this is a zigbee2mqtt issue.

z4rn0x, 2019-03-07:
I don't see these \ / in a log (at least not in a log that I know...) but in the mqtt plugin in Jeedom.
But here are some files that might help to understand my problem: 
Service journal:
[image]
Configuration.yaml:
[image]
State.json (the bulb doesn't appear in this one...):
[image]
What kind of log do you need me to upload to help you?

Koenkk, 2019-03-07:
This looks like an issue with the Jeedom MQTT plugin, not with zigbee2mqtt. I would recommend creating an issue there.

z4rn0x, 2019-03-08:
Might be... except that it's working fine for the xiaomi sensors and switch...
Just to be sure that I was clear: Jeedom doesn't see any MQTT message from the bulb (and only from the bulb) unless I change the homeassistant value to true (which I don't need to change for the sensors and switch). But if I do so, the parsejson gives "\ /"... I think I put you on the wrong track by mentioning homeassistant value.
In short: after pairing the bulb (which went fine according to the logs), I don't see any other message from the bulb than "payload '{"type":"pairing","message":"device incoming"}'
Does it make sense?

Koenkk, 2019-03-09:
Can you turn it on/off directly via MQTT? (http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttdevice_idset)

z4rn0x, 2019-03-10:
Yeaahhh, it's now included! For any reason, it wasn't going through auto inclusion as it does for the Xiaomi sensors... But anyway, for those who will meet the same issue:
- In your MQTT plugin, go to the zigbee2mqtt equipment
- Go to Commands
- Add a command action
- Name it
- Topic, enter zigbee2mqtt/0x000b57fffed291d3/set
- Value, enter {"state":"ON","brightness":255,"color_temp":300,"transition":1}
- And Voila! the new equipment is identified
Thank you so much for your help @Koenkk !

closed 2019-03-10 by z4rn0x