Koenkk/zigbee2mqtt #1202: Ikea LED1624G9 undefined

SBTephan, 2019-03-06:
I bought the  Ikea LED1624G9 Color lamp. Strangly enough it says undefined when connecting it.
But it is in the supported list.
So I installed the DEV branch to see if that makes any difference. removed the device from the configuration and DB an reconnected it.
Same message "undefined" 

here is the DB entry:

{"id":20,"type":"Router","ieeeAddr":"0x000d6ffffe21dfca","nwkAddr":41068,"manufId":4476,"epList":[1],"status":"online","joinTime":1551903007,"endpoints":{"1":{"profId":49246,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":0,"16398":0,"65533":1,"remainingTime":0,"currentX":30015,"currentY":26870,"colorMode":2,"numPrimaries":0,"enhancedColorMode":1,"colorCapabilities":8,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":124,"lastMessageRssi":-69}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"neg0kjloQgAxqOZM"}

Any idea why it is saying undefined?

Best regards,
Stephan

Koenkk, 2019-03-06:
It failed to determine the modelId.

- Stop zigbee2mqtt
- Remove all entries of `0x000d6ffffe21dfca` from `database.db`
- Start zigbee2mqtt
- Re-pair device.

SBTephan, 2019-03-06:
That was the trick, Thanks!

closed 2019-03-07 by Koenkk

Pockrandt, 2020-01-07:
who ever may find this I had the same issue with the same bulb but the solution was slightly different.
Needed to repair it but ran into `modelID 'undefined'`  with no entry in the `database.db` after removing it from the network and restart & re-stick. 

```
info  2020-01-07 18:43:52: Zigbee: allowing new devices to join.
info  2020-01-07 18:43:52: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0-dev","commit":"2451d536f2c9e446d396df55c9493f256ab9352e","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190223}},"log_level":"info","permit_join":true}'
info  2020-01-07 18:45:27: Device 'ikea_wz_rgb_01' joined
info  2020-01-07 18:45:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"ikea_wz_rgb_01"}}'
info  2020-01-07 18:45:27: Starting interview of 'ikea_wz_rgb_01'
info  2020-01-07 18:45:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"ikea_wz_rgb_01"}}'
info  2020-01-07 18:45:27: MQTT publish: topic 'zigbee2mqtt/ikea_wz_rgb_01/availability', payload 'online'
warn  2020-01-07 18:46:01: Device with modelID 'undefined' is not supported.
warn  2020-01-07 18:46:01: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

after removing the device from `devices.yaml` the pairing worked.

```
zigbee2mqtt:info  2020-01-07 18:54:15: Starting interview of '0x000d6xxxxxxxxxxx'
zigbee2mqtt:info  2020-01-07 18:54:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6xxxxxxxxxxx"}}'
zigbee2mqtt:info  2020-01-07 18:54:15: MQTT publish: topic 'zigbee2mqtt/0x000d6xxxxxxxxxxx/availability', payload 'online'
zigbee2mqtt:info  2020-01-07 18:54:25: MQTT publish: topic 'homeassistant/light/0x000d6xxxxxxxxxxx/light/config', payload '{"brightness":true,"xy":true,"schema":"json","command_topic":"zigbee2mqtt/0x000d6xxxxxxxxxxx/set","state_topic":"zigbee2mqtt/0x000d6xxxxxxxxxxx","json_attributes_topic":"zigbee2mqtt/0x000d6xxxxxxxxxxx","name":"0x000d6xxxxxxxxxxx_light","unique_id":"0x000d6xxxxxxxxxxx_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6xxxxxxxxxxx"],"name":"0x000d6xxxxxxxxxxx","sw_version":"Zigbee2mqtt 1.8.0-dev","model":"TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/0x000d6xxxxxxxxxxx/availability"}'
zigbee2mqtt:info  2020-01-07 18:54:25: MQTT publish: topic 'homeassistant/sensor/0x000d6xxxxxxxxxxx/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6xxxxxxxxxxx","json_attributes_topic":"zigbee2mqtt/0x000d6xxxxxxxxxxx","name":"0x000d6xxxxxxxxxxx_linkquality","unique_id":"0x000d6xxxxxxxxxxx_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6xxxxxxxxxxx"],"name":"0x000d6xxxxxxxxxxx","sw_version":"Zigbee2mqtt 1.8.0-dev","model":"TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/0x000d6xxxxxxxxxxx/availability"}'
zigbee2mqtt:info  2020-01-07 18:54:27: Successfully interviewed '0x000d6xxxxxxxxxxx', device has successfully been paired
zigbee2mqtt:info  2020-01-07 18:54:27: Device '0x000d6xxxxxxxxxxx' is supported, identified as: IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)
```