Koenkk/zigbee2mqtt #1203: Unable to turn light off with brightness control

GForce2010, 2019-03-06:
I am using Zigbee2MQTT to control several Ikea bulbs through OpenHAB.

I know that the bulb has two control items, State ON/OFF and Brightness 0-255.

What I would like to be able to do is control the bulb fully just by using the brightness control i.e if I want to turn the bulb off I set Brightness to 0. Unfortunately this is not what happens.

When I send the command to set Brightness to 0 the bulb stays on at low brightness and the state reports the Brightness is 1.

Because of this I am having to do extra work in OpenHAB so that it sends a State OFF when the brightness is set to 0. 

Is there a better way of doing it?

Thanks.

rorie1982, 2019-03-07:
Using the Ikea GU10 spots sending brightness 0 will turn my spots to off. Sending state OFF nothing happens. So maybe this is A bug, I can live with this. What kind of bulbs are you using?

GForce2010, 2019-03-07:
I have just tried again and the bulbs are now turning off but I still have the problem of the reported state.

If I send the following command to any of my bulbs they turn off.

`{  "brightness": 0}`

But it then reports this as its state

`{"state":"ON","brightness":1}`

Meaning my OpenHAB control panel show that the light is still on, 

I believe this state is being reported from the bulb itself so I am starting to think that it is a limitation of the bulbs firmware.

closed 2019-03-08 by GForce2010