Koenkk/zigbee2mqtt #1204: Question abut pairing/sniffing E1524

vasil-d, 2019-03-07:
I am using HomeAssistant on virtual env. with mosquitto mqtt broker.
The instructions are very well/clear written, but I have a question about CC2531 with no clear answer.
Because the Ikea E1524 does requires extra setup with sniffing my question/questions are the following:
1. Do i need a second CC2531 for sniffing the data (as described in the documentation) or 1 is enough for pairing and sniffing.
2. If 1 is enough - do I use the ZBOSE firmware on the device or the CC2531ZNP-Prod.hex?

vasil-d, 2019-03-07:
really, no one?

Koenkk, 2019-03-07:
Please be a less impatient next time.

I would recommend updating to the latest dev branch and dev firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin). This does not require any special setup for the E1524 anymore.

vasil-d, 2019-03-07:
I am very sorry if i did leave the impression of impatience. I meant no disrespect.
Just my question is not complicated, rater simple one.
Thank you!

closed 2019-03-07 by Koenkk

foxy82, 2019-07-28:
Hi - it looks like this has now been merged so link above is no longer valid. Can you confirm this is the 1.2 firmware or do I need the 3.0 firmware?