Koenkk/zigbee2mqtt #1205: Incorrect state reported after group control

GForce2010, 2019-03-07:
I have been setting up some groups and I have come across what I believe is a bug.

I have a group called study, if I send the following: -
Topic `zigbee2mqtt/Study/set`

Sent
`{"state":"ON"}`
Received
`{"brightness":100,"state":"ON"}`

Sent
`{"state":"OFF"}`
Received
`{"brightness":100,"state":"OFF"}`
The above work how they should, below is the problem

Sent
`{"brightness":255}`
Received
`{"brightness":255,"state":"OFF"}`

The lights turn on to full brightness but the state is still reported as OFF

Sent
`{"state":"ON","brightness":255}`
Received
`{"brightness":255,"state":"OFF"}`
Sending a state ON also doesn't change the state reported back.

This is different to controlling my single lights, if you send a brightness value to them they always report back a state of ON.

This is messing up my control board because if the state id OFF it assumes the light is off even though the group is actually on.

below is the log from me sending the above commands.

```
3/7/2019, 5:28:03 PM - debug: Ping 0x00124b001bd1250b
3/7/2019, 5:28:04 PM - debug: Ping 0x000b57fffec348ae
3/7/2019, 5:28:04 PM - debug: Successfully pinged 0x000b57fffec348ae
3/7/2019, 5:28:04 PM - info: MQTT publish: topic 'zigbee2mqtt/Bathroom/availability', payload 'online'
3/7/2019, 5:28:11 PM - debug: Ping 0x000b57fffee3187f
3/7/2019, 5:28:11 PM - debug: Successfully pinged 0x000b57fffee3187f
3/7/2019, 5:28:11 PM - info: MQTT publish: topic 'zigbee2mqtt/Dining_Room/availability', payload 'online'
3/7/2019, 5:28:12 PM - debug: Ping 0x000b57fffeeb302e
3/7/2019, 5:28:12 PM - debug: Successfully pinged 0x000b57fffeeb302e
3/7/2019, 5:28:12 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Livingroom_4/availability', payload 'online'
3/7/2019, 5:28:13 PM - debug: Ping 0x000b57fffee1c9f7
3/7/2019, 5:28:13 PM - debug: Successfully pinged 0x000b57fffee1c9f7
3/7/2019, 5:28:13 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_2/availability', payload 'online'
3/7/2019, 5:28:13 PM - debug: Ping 0x0017880102ccd362
3/7/2019, 5:28:13 PM - debug: Successfully pinged 0x0017880102ccd362
3/7/2019, 5:28:13 PM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen/availability', payload 'online'
3/7/2019, 5:28:13 PM - debug: Failed to ping 0x00124b001bd1250b
3/7/2019, 5:28:13 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00124b001bd1250b/availability', payload 'offline'
3/7/2019, 5:28:13 PM - debug: Ping 0x00124b001bd1250b
3/7/2019, 5:28:14 PM - debug: Ping 0x90fd9ffffe10a97c
3/7/2019, 5:28:14 PM - debug: Successfully pinged 0x90fd9ffffe10a97c
3/7/2019, 5:28:14 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_3/availability', payload 'online'
3/7/2019, 5:28:14 PM - debug: Ping 0x0017880102cf693e
3/7/2019, 5:28:14 PM - debug: Successfully pinged 0x0017880102cf693e
3/7/2019, 5:28:14 PM - info: MQTT publish: topic 'zigbee2mqtt/Hall/availability', payload 'online'
3/7/2019, 5:28:14 PM - debug: Ping 0x000b57fffed885e1
3/7/2019, 5:28:14 PM - debug: Successfully pinged 0x000b57fffed885e1
3/7/2019, 5:28:14 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Bedroom_1/availability', payload 'online'
3/7/2019, 5:28:14 PM - debug: Ping 0x00158d00020a5adc
3/7/2019, 5:28:14 PM - debug: Successfully pinged 0x00158d00020a5adc
3/7/2019, 5:28:14 PM - info: MQTT publish: topic 'zigbee2mqtt/His_Lamp/availability', payload 'online'
3/7/2019, 5:28:16 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3125,"3":12,"4":5032,"5":20,"6":[0,3],"10":41344,"100":0}}}' of device 'lumi.sensor_magnet.aq2' (0x00158d000312f736)
3/7/2019, 5:28:16 PM - info: MQTT publish: topic 'zigbee2mqtt/Mag_Sen_1', payload '{"contact":true,"linkquality":36,"battery":100,"voltage":3125}'
3/7/2019, 5:28:16 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3125,null,12,5032,20,[0],null,null,null,41344,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0]}}' of device 'lumi.sensor_magnet.aq2' (0x00158d000312f736)
3/7/2019, 5:28:18 PM - debug: Ping 0x000b57fffed6f0c3
3/7/2019, 5:28:18 PM - debug: Successfully pinged 0x000b57fffed6f0c3
3/7/2019, 5:28:18 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Livingroom_2/availability', payload 'online'
3/7/2019, 5:28:19 PM - debug: Received MQTT message on 'zigbee2mqtt/Study/set' with data '{"state":"ON"}'
3/7/2019, 5:28:19 PM - info: Zigbee publish to group '1', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/7/2019, 5:28:19 PM - info: MQTT publish: topic 'zigbee2mqtt/Study', payload '{"brightness":100,"state":"ON"}'
3/7/2019, 5:28:23 PM - error: Failed to ping 0x00124b001bd1250b
3/7/2019, 5:28:25 PM - debug: Ping 0x90fd9ffffe10a729
3/7/2019, 5:28:25 PM - debug: Successfully pinged 0x90fd9ffffe10a729
3/7/2019, 5:28:25 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Livingroom_1/availability', payload 'online'
3/7/2019, 5:28:26 PM - debug: Received MQTT message on 'zigbee2mqtt/Study/set' with data '{"state":"OFF"}'
3/7/2019, 5:28:26 PM - info: Zigbee publish to group '1', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/7/2019, 5:28:26 PM - info: MQTT publish: topic 'zigbee2mqtt/Study', payload '{"brightness":100,"state":"OFF"}'
3/7/2019, 5:28:28 PM - debug: Ping 0x000b57fffe9f4635
3/7/2019, 5:28:29 PM - debug: Successfully pinged 0x000b57fffe9f4635
3/7/2019, 5:28:29 PM - info: MQTT publish: topic 'zigbee2mqtt/Porch/availability', payload 'online'
3/7/2019, 5:28:30 PM - debug: Ping 0x000b57fffec2c92d
3/7/2019, 5:28:30 PM - debug: Successfully pinged 0x000b57fffec2c92d
3/7/2019, 5:28:30 PM - info: MQTT publish: topic 'zigbee2mqtt/Landing/availability', payload 'online'
3/7/2019, 5:28:33 PM - debug: Received MQTT message on 'zigbee2mqtt/Study/set' with data '{"brightness":255}'
3/7/2019, 5:28:34 PM - info: Zigbee publish to group '1', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
3/7/2019, 5:28:34 PM - info: MQTT publish: topic 'zigbee2mqtt/Study', payload '{"brightness":255,"state":"OFF"}'
3/7/2019, 5:28:41 PM - debug: Received MQTT message on 'zigbee2mqtt/Study/set' with data '{"state":"ON","brightness":255}'
3/7/2019, 5:28:41 PM - debug: Skipping 'state' because of 'brightness'
3/7/2019, 5:28:41 PM - debug: Ping 0x90fd9ffffe13a62b
3/7/2019, 5:28:41 PM - debug: Successfully pinged 0x90fd9ffffe13a62b
3/7/2019, 5:28:41 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_1/availability', payload 'online'
3/7/2019, 5:28:41 PM - info: Zigbee publish to group '1', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
3/7/2019, 5:28:41 PM - info: MQTT publish: topic 'zigbee2mqtt/Study', payload '{"brightness":255,"state":"OFF"}'
3/7/2019, 5:28:46 PM - debug: Ping 0x00158d00020a5a56
3/7/2019, 5:28:46 PM - debug: Successfully pinged 0x00158d00020a5a56
3/7/2019, 5:28:46 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp/availability', payload 'online'
```

Thanks.

Koenkk, 2019-03-08:
Can you check if this is fixed in the dev branch?

GForce2010, 2019-03-08:
I can corm that this is now fixed in the latest dev branch.

Thank you for all your hard work.

closed 2019-03-08 by GForce2010