Koenkk/zigbee2mqtt #1208: Add support for 'FNB56-ZSW01LX2.0'

peter1960, 2019-03-08:
The 'FNB56-ZSW01LX2.0' is a newer version of 'FB56+ZSW05HG1.2'.

Can the following be added to the devices.js please under Nue, the existing converter works

    {
        zigbeeModel: ['FNB56-ZSW01LX2.0'],
        model: 'FNB56-ZSW01LX2.0',
        vendor: 'Nue',
        description: 'ZigBee one gang smart switch',
        supports: 'on/off',
        fromZigbee: [fz.generic_state],
        toZigbee: [tz.on_off],
    },

Koenkk, 2019-03-08:
Just to be sure, you got it working with the above addition to `devices.js`, right?

peter1960, 2019-03-09:
Below is the results after adding that and restart:

Start:
2019-3-8 15:30:37 - info: GarageLight (0x00158d0002926dad): FNB56-ZSW01LX2.0 - Nue ZigBee one gang smart switch (Router)

2019-3-8 15:30:37 - info: MQTT publish: topic 'zigbee2mqtt/GarageLight', payload '{"state":"ON","linkquality":49,"last_seen":"2019-03-08T04:23:40.164Z","device":{"ieeeAddr":"0x00158d0002926dad","friendlyName":"GarageLight","type":"Router","nwkAddr":20560,"manufId":4478,"manufName":
"FeiBit","powerSource":"Mains (single phase)","modelId":"FNB56-ZSW01LX2.0","status":"offline"},"elapsed":1200102}'

Turn On:
2019-3-9 13:43:46 - info: MQTT publish: topic 'zigbee2mqtt/GarageLight', payload '{"state":"ON","linkquality":57,"last_seen":"2019-03-09T02:43:46.827Z","device":{"ieeeAddr":"0x00158d0002926dad","friendlyName":"GarageLight","type":"Router","nwkAddr":20560,"manufId":4478,"manufName":"FeiBit","powerSource":"Mains (single phase)","modelId":"FNB56-ZSW01LX2.0","status":"offline"},"elapsed":1200088}'

Turn Off
2019-3-9 13:57:44 - info: MQTT publish: topic 'zigbee2mqtt/GarageLight', payload '{"state":"OFF","linkquality":65,"last_seen":"2019-03-09T02:57:44.135Z","device":{"ieeeAddr":"0x00158d0002926dad","friendlyName":"GarageLight","type":"Router","nwkAddr":20560,"manufId":4478,"manufName":"FeiBit","powerSource":"Mains (single phase)","modelId":"FNB56-ZSW01LX2.0","status":"offline"},"elapsed":837308}'

highground88, 2019-03-09:
I didn't see this until just now but I've already tested and submitted a pull request earlier as I just got one of these:

https://github.com/Koenkk/zigbee-shepherd-converters/pull/325

@Koenkk , I just got a new batch of switches and they have indeed changed models. I'll continue to work through my current and new models, but so far there seem to be some changes internally. The CC2530 chip is replaced with a JN5169 chip. I'll get back to these changes with you eventually but anything with '2.0' on the end of the model migth have different endpoints - e.g. the onlder 3 gang uses 11,12,13. New '2.0' one uses 1,2,3 and adds another undefined state, plus another "state":'OFF/ON'.

You can probably close this one though....

Koenkk, 2019-03-09:
@highground88 great, so it's already supported, closing this.

closed 2019-03-09 by Koenkk