Koenkk/zigbee2mqtt #1209: Availability functionality leads to z2m reboot

milakov, 2019-03-08:
I am seeing zigbee2mqtt restarting in a minute after it fails to ping a router, for example below is the tail of log file:
```
3/8/2019, 1:24:58 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00124b001b50c0a6/availability', payload 'online'
3/8/2019, 1:25:05 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":36,"linkquality":70,"last_seen":"2019-03-08T13:25:05+03:00"}'
3/8/2019, 1:25:06 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002561266/availability', payload 'online'
3/8/2019, 1:25:10 PM - error: Failed to ping 0x00124b001b50416b
3/8/2019, 1:25:10 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00124b001b50416b/availability', payload 'offline'
3/8/2019, 1:25:21 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":36,"linkquality":68,"last_seen":"2019-03-08T13:25:21+03:00"}'
3/8/2019, 1:25:21 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002561266', payload '{"state":"ON","power":0,"voltage":null,"consumption":455.35,"temperature":39,"linkquality":44,"last_seen":"2019-03-08T13:25:21+03:00"}'
3/8/2019, 1:25:29 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00124b001b505747/availability', payload 'online'
3/8/2019, 1:25:37 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":36,"linkquality":70,"last_seen":"2019-03-08T13:25:37+03:00"}'
3/8/2019, 1:25:53 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":36,"linkquality":73,"last_seen":"2019-03-08T13:25:53+03:00"}'
3/8/2019, 1:25:58 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00124b001b50c0a6/availability', payload 'online'
3/8/2019, 1:26:09 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":36,"linkquality":73,"last_seen":"2019-03-08T13:26:09+03:00"}'
3/8/2019, 1:26:09 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":36,"linkquality":73,"last_seen":"2019-03-08T13:26:09+03:00"}'
```

milakov, 2019-03-08:
Just caught it with debug on:
```  zigbee2mqtt:info 3/8/2019, 2:07:30 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001b505747/availability', payload 'online'
  zigbee2mqtt:error 3/8/2019, 2:07:38 PM Failed to ping 0x00124b001b50416b
  zigbee2mqtt:info 3/8/2019, 2:07:38 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001b50416b/availability', payload 'offline'
  zigbee2mqtt:debug 3/8/2019, 2:07:39 PM Received zigbee message of type 'attReport' with data '{"cid":"genAnalogInput","data":{"presentValue":0}}' of device 'lumi.plug' (0x00158d00024f2e59)
  zigbee2mqtt:info 3/8/2019, 2:07:39 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":37,"linkquality":84,"last_seen":"2019-03-08T14:07:39+03:00"}'
  zigbee2mqtt:debug 3/8/2019, 2:07:53 PM Ping 0x00158d00024f2e59
  zigbee2mqtt:debug 3/8/2019, 2:07:53 PM Successfully pinged 0x00158d00024f2e59
  zigbee2mqtt:debug 3/8/2019, 2:07:53 PM Ping 0x00158d0002561266
  zigbee2mqtt:debug 3/8/2019, 2:07:53 PM Successfully pinged 0x00158d0002561266
  zigbee2mqtt:debug 3/8/2019, 2:07:55 PM Received zigbee message of type 'attReport' with data '{"cid":"genAnalogInput","data":{"presentValue":0}}' of device 'lumi.plug' (0x00158d00024f2e59)
  zigbee2mqtt:info 3/8/2019, 2:07:55 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":37,"linkquality":84,"last_seen":"2019-03-08T14:07:55+03:00"}'
  zigbee2mqtt:debug 3/8/2019, 2:07:59 PM Ping 0x00124b001b50c0a6
  zigbee2mqtt:debug 3/8/2019, 2:07:59 PM Successfully pinged 0x00124b001b50c0a6
  zigbee2mqtt:info 3/8/2019, 2:07:59 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001b50c0a6/availability', payload 'online'
  zigbee2mqtt:debug 3/8/2019, 2:08:09 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3082},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":823},{"elmType":32,"elmVal":104}]}}' of device 'lumi.sensor_switch' (0x00158d00026b76b6)
  zigbee2mqtt:info 3/8/2019, 2:08:09 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00026b76b6', payload '{"battery":100,"voltage":3082,"linkquality":63,"last_seen":"2019-03-08T14:08:09+03:00"}'
  zigbee2mqtt:debug 3/8/2019, 2:08:09 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3082},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":823},{"elmType":32,"elmVal":104}]}}' of device 'lumi.sensor_switch' (0x00158d00026b76b6)
  zigbee2mqtt:debug 3/8/2019, 2:08:11 PM Received zigbee message of type 'attReport' with data '{"cid":"genAnalogInput","data":{"presentValue":0}}' of device 'lumi.plug' (0x00158d00024f2e59)
  zigbee2mqtt:info 3/8/2019, 2:08:11 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":37,"linkquality":86,"last_seen":"2019-03-08T14:08:11+03:00"}'
  zigbee2mqtt:debug 3/8/2019, 2:08:16 PM Ping 0x00158d0002561266
  zigbee2mqtt:debug 3/8/2019, 2:08:16 PM Successfully pinged 0x00158d0002561266
  zigbee2mqtt:info 3/8/2019, 2:08:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002561266/availability', payload 'online'
  zigbee2mqtt:debug 3/8/2019, 2:08:27 PM Received zigbee message of type 'attReport' with data '{"cid":"genAnalogInput","data":{"presentValue":0}}' of device 'lumi.plug' (0x00158d00024f2e59)
  zigbee2mqtt:info 3/8/2019, 2:08:27 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"ON","power":0,"voltage":193.8,"consumption":103.39,"temperature":37,"linkquality":84,"last_seen":"2019-03-08T14:08:27+03:00"}'
  zigbee2mqtt:debug 3/8/2019, 2:08:30 PM Ping 0x00124b001b505747
  zigbee2mqtt:debug 3/8/2019, 2:08:30 PM Successfully pinged 0x00124b001b505747
  zigbee2mqtt:info 3/8/2019, 2:08:30 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001b505747/availability', payload 'online'
  zigbee2mqtt:info 3/8/2019, 2:08:35 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00029c0149', payload '{"illuminance":119,"linkquality":28,"last_seen":"2019-03-08T14:08:35+03:00","occupancy":false,"battery":100,"voltage":3035}'
  zigbee2mqtt:debug 3/8/2019, 2:08:38 PM Ping 0x00124b001b50416b
  zigbee2mqtt:debug 3/8/2019, 2:08:38 PM Successfully pinged 0x00124b001b50416b
/app/node_modules/q/q.js:155
                throw e;
                ^
TypeError: Cannot read property 'hasOwnProperty' of null
    at Object.convert (/app/node_modules/zigbee-shepherd-converters/converters/toZigbee.js:1211:25)
    at converters.forEach (/app/lib/extension/deviceAvailability.js:115:45)
    at Array.forEach (<anonymous>)
    at DeviceAvailability.onReconnect (/app/lib/extension/deviceAvailability.js:114:24)
    at DeviceAvailability.publishAvailability (/app/lib/extension/deviceAvailability.js:128:18)
    at zigbee.ping (/app/lib/extension/deviceAvailability.js:73:18)
    at shepherd.controller.checkOnline (/app/lib/zigbee.js:288:25)
    at /app/node_modules/q/q.js:2055:17
    at runSingle (/app/node_modules/q/q.js:137:13)
    at flush (/app/node_modules/q/q.js:125:13)
npm
 ERR!
 code ELIFECYCLE
npm ERR!
 errno 1
npm
 ERR! zigbee2mqtt@1.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR!
 Failed at the zigbee2mqtt@1.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.```

Koenkk, 2019-03-08:
Should be fixed in the latest dev branch, can you verify?

milakov, 2019-03-08:
Still crashes
```
  zigbee2mqtt:info 3/8/2019, 8:37:31 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001b505747/availability', payload 'online'
  zigbee2mqtt:error 3/8/2019, 8:37:39 PM Failed to ping 0x00124b001b50416b
  zigbee2mqtt:info 3/8/2019, 8:37:39 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001b50416b/availability', payload 'offline'
  zigbee2mqtt:debug 3/8/2019, 8:37:53 PM Ping 0x00158d00024f2e59
  zigbee2mqtt:debug 3/8/2019, 8:37:53 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3055,"3":33,"4":5032,"5":608,"6":[0,1],"8":5125,"10":45761}}}' of device 'lumi.sensor_86sw2Un' (0x00158d000254cced)
  zigbee2mqtt:info 3/8/2019, 8:37:53 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000254cced', payload '{"battery":100,"voltage":3055,"linkquality":23,"last_seen":"2019-03-08T20:37:53+03:00"}'
  zigbee2mqtt:debug 3/8/2019, 8:37:53 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3055,null,33,5032,608,[0,1],null,5125,null,45761]}}' of device 'lumi.sensor_86sw2Un' (0x00158d000254cced)
  zigbee2mqtt:debug 3/8/2019, 8:37:53 PM Successfully pinged 0x00158d00024f2e59
  zigbee2mqtt:debug 3/8/2019, 8:37:53 PM Ping 0x00158d0002561266
  zigbee2mqtt:debug 3/8/2019, 8:37:53 PM Successfully pinged 0x00158d0002561266
  zigbee2mqtt:debug 3/8/2019, 8:37:57 PM Ping 0x00158d00024f2e59
  zigbee2mqtt:debug 3/8/2019, 8:37:57 PM Successfully pinged 0x00158d00024f2e59
  zigbee2mqtt:info 3/8/2019, 8:37:57 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59/availability', payload 'online'
  zigbee2mqtt:debug 3/8/2019, 8:37:58 PM Ping 0x00158d0002561266
  zigbee2mqtt:debug 3/8/2019, 8:37:59 PM Successfully pinged 0x00158d0002561266
  zigbee2mqtt:info 3/8/2019, 8:37:59 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002561266/availability', payload 'online'
  zigbee2mqtt:debug 3/8/2019, 8:38:00 PM Ping 0x00124b001b50c0a6
  zigbee2mqtt:debug 3/8/2019, 8:38:00 PM Successfully pinged 0x00124b001b50c0a6
  zigbee2mqtt:info 3/8/2019, 8:38:00 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001b50c0a6/availability', payload 'online'
  zigbee2mqtt:debug 3/8/2019, 8:38:07 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"3":39,"5":49,"7":"0x0000000000000000","8":4894,"9":3339,"100":1,"149":455.3517150878906,"152":0,"154":16}}}' of device 'lumi.plug' (0x00158d0002561266)
  zigbee2mqtt:info 3/8/2019, 8:38:07 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002561266', payload '{"state":"ON","power":0,"voltage":null,"consumption":455.35,"temperature":39,"linkquality":26,"last_seen":"2019-03-08T20:38:07+03:00"}'
  zigbee2mqtt:debug 3/8/2019, 8:38:07 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,null,null,39,null,49,null,"0x0000000000000000",4894,3339,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,1,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,455.3517150878906,null,null,0,null,16]}}' of device 'lumi.plug' (0x00158d0002561266)
  zigbee2mqtt:debug 3/8/2019, 8:38:31 PM Ping 0x00124b001b505747
  zigbee2mqtt:debug 3/8/2019, 8:38:31 PM Successfully pinged 0x00124b001b505747
  zigbee2mqtt:info 3/8/2019, 8:38:31 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001b505747/availability', payload 'online'
  zigbee2mqtt:debug 3/8/2019, 8:38:39 PM Ping 0x00124b001b50416b
  zigbee2mqtt:debug 3/8/2019, 8:38:39 PM Successfully pinged 0x00124b001b50416b
/app/node_modules/q/q.js:155
                throw e;
                ^
TypeError: Cannot read property 'hasOwnProperty' of null
    at Object.convert (/app/node_modules/zigbee-shepherd-converters/converters/toZigbee.js:1211:25)
    at converters.forEach (/app/lib/extension/deviceAvailability.js:115:45)
    at Array.forEach (<anonymous>)
    at DeviceAvailability.onReconnect (/app/lib/extension/deviceAvailability.js:114:24)
    at DeviceAvailability.publishAvailability (/app/lib/extension/deviceAvailability.js:128:18)
    at zigbee.ping (/app/lib/extension/deviceAvailability.js:73:18)
    at shepherd.controller.checkOnline (/app/lib/zigbee.js:288:25)
    at /app/node_modules/q/q.js:2055:17
    at runSingle (/app/node_modules/q/q.js:137:13)
    at flush (/app/node_modules/q/q.js:125:13)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
 
npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
npm
 ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-08T17_38_39_642Z-debug.log
2019-03-08T20:38:44: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-03-08T20:38:44: PM2 log: App [npm:0] starting in -fork mode-
2019-03-08T20:38:44: PM2 log: App [npm:0] online
```

Koenkk, 2019-03-08:
Can you post the startup log?

milakov, 2019-03-08:
```
3/8/2019, 8:14:50 PM - info: Logging to directory: '/share/zigbee2mqtt/log/2019-03-08.20-14-50'
3/8/2019, 8:14:51 PM - info: Starting zigbee2mqtt version 1.1.1 (commit #unknown)
3/8/2019, 8:14:51 PM - info: Starting zigbee-shepherd
3/8/2019, 8:14:53 PM - info: zigbee-shepherd started
3/8/2019, 8:14:53 PM - info: Coordinator firmware version: '20190223'
3/8/2019, 8:14:53 PM - info: Currently 24 devices are joined:
3/8/2019, 8:14:53 PM - info: 0x00158d000254cced (0x00158d000254cced): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00024f2e59 (0x00158d00024f2e59): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
3/8/2019, 8:14:53 PM - info: 0x00158d0002561266 (0x00158d0002561266): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
3/8/2019, 8:14:53 PM - info: 0x00158d0001ef8655 (0x00158d0001ef8655): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00024a5018 (0x00158d00024a5018): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d0001ef852b (0x00158d0001ef852b): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d0002720cd1 (0x00158d0002720cd1): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d0002775e09 (0x00158d0002775e09): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00124b001b50416b (0x00124b001b50416b): GL-B-007Z - Gledopto Smart 6W E27 RGB / CW LED bulb (Router)
3/8/2019, 8:14:53 PM - info: 0x00158d000288e3c5 (0x00158d000288e3c5): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00022cb428 (0x00158d00022cb428): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00029c070e (0x00158d00029c070e): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00023891e9 (0x00158d00023891e9): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d000288e3b1 (0x00158d000288e3b1): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d000101b949 (0x00158d000101b949): MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d000288e352 (0x00158d000288e352): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00024a4e92 (0x00158d00024a4e92): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00029c0149 (0x00158d00029c0149): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00027c1b4c (0x00158d00027c1b4c): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d00026b76b6 (0x00158d00026b76b6): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00124b001b50c0a6 (0x00124b001b50c0a6): GL-B-007Z - Gledopto Smart 6W E27 RGB / CW LED bulb (Router)
3/8/2019, 8:14:53 PM - info: 0x00124b001b505747 (0x00124b001b505747): GL-B-008Z - Gledopto Smart 12W E27 RGB / CW LED bulb (Router)
3/8/2019, 8:14:53 PM - info: 0x00158d00026b773d (0x00158d00026b773d): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
3/8/2019, 8:14:53 PM - info: 0x00158d000272652f (0x00158d000272652f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/8/2019, 8:14:53 PM - info: Zigbee: disabling joining new devices.
3/8/2019, 8:14:53 PM - info: Connecting to MQTT server at mqtt://192.168.1.170
3/8/2019, 8:14:54 PM - info: zigbee-shepherd ready
3/8/2019, 8:14:54 PM - info: Connected to MQTT server
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000254cced', payload '{"battery":100,"voltage":3045,"linkquality":84,"last_seen":"2019-03-08T19:47:59+03:00"}'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"OFF","power":0,"voltage":204,"consumption":103.39,"temperature":33,"linkquality":60,"last_seen":"2019-03-08T20:02:06+03:00"}'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002561266', payload '{"state":"ON","power":0,"voltage":null,"consumption":455.35,"temperature":39,"linkquality":68,"last_seen":"2019-03-08T20:05:46+03:00"}'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ef8655', payload '{"battery":100,"voltage":3075,"linkquality":73,"last_seen":"2019-03-08T19:39:58+03:00"}'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024a5018', payload '{"battery":100,"voltage":3025,"linkquality":18,"last_seen":"2019-03-08T19:38:50+03:00","water_leak":false}'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ef852b', payload '{"battery":100,"voltage":3065,"linkquality":81,"last_seen":"2019-03-08T19:46:14+03:00"}'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002720cd1', payload '{"humidity":50.8,"linkquality":26,"last_seen":"2019-03-08T19:41:15+03:00","pressure":987,"temperature":22.1,"battery":100,"voltage":3015}'
[... cut ...]
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00124b001b505747/availability', payload 'online'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00026b773d/availability', payload 'online'
3/8/2019, 8:14:54 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d000272652f/availability', payload 'online'
3/8/2019, 8:14:55 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024f2e59', payload '{"state":"OFF","power":0,"voltage":204,"consumption":103.39,"temperature":33,"linkquality":78,"last_seen":"2019-03-08T20:14:55+03:00"}'
```

Koenkk, 2019-03-09:
Can you try again? There was some build infrastructure problem and because of this new docker images werent generated

milakov, 2019-03-09:
Seems to work fine now. Thanks!

Koenkk, 2019-03-09:
Great, thanks!

closed 2019-03-09 by Koenkk