Koenkk/zigbee2mqtt #1211: Xiaomi Aqara wireless switch WXKG11LM - hold not working

cela96, 2019-03-08:
Hi all, I have correctly configured the device but unfortunately the "hold" function does not work.
Thanks in advance

Koenkk, 2019-03-09:
Can you post your log when performing the hold with `log_level: debug`? (http://www.zigbee2mqtt.io/configuration/configuration.html)

sermayoral, 2019-03-12:
it happens the same to me. Here is the log, but no new lines appear when I hold the button. Here is the log with a single click and hold:

```
[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-03-12T23:33:11: PM2 log: Launching in no daemon mode
2019-03-12T23:33:11: PM2 log: App [npm:0] starting in -fork mode-
2019-03-12T23:33:11: PM2 log: App [npm:0] online
> zigbee2mqtt@1.2.1 start /zigbee2mqtt-1.2.1
> node index.js
  zigbee2mqtt:info 3/12/2019, 11:33:16 PM Logging to directory: '/share/zigbee2mqtt/log/2019-03-12.23-33-16'
  zigbee2mqtt:debug 3/12/2019, 11:33:16 PM Removing old log directory '/share/zigbee2mqtt/log/2019-02-28.11-25-52'
  zigbee2mqtt:debug 3/12/2019, 11:33:17 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 3/12/2019, 11:33:18 PM Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:debug 3/12/2019, 11:33:18 PM Saving state to file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 3/12/2019, 11:33:18 PM Starting zigbee2mqtt version 1.2.1 (commit #unknown)
  zigbee2mqtt:info 3/12/2019, 11:33:18 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/12/2019, 11:33:21 PM zigbee-shepherd started
  zigbee2mqtt:info 3/12/2019, 11:33:21 PM Coordinator firmware version: '20190109'
  zigbee2mqtt:debug 3/12/2019, 11:33:21 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018df4679","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190109},"startTime":1552430001,"joinTimeLeft":0}
  zigbee2mqtt:info 3/12/2019, 11:33:21 PM Currently 2 devices are joined:
  zigbee2mqtt:info 3/12/2019, 11:33:21 PM xa_dswitch_salon (0x00158d0003152dd8): QBKG03LM - Xiaomi Aqara double key wired wall switch (Router)
  zigbee2mqtt:info 3/12/2019, 11:33:21 PM xa_button_hall (0x00158d0002008976): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
  zigbee2mqtt:info 3/12/2019, 11:33:21 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 3/12/2019, 11:33:21 PM Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 3/12/2019, 11:33:21 PM zigbee-shepherd ready
  zigbee2mqtt:debug 3/12/2019, 11:33:22 PM Mounted the cieApp (epId 11)
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM Connected to MQTT server
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM MQTT publish: topic 'zigbee2mqtt/xa_dswitch_salon', payload '{"state_left":"OFF","linkquality":76,"state_right":"ON"}'
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM MQTT publish: topic 'zigbee2mqtt/xa_button_hall', payload '{"battery":91,"voltage":2985,"linkquality":128}'
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":false}'
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM MQTT publish: topic 'homeassistant/switch/0x00158d0003152dd8/switch_left/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state_left }}","command_topic":"zigbee2mqtt/xa_dswitch_salon/left/set","state_topic":"zigbee2mqtt/xa_dswitch_salon","name":"xa_dswitch_salon_switch_left","unique_id":"0x00158d0003152dd8_switch_left_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0003152dd8","name":"xa_dswitch_salon","sw_version":"Zigbee2mqtt 1.2.1","model":"Aqara double key wired wall switch (QBKG03LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM MQTT publish: topic 'homeassistant/switch/0x00158d0003152dd8/switch_right/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state_right }}","command_topic":"zigbee2mqtt/xa_dswitch_salon/right/set","state_topic":"zigbee2mqtt/xa_dswitch_salon","name":"xa_dswitch_salon_switch_right","unique_id":"0x00158d0003152dd8_switch_right_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0003152dd8","name":"xa_dswitch_salon","sw_version":"Zigbee2mqtt 1.2.1","model":"Aqara double key wired wall switch (QBKG03LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM MQTT publish: topic 'homeassistant/sensor/0x00158d0003152dd8/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/xa_dswitch_salon","json_attributes_topic":"zigbee2mqtt/xa_dswitch_salon","name":"xa_dswitch_salon_linkquality","unique_id":"0x00158d0003152dd8_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0003152dd8","name":"xa_dswitch_salon","sw_version":"Zigbee2mqtt 1.2.1","model":"Aqara double key wired wall switch (QBKG03LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/12/2019, 11:33:26 PM MQTT publish: topic 'homeassistant/sensor/0x00158d0002008976/click/config', payload '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","expire_after":1,"state_topic":"zigbee2mqtt/xa_button_hall","json_attributes_topic":"zigbee2mqtt/xa_button_hall","name":"xa_button_hall_click","unique_id":"0x00158d0002008976_click_zigbee [... cut ...] null,null,null,null,null,null,1,1,null,null,null,null,null,null,null,null,1,1,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,0]}}' of device 'lumi.ctrl_neutral2' (0x00158d0003152dd8)
  zigbee2mqtt:debug 3/12/2019, 11:37:22 PM Ping 0x00158d0003152dd8
  zigbee2mqtt:debug 3/12/2019, 11:37:22 PM Successfully pinged 0x00158d0003152dd8
```

Koenkk, 2019-03-13:
To really check what is going on, could you sniff the zigbee traffic on a long hold? (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

sermayoral, 2019-03-13:
I only have Windows 10 OS, and the ZBOSS page doesn't work: http://zboss.dsr-wireless.com/

Do you have an alternative? Sorry and thanks!!

Koenkk, 2019-03-14:
Unfortunately not, please try with a virtualbox linux vm.

sermayoral, 2019-03-14:
It seems that the web is working again. When I have a little time I sniff the network and report the result.

Thank you!

sermayoral, 2019-03-20:
After flashing the CC2531 with the sniffer firmware and reboot my rpi with de CC2531 plugged, zigbee2mqtt logs show this: 

```
[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-03-20T02:58:39: PM2 log: Launching in no daemon mode
2019-03-20T02:58:40: PM2 log: App [npm:0] starting in -fork mode-
2019-03-20T02:58:40: PM2 log: App [npm:0] online
> zigbee2mqtt@1.2.1 start /zigbee2mqtt-1.2.1
> node index.js
  zigbee2mqtt:info 3/20/2019, 2:58:44 AM Logging to directory: '/share/zigbee2mqtt/log/2019-03-20.02-58-44'
  zigbee2mqtt:info 3/20/2019, 2:58:47 AM Starting zigbee2mqtt version 1.2.1 (commit #unknown)
  zigbee2mqtt:info 3/20/2019, 2:58:47 AM Starting zigbee-shepherd
  zigbee2mqtt:info 3/20/2019, 2:58:48 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 3/20/2019, 2:59:48 AM Starting zigbee-shepherd
  zigbee2mqtt:error 3/20/2019, 2:59:50 AM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 3/20/2019, 2:59:50 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 3/20/2019, 2:59:50 AM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-1.2.1/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/zigbee2mqtt-1.2.1/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 3/20/2019, 2:59:50 AM Exiting...
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
```

sermayoral, 2019-03-20:
Here you are :-)

[WXKG11LM_sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/2986421/WXKG11LM_sniff.zip)

Koenkk, 2019-03-21:
As it indeed send nothing on a long press, I think this variant of the WXKG11LM doens't support long clicks.

sermayoral, 2019-03-22:
Ok @Koenkk. It has been tried. Thanks!!

closed 2019-05-28 by stale[bot]

pszeremeta86, 2019-06-17:
The problem is in a different payload: Single and Double clicks are "click", but Hold is and action. 

payload '{"linkquality":47,"battery":100,"voltage":3065,"click":"double"}'
payload '{"linkquality":23,"battery":100,"voltage":3065,"action":"hold"}'

In my case only "click sensor" was autodiscovered by Home Assistant "sensor.0x00158d0001e88543_click"

Please add "action sensor" to your configuration.yaml file manually to solve the problem.

sensor:
  - platform: "mqtt"
    name: "0x00158d0001e88543_action"
    state_topic: "zigbee2mqtt/0x00158d0001e88543"
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:toggle-switch"
    value_template: "{{ value_json.action }}"

Koenkk, 2019-06-18:
@pszeremeta86 yes, this has been fixed in #1640

Raeymeister, 2022-10-14:
@Koenkk 
I see you fixed above error for WXKG03LM in #1640.
I use device WXKG11LM, where the problem still exists (Hold is published as an Action, not as a Click)

Environment: Z2M v. 1.28.0

**MQTT output when performing HOLD**:

zigbee2mqtt/bridge/logging {"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/0x00158d000401886f', payload '{\"action\":\"hold\",\"battery\":100,\"click\":null,\"device_temperature\":27,\"linkquality\":168,\"power_outage_count\":29,\"voltage\":3005}'"}
zigbee2mqtt/0x00158d000401886f {"action":"hold","battery":100,"click":null,"device_temperature":27,"linkquality":168,"power_outage_count":29,"voltage":3005}
zigbee2mqtt/bridge/logging {"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/0x00158d000401886f', payload '{\"action\":\"\",\"battery\":100,\"click\":null,\"device_temperature\":27,\"linkquality\":168,\"power_outage_count\":29,\"voltage\":3005}'"}
zigbee2mqtt/0x00158d000401886f {"action":"","battery":100,"click":null,"device_temperature":27,"linkquality":168,"power_outage_count":29,"voltage":3005}
zigbee2mqtt/bridge/logging {"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/0x00158d000401886f/action', payload 'hold'"}
zigbee2mqtt/0x00158d000401886f/action hold

@pszeremeta86 , can you please point me to more information on the correct syntax to add your workaround to my configuration.yaml? Thanks!