Koenkk/zigbee2mqtt #1212: update to latest Dev branch - Now receiving multiple state reports

GForce2010, 2019-03-08:
Earlier today I updates to the latest Dev branch to fix an issue with group state reporting.

I now have an issue where I am receiving multiple state message after sending a bulb a command.

One example is I sent the following command
`{"state":"ON","brightness":155}`

and I received the following 7 responses
```
{"state":"ON","brightness":155,"linkquality":21}
{"state":"ON","brightness":155,"linkquality":21}
{"state":"ON","brightness":155,"linkquality":21}
{"state":"ON","brightness":155,"linkquality":21}
{"state":"ON","brightness":155,"linkquality":21}
{"state":"ON","linkquality":21}
{"state":"ON","brightness":155,"linkquality":21}
```

Sending this to another bulb
`{"state":"ON","brightness":255}`

results in these 4 messages
```
{"state":"ON","linkquality":5,"brightness":255}
{"state":"ON","linkquality":15,"brightness":254}
{"state":"ON","linkquality":18,"brightness":254}
{"state":"ON","linkquality":15}
```

Because the last message doesn't include the brightness this also messes up my control as it reports a NULL state.

Is there anything I can do to stop these extra messages?

GForce2010, 2019-03-08:
Log extract below.

```
3/8/2019, 9:04:12 PM - debug: Received MQTT message on 'zigbee2mqtt/Porch/set' with data '{"state":"ON","brightness":255}'
3/8/2019, 9:04:12 PM - info: Zigbee publish to device '0x000b57fffe9f4635', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
3/8/2019, 9:04:13 PM - info: MQTT publish: topic 'zigbee2mqtt/Porch', payload '{"state":"ON","linkquality":10,"brightness":255}'
3/8/2019, 9:04:13 PM - info: Zigbee publish to device '0x000b57fffe9f4635', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
3/8/2019, 9:04:13 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b57fffe9f4635)
3/8/2019, 9:04:13 PM - info: MQTT publish: topic 'zigbee2mqtt/Porch', payload '{"state":"ON","linkquality":10,"brightness":254}'
3/8/2019, 9:04:13 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genLevelCtrl","data":{"15":0,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b57fffe9f4635)
3/8/2019, 9:04:13 PM - info: MQTT publish: topic 'zigbee2mqtt/Porch', payload '{"state":"ON","linkquality":10,"brightness":254}'
3/8/2019, 9:04:13 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genLevelCtrl","data":{"16384":254,"65533":1}}' of device 'TRADFRI bulb E27 W opal 1000lm' (0x000b57fffe9f4635)
3/8/2019, 9:04:13 PM - info: MQTT publish: topic 'zigbee2mqtt/Porch', payload '{"state":"ON","linkquality":7}'
3/8/2019, 9:04:22 PM - debug: Ping 0x00124b001bd1250b
```

I also noticed that when I restart the Zigbee2MQTT service that I had a a state report from every device added, unfortunately these states were reported wrong. 

Looking at the logs it would appear that the bulbs in a group are being reported with a state of Off even though they are on.

Koenkk, 2019-03-09:
The latest dev branch parses more messages send from the devices (all `readRsp` weren't parsed before). So more messages are expected for device which sends these.

The issue that the `brightness` is missing should be fixed in the latest dev branch, can you confirm this?

GForce2010, 2019-03-09:
I can confirm that brightness is now listed in all the responses.

One other thing I noticed, when Zigbee2MQTT starts there is a state message sent for each device. This information is not correct or consistent. All the messages include State and Link Quality but not all of them include Brightness. Also some lights that are ON are reported with a state of Off.

Example.
I send the following to three bulbs.
`{"state":"ON","brightness":"255"}`

After a restart I receive these responses.
```
{"state":"ON","linkquality":113,"brightness":254}   //Bulb 1 Correct
{"state":"ON","linkquality":115,"brightness":254}  //Bulb 2 Correct
{"state":"OFF","linkquality":115,"brightness":140}  //Bulb 3 Incorrect Brightness.
```

Thanks again for all your hard work.

Koenkk, 2019-03-09:
@GForce2010 the values that are published are cached in `data/states.json`, the actual values are not retrieved on startup. Did you ever got an update for e.g. bulb 3 with the correct brightness?

closed 2019-03-09 by Koenkk

matlab22, 2019-04-15:
I would like to add support for LXZB-02A, but I am not able to make a pull request.
It is working fine by adding this:
`
    {
        zigbeeModel: ['LXN56-DC27LX1.1'],
        model: 'LXZB-02A',
        vendor: 'Nue / 3A',
        description: 'Smart light controller',
        extend: generic.light_onoff_brightness,
    },
`
And it can be controlled by {"state":"ON","brightness":255}

Koenkk, 2019-04-17:
@matlab22 added, thanks!