Koenkk/zigbee2mqtt #1213: Publish hardware and software version

tb-killa, 2019-03-08:
To be in a complete way as deconz is doing we should publish the hardware version and  software version of the end device too. 

It would also be a great hint of someone report bug report and use maybe some sort of newer or older hard or software version. 

Hint. With this we could also do some tricky hacks on converter themselve to maybe ship around buggy firmware if it's possible if something isn't working as expected like e.g. Reporting or others.

Koenkk, 2019-03-10:
Note to self: when implementing at this look at #1226

tb-killa, 2019-03-16:
I extend the ``bridgeconfig.js`` to show the coordinator_firmware
```
publish() {
        const topic = `bridge/config`;
        const payload = {
			coordinator_firmware: this.zigbee.shepherd.info().firmware.revision,
            log_level: logger.transports.console.level,
            permit_join: this.zigbee.getPermitJoin(),
        };

        this.mqtt.publish(topic, JSON.stringify(payload), {retain: true, qos: 0}, null);
    }
```
We should check if we want to move the ``startupLogVersion`` from controller.js to something else or extend them to return the needing values ?

Koenkk, 2019-03-16:
With these code snippets it's hard to review, could you make a work in progress PR? We can work further there.

tb-killa, 2019-03-16:
@Koenkk I build up my two [WIP] PRs.
Maybe we both could do our work to implement both ?

Koenkk, 2019-03-16:
@tb-killa good, let's continue there!

tb-killa, 2019-03-17:
Hi @Koenkk i build another PR #1274 
What does we also need to get the informations in our zigbee2mqtt output payload ?

tb-killa, 2019-03-18:
Sadly doesnt run as expected :(
```
{ type: 'EndDevice',
  ieeeAddr: '0x00158d000278e5fc',
  nwkAddr: 18863,
  manufId: 4151,
  manufName: 'LUMI',
  powerSource: 'Battery',
  modelId: 'lumi.sensor_cube',
  epList: [ 1, 2, 3 ],
  zclVersion: undefined,
  appVersion: undefined,
  stackVersion: undefined,
  hwVersion: undefined,
  swBuildId: undefined,
  status: 'online',
  joinTime: 1552925179 }
```
The needing Informations doesnt return.
We need to figure out what doesnt work.

tb-killa, 2019-03-18:
Okay i modify the output for ``include_device_information`` as show in the PR.
Now we got the the Output like this:
```
"device":{"ieeeAddr":"0x00158d000252XXX","friendlyName":"WXKG03LM_1","type":"EndDevice","nwkAddr":550XX,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_86sw1\u0000lu","hwVersion":"unknown","swBuildId":"unknown","status":"online"}
```
We need to check why not every device is publishing the basic cluster informations?
Maybe we doesnt define to configure the bind to them ?

tb-killa, 2019-03-19:
@Koenkk we should globally define report of basic cluster with the both informations.
Sadly not every device support per basic the data as we need. 

Would you please test with your own infrastructures as we could do the integration after that results.

Koenkk, 2019-03-19:
@tb-killa can you make a pr for this so I can test?

tb-killa, 2019-03-22:
@Koenkk do some tests via ``deviceConfigure.js`` with basic cluster and the needing attributes but this doesn´t work.

Strange would be that not every device support this by themselve, as if i expect.

this is a sample of the ``ecozy``

``
"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"eCozy","modelId":"Thermostat","dateCode":"20160826eCozy428","powerSource":3,"locationDesc":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","deviceEnabled":1,"disableLocalConfig":0}}
``

this of the ``bitron``

``
"genBasic":{"dir":{"value":3},"attrs":{"zclVersion":1,"appVersion":17,"hwVersion":0,"manufacturerName":"Bitron Home","modelId":"902010/22","dateCode":"20150216        ","powerSource":3,"locationDesc":"                ","physicalEnv":0,"deviceEnabled":1}}
``

this of the ``osram`` 4xswitch

``
"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Switch 4x EU-LIGHTIFY","dateCode":"201707051445390","powerSource":3,"swBuildId":"e.1.0.0M"}}
``

It seems that ikea devices doesnt return this value per default. As i read at another source we need here define specific cluster reporting only for vendor ikea. 
Do you know how we could do this ?

Sadly non of the xiaomi devices seems to return this standard values too.
@kirovilya Do you could help ??

As i know from deconz there this is working, maybe we forgot something ?

Koenkk, 2019-03-22:
For xiaomi, I have `"dateCode":"2018-09-26"` in my `database.db`. I think every device just uses different values for this.

tb-killa, 2019-03-24:
3.2.2.2.8 DateCode Attribute 
The DateCode attribute is a ZigBee character string with a maximum length of 16
bytes . The first 8 characters specify the date of manufacturer of the device in
international date notation according to ISO 8601, i.e. YYYYMMDD, e.g.
20060814.

tb-killa, 2019-03-24:
What about do some ``foundation genbasic read`` of the attributes by all devices and push them back to the database?

Koenkk, 2019-03-24:
Does this work for devices that dont initially support it?

tb-killa, 2019-03-24:
As with everything who doesn't follow the standards... I don't know ;)
Maybe what we could check too if we extend our zcl-id and zcl-packet with the genOta fields because there should be the version of sw and hardware published because of the update functionality.

tb-killa, 2019-03-24:
> For xiaomi, I have `"dateCode":"2018-09-26"` in my `database.db`. I think every device just uses different values for this.

Non of my xiaomi devices got this value inside the database file. 
What type of device does this use?

Koenkk, 2019-03-26:
@tb-killa I was mistaken, it was the `lumi.router` (CC2530 router)

tb-killa, 2019-03-26:
@Koenkk yes I see this entry too.
Sadly it seems to be that actually the xiaomi devices doesn't return the value inside the basic cluster. 

Maybe there is some failure inside the xiaomi structure as we doesn't see them?
Or they need some special way of report. 
We need someone who could sniff with original gateway. 

From other point of view I would prefer the include of the genOta fields and parse this informations, maybe xiaomi follow them.?

Koenkk, 2019-03-27:
Xiaomi devices don't seem to support `genOta`, you can easily check this by putting this in the `onReady()` in `zigbee.js`:

```js
const ep = this.getEndpoint('0x00158d00018255df');
        ep.foundation('genOta', 'read', [ { attrId: 0 }, { attrId: 1 }, {attrId: 2}, {attrId: 3}, {attrId: 8} ], function (err, rsp) {
            if (!err)
                console.log(rsp);
        });
```

However, I've found out that they **do** support the `dateCode`, will add this.

Koenkk, 2019-03-27:
Added! I think we now retrieve and publish all version related attributes that devices have, can this be closed?

tb-killa, 2019-03-27:
I think you forget the ``getDeviceInfoForMqtt`` output.?
Or may I doesn't see that correctly?

Koenkk, 2019-03-28:
Added! Assuming this is done now, feel free to re-open if needed.

closed 2019-03-28 by Koenkk

tb-killa, 2019-03-31:
@Koenkk We need some tweak for the ``dateCode`` Value.

My Devices reporting this value in strange Output, for e.g.
```
"dateCode":"20150216        "
"dateCode":"201707051445390"
"dateCode":"2018-09-26"
"dateCode":"20160826eCozy428"
```

The output should be transferred to a uniform state and then output.
Maybe the third ones seems to be the best one.

Koenkk, 2019-04-01:
How to do this? We don't have any influence on what values are reported, how to e.g. make `20160826eCozy428` uniform?

tb-killa, 2019-04-02:
@Koenkk 
We could do something like this (Prototype to show) :
```
var test = '20150216        ';

if (!(/(\d{4})-(\d{2})-(\d{2})/.test(test))) {
var t = test.slice(0,8).replace(/(\d{4})(\d{2})(\d{2})/, '$1-$2-$3');
alert(t);
}
```

With this and the input from above i get the needing result like ``20160826eCozy428`` to ``2016-08-26``

Please check and maybe we could build it in correct way ?

Koenkk, 2019-04-03:
If that works with all the cases from https://github.com/Koenkk/zigbee2mqtt/issues/1213#issuecomment-478374165 I'm ok with that.