Koenkk/zigbee2mqtt #1214: Implement time cluster

tb-killa, 2019-03-08:
To get more complete integration we should build the possibilitie to get our end devices become the actually time. 

This would help by radiator or other programmable devices to be in the right time with the stuff around like zone, week and much more. 

Maybe some sort of zapp extension?

milakov, 2019-03-08:
Interesting. While sniffing zigbee ttraffic I noticed some routers were regukarky sending "Read Attributes" packets to the coordinator specifyinf Time attribute:

[image]

The coordinator never responds.

tb-killa, 2019-03-09:
Yes the same I see on different xiaomi end devices too.

tb-killa, 2019-03-10:
@Koenkk Found a implementation where we could do some check of how this could work:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/time.cpp

Koenkk, 2019-03-11:
The first thing to do here is to make sure these time messages make it into zigbee2mqtt. Do these already appear in the log as no converter available? If not, can you provide logging (with `DEBUG=*`)?

tb-killa, 2019-03-11:
You see them not as converter messages. Only if you use a zigbee sniffer.

milakov, 2019-03-11:
@Koenkk Not sure how to do that with hass.io installation...

Koenkk, 2019-03-11:
@tb-killa can you post the debug logging then?

tb-killa, 2019-03-12:
@Koenkk Will provide if i got the time.

PS: We need the Time Cluster (Simple Answer to the ZCL Read Request) because this knowing devices need them:

- lumi.ctrl_neutral` switches (Send Periodic Read Attr Command to Coordinator to check if they are connected)
- Thermostat devices who need daylight saving time and general the actually time to work autonom if lost connection to coordinator.

itsmepetrov, 2019-04-03:
Request from aqara wall switches:

```js
{
  sof: 254,
  len: 25,
  type: 'AREQ',
  subsys: 'AF',
  cmd: 'incomingMsg',
  payload: {
    groupid: 0,
    clusterid: 10,
    srcaddr: 45736,
    srcendpoint: 1,
    dstendpoint: 1,
    wasbroadcast: 0,
    linkquality: 78,
    securityuse: 0,
    timestamp: 1805695,
    transseqnumber: 0,
    len: 5,
    data: <Buffer 10 44 00 00 00>
  },
  fcs: 205,
  csum: 205
}
```

tb-killa, 2019-04-07:
WIP with some basic stuff in the near of future: https://github.com/tb-killa/zigbee2mqtt/tree/%231214

antst, 2019-04-16:
What is the status?
Falling out aqara switches kill me (

antst, 2019-04-16:
BTW, I see some commits related to time cluster in your repo. Do you need some testing?

itsmepetrov, 2019-04-16:
There is a problem with aqara switches and time cluster in shepherd: the cluster mounts on new endpoint (not first), but aqara devices doesn't support binding attributes on specific endpoint so they use first endpoint for everything, in this case we need to implement some mechanism to redirect such requests from first endpoint to endpoint with time cluster (internally). I've tried to hardcode redirection for every time cluster request to the new endpoint and it worked, but it's not the best solution.

antst, 2019-04-16:
Is your solution available?
Its maybe not the best, but bunch of not-working switches is way worse )))

Koenkk, 2019-04-24:
Instead of doing the hacky forwarding of messages, we could directly add this code in zigbee-shepherd. 

- Ziee app for the default endpoints is created here: https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/initializers/init_controller.js#L121
- A tutorial how to add a cluster can be found here: https://github.com/zigbeer/ziee/wiki#user-content-3-initialization-of-clusters-step-by-step
- @itsmepetrov already has prepared some code here: https://github.com/tb-killa/zigbee2mqtt/commit/06464a94cf27407b0f36423c77e2ad1a0110f43e#commitcomment-33194120

Koenkk, 2019-04-26:
Implemented!

tb-killa, 2019-07-05:
Hi @Koenkk we need to reopen this issue and work on your implementation because it doesn't seems to work correctly.

Devices requesting for the TimeStatus Bits on first and need an answer for this.

After that the time themselves are requested.

tb-killa, 2019-07-05:
Thanks.
I do some checkup and this are requested by my device Bitron ``AV2010/32`` on Startup.

For the TimeStatus I do some research and found the needing Bits.
I added for everything the Comments to be clear what this is meaning.

But I need help to get the right calculations for the other bits to get them complete functional.

```
if (cluster === 'genTime' && attribute.key === 'timeStatus') {
		/*
		* timestatus is a mandatory 8-bit attribute containing the following bitmap:
		*
		* Bits 0 - Master -  1: Time-master for network
		*					 0: Not time-master for network
		*
		* Bits 1 - Synchronised -  1: Synchronised to another device
		*						   0: Not synchronised to another device
		*
		* Bits 2 - Master for TimeZone and DST * - 1: Master for time-zone and DST
		*										   0: Not master for time-zone and DST
		*
		* Bits 3-7 - Reserved
		*/
		const timeStatusBits = '0x0D'; //sets bit#0 (Master) AND/or bit#3 (Superseding) AND sets bit #2 (Master Zone Dst)
		response.push(this.createReadResponseRec(clusterID, attribute.value, timeStatusBits));
}
if (cluster === 'genTime' && attribute.key === 'time') {
		/*
		* The time attribute is referenced to UTC (Co-ordinated Universal Time) and based on the type
		* UTCTime, which is defined in the ZigBee standard as:
		*
		* UTCTime is an unsigned 32 bit value representing the number of seconds since 0 hours, 
		* 0 minutes, 0 seconds, on the 1st of January, 2000 UTC
		*
		*/
		const time = Math.round(((new Date()).getTime() - OneJanuary2000) / 1000);
		response.push(this.createReadResponseRec(clusterID, attribute.value, time));
}
if (cluster === 'genTime' && attribute.key === 'timeZone') {
		/*
		* TimeZone is an optional attribute which indicates the local time-zone
		* expressed as an offset from UTC, in seconds. 
		*
		*/
}
if (cluster === 'genTime' && attribute.key === 'dstStart') {
		/*
		* DstStart is an optional attribute which contains the start-time (UTC),
		* in seconds, for daylight saving for the current year.
		*
		* If DstStart is used then DstEnd and DstShift are also required!
		*
		*
		*/
}
if (cluster === 'genTime' && attribute.key === 'dstEnd') {
		/*
		* DstEnd is an optional attribute which contains the end-time (UTC),
		* in seconds, for daylight saving for the current year.
		*
		* If DstEnd is used then DstStart and DstShift are also required!
		*
		*
		*/
}
if (cluster === 'genTime' && attribute.key === 'dstShift') {
		/*
		* DstShift is an optional attribute which contains the local time-shift,
		* in seconds, relative to standard local time that is applied during the
		* daylight saving period.
		*
		* If DstShift is used then DstStart and DstEnd are also required!
		*
		*
		*/
}
if (cluster === 'genTime' && attribute.key === 'standardTime') {
		/*
		* StandardTime is an optional attribute which contains the local standard time
		* (equal to time + timeZone).
		*
		*
		*/
}
if (cluster === 'genTime' && attribute.key === 'localTime') {
		/*
		* LocalTime is an optional attribute  which contains the local time taking into
		* account daylight saving, if applicable (equal to time + timezone +
		* dstshift during the daylight saving period and equal to standardtime
		* outside of the daylight saving period).
		*
		*
		*/
}
if (cluster === 'genTime' && attribute.key === 'lastSetTime') {
		/*
		* LastSetTime is an optional attribute which indicates the most recent UTC time
		* at which the Time attribute (time) was set, either internally or over the ZigBee
		* network. 
		*
		*
		*/
}
if (cluster === 'genTime' && attribute.key === 'validUntilTime') {
		/*
		* ValidUntilTime is an optional attribute and indicates a UTC time (later than
		* LastSetTime) up to which the Time attribute (Time) value may be trusted.
		*
		*
		*/
}
```

closed 2019-09-10 by stale[bot]