Koenkk/zigbee2mqtt #1216: E1743 Switch : No click information in the payload

cedricous, 2019-03-09:
The E1743 is correclty paired but i receive only battery and linkquality on the log:
zigbee2mqtt:info 3/8/2019, 4:56:29 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2f1753', payload '{"battery":60,"linkquality":120}'

I use the edge version of the zigbee2mqtt on the Hass.io, put the parameter "zigbee_shepherd_devices": true, and put the devices.js from the zigbee-shepherd-converters github in the share/zigbee2mqtt directory.

Maybe I forgot something ?

Here the log of the binding :
zigbee2mqtt:info 3/8/2019, 4:54:57 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
zigbee2mqtt:info 3/8/2019, 4:55:32 PM New device 'undefined' with address 0x000d6ffffe2f1753 connected!
zigbee2mqtt:info 3/8/2019, 4:55:32 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6ffffe2f1753","meta":{}}'
zigbee2mqtt:warn 3/8/2019, 4:55:32 PM Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn 3/8/2019, 4:55:32 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:warn 3/8/2019, 4:55:33 PM Failed to configure 0x000d6ffffe2ce7ca (0x000d6ffffe2ce7ca) ('Error: Timed out after 10000 ms') (attempt #1)
zigbee2mqtt:warn 3/8/2019, 4:55:33 PM This can be ignored if the device is working properly
zigbee2mqtt:info 3/8/2019, 4:55:35 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2f1753/click/config', payload '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","expire_after":1,"state_topic":"zigbee2mqtt/0x000d6ffffe2f1753","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753_click","unique_id":"0x000d6ffffe2f1753_click_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info 3/8/2019, 4:55:35 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2f1753/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x000d6ffffe2f1753","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753_battery","unique_id":"0x000d6ffffe2f1753_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info 3/8/2019, 4:55:35 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2f1753/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6ffffe2f1753","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753_linkquality","unique_id":"0x000d6ffffe2f1753_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info 3/8/2019, 4:55:45 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2f1753', payload '{"battery":100,"linkquality":128}'
zigbee2mqtt:warn 3/8/2019, 4:56:01 PM Failed to configure 0x000d6ffffe2f1753 (0x000d6ffffe2f1753) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
zigbee2mqtt:warn 3/8/2019, 4:56:01 PM This can be ignored if the device is working properly
zigbee2mqtt:info 3/8/2019, 4:56:05 PM Device incoming...
zigbee2mqtt:info 3/8/2019, 4:56:05 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
zigbee2mqtt:info 3/8/2019, 4:56:12 PM Successfully configured 0x000d6ffffe2f1753 (0x000d6ffffe2f1753)
zigbee2mqtt:info 3/8/2019, 4:56:29 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2f1753', payload '{"battery":60,"linkquality":120}'

Koenkk, 2019-03-09:
With the edge version, you don't need a separate `devices.js`, edge already includes support for this device.

Can you post your `debug` log when pressing the button on this switch?

cedricous, 2019-03-10:
First many Thx for your help,
So I put this parameter on the file share/zigbee2mqtt/configuration.yaml
advanced:
  log_level: debug

Here is the log when I pressing the button :

  zigbee2mqtt:debug 3/10/2019, 7:43:26 AM Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":87}}' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)
  zigbee2mqtt:info 3/10/2019, 7:43:26 AM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2ce7ca', payload '{"battery":87,"linkquality":70}'
  zigbee2mqtt:debug 3/10/2019, 7:43:26 AM Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":87}}' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)

Koenkk, 2019-03-10:
Something strange is going on. Can you try to

- Stop zigbee2mqtt
- Remove all entries of `0x000d6ffffe2ce7ca` from `database.db`
- Start zigbee2mqtt
- Re-pair device.

cedricous, 2019-03-10:
So, I removed the IKEA controler outlet and the 0x000d6ffffe2ce7ca from database.db (I don't know why but if the IKEA controler outlet is paired it's impossible to paired other think but it's another topic)

After repair the Ikea switche I have the same problem, here the result :

 zigbee2mqtt:debug 3/10/2019, 3:41:21 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI on/off switch","powerSource":3}}' of device 'undefined' (0x000d6ffffe2ce7ca)
  zigbee2mqtt:warn 3/10/2019, 3:41:21 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 3/10/2019, 3:41:21 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 3/10/2019, 3:41:24 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:25 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:30 PM Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)
  zigbee2mqtt:debug 3/10/2019, 3:41:30 PM Configuring 0x000d6ffffe2ce7ca (0x000d6ffffe2ce7ca) ...
  zigbee2mqtt:info 3/10/2019, 3:41:30 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2ce7ca/click/config', payload '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","expire_after":1,"state_topic":"zigbee2mqtt/0x000d6ffffe2ce7ca","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2ce7ca","name":"0x000d6ffffe2ce7ca_click","unique_id":"0x000d6ffffe2ce7ca_click_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2ce7ca","name":"0x000d6ffffe2ce7ca","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/10/2019, 3:41:30 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2ce7ca/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x000d6ffffe2ce7ca","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2ce7ca","name":"0x000d6ffffe2ce7ca_battery","unique_id":"0x000d6ffffe2ce7ca_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2ce7ca","name":"0x000d6ffffe2ce7ca","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/10/2019, 3:41:30 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2ce7ca/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6ffffe2ce7ca","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2ce7ca","name":"0x000d6ffffe2ce7ca_linkquality","unique_id":"0x000d6ffffe2ce7ca_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2ce7ca","name":"0x000d6ffffe2ce7ca","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:debug 3/10/2019, 3:41:30 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:32 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:35 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:36 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:37 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:40 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:44 PM Received zigbee message of type 'readRsp' with data '{"cid":"closuresWindowCovering","data":{"windowCoveringType":0,"configStatus":3,"installedOpenLimitLiftCm":0,"installedClosedLimitLiftCm":65535,"installedOpenLimitTiltDdegree":0}}' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)
  zigbee2mqtt:debug 3/10/2019, 3:41:47 PM Received zigbee message of type 'readRsp' with data '{"cid":"closuresWindowCovering","data":{"19":65535,"65533":1,"windowCoveringMode":20}}' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)
  zigbee2mqtt:debug 3/10/2019, 3:41:47 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:48 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:49 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe2ce7ca"'
  zigbee2mqtt:debug 3/10/2019, 3:41:49 PM Received zigbee message of type 'devIncoming' with data '"0x000d6ffffe2ce7ca"' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)
  zigbee2mqtt:info 3/10/2019, 3:41:49 PM Device incoming...
  zigbee2mqtt:info 3/10/2019, 3:41:49 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 3/10/2019, 3:41:49 PM Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)
[... cut ...]
  zigbee2mqtt:debug 3/10/2019, 3:42:03 PM Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":60}}' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)
  zigbee2mqtt:info 3/10/2019, 3:42:03 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2ce7ca', payload '{"battery":60,"linkquality":70}'
  zigbee2mqtt:debug 3/10/2019, 3:42:03 PM Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":60}}' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)

booxdk, 2019-03-10:
Hi. I'm having the same problem with the E1743 running on Ubuntu, only receiving payload '{"battery":34,"linkquality":0}'. I've updates the converters using yarn add zigbee-shepherd-converters and removed old entries from the database with no luck - anyone actually got this one working ?

tynor88, 2019-03-10:
I have it working just fine with latest firmware.

booxdk, 2019-03-11:
> I have it working just fine with latest firmware.

Thank you for the feedback. It makes me wonder if the Tradfri firmware version on the of/off remote could be an issue ? - Mine has never been connected to the Ikea hub so fw could be any version - I'll try to connect it to the Ikea hub and update - do you have any idea of what fw thats in your remote ?

tynor88, 2019-03-11:
Mine was never connected to the IKEA hub either. However I was referring to
the latest zigbee2mqtt dev firmware.

I'm not sure how to check the firmware of the device itself without pairing
it with the IKEA hub?

cedricous, 2019-03-11:
I have the last mqtt firmware and I have this issue, for me you cannot update the firmware of the device himself.

Koenkk, 2019-03-11:
Could you do some zigbee sniffing of the button clicks to find what is going on? http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

tynor88, 2019-03-11:
My logging looks like this:

On Click:
> zigbee2mqtt:info 3/11/2019, 8:03:47 PM MQTT publish: topic 'zigbee2mqtt/IKEA On Off Switch 1', payload '{"battery":74,"linkquality":68,"click":"on"}'

Off Click:

> zigbee2mqtt:info 3/11/2019, 8:02:41 PM MQTT publish: topic 'zigbee2mqtt/IKEA On Off Switch 1', payload '{"battery":74,"linkquality":68,"click":"off"}'

And the E1603 responds nicely (not sure why it publishes 3 times though, but it works):

>  zigbee2mqtt:info 3/11/2019, 8:06:20 PM Zigbee publish to device '0x000d6ffffec7d6e2', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 3/11/2019, 8:06:20 PM MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 1', payload '{"state":"OFF","linkquality":70}'
  zigbee2mqtt:info 3/11/2019, 8:06:20 PM MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 1', payload '{"state":"OFF","linkquality":70}'
  zigbee2mqtt:info 3/11/2019, 8:06:20 PM MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 1', payload '{"state":"OFF","linkquality":70}'

cedricous, 2019-03-14:
Here the log :
In Zigbee2mqtt the issue is steal there

  zigbee2mqtt:debug 3/14/2019, 6:25:45 PM Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":87}}' of device 'TRADFRI on/off switch' (0x000d6ffffe2ce7ca)
  zigbee2mqtt:info 3/14/2019, 6:25:45 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2ce7ca', payload '{"battery":87,"linkquality":86}'

In WireShark here the log :
[image]

cedricous, 2019-03-14:
The screen shot of the first panel of Wireshark, maybe it will help you : 
[image]

Koenkk, 2019-03-15:
Can you post your zigbee2mqtt startup log?

cedricous, 2019-03-15:
Here the log of the zigbee2mqtt startup : 

/app/run.sh: line 10:    10 Segmentation fault      (core dumped) python3 set_config.py "$CONFIG_PATH" "$DATA_PATH"
2019-03-15T16:46:05: PM2 log: Launching in no daemon mode
2019-03-15T16:46:05: PM2 log: App [npm:0] starting in -fork mode-
2019-03-15T16:46:05: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /app
> node index.js
  zigbee2mqtt:info 3/15/2019, 4:46:10 PM Logging to directory: '/share/zigbee2mqtt/log/2019-03-15.16-46-10'
  zigbee2mqtt:debug 3/15/2019, 4:46:10 PM Removing old log directory '/share/zigbee2mqtt/log/2019-03-08.06-37-28'
  zigbee2mqtt:debug 3/15/2019, 4:46:12 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 3/15/2019, 4:46:12 PM Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:debug 3/15/2019, 4:46:12 PM Saving state to file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 3/15/2019, 4:46:12 PM Starting zigbee2mqtt version 1.1.1 (commit #unknown)
  zigbee2mqtt:info 3/15/2019, 4:46:12 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM zigbee-shepherd started
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM Coordinator firmware version: '20190109'
  zigbee2mqtt:debug 3/15/2019, 4:46:13 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ecdfac","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190109},"startTime":1552668373,"joinTimeLeft":0}
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM Currently 5 devices are joined:
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM 0x00158d000238abfc (0x00158d000238abfc): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM 0x00158d0002a4e69e (0x00158d0002a4e69e): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM 0x000d6ffffe2f1753 (0x000d6ffffe2f1753): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM 0x00158d0002b430cf (0x00158d0002b430cf): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM 0x000d6ffffe2ce7ca (0x000d6ffffe2ce7ca): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:debug 3/15/2019, 4:46:13 PM Configuring 0x000d6ffffe2f1753 (0x000d6ffffe2f1753) ...
  zigbee2mqtt:info 3/15/2019, 4:46:13 PM zigbee-shepherd ready
  zigbee2mqtt:debug 3/15/2019, 4:46:13 PM Mounted the cieApp (epId 11)
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM Connected to MQTT server
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000238abfc', payload '{"battery":100,"voltage":3045,"linkquality":107}'
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002a4e69e', payload '{"battery":100,"voltage":3035,"linkquality":184,"unknown_data":1376256,"angle_x":0,"angle_y":4,"angle_z":86,"angle_x_absolute":90,"angle_y_absolute":86}'
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2f1753', payload '{"battery":87,"linkquality":97}'
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002b430cf', payload '{"battery":100,"voltage":3025,"linkquality":97,"illuminance":12,"occupancy":false}'
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2ce7ca', payload '{"battery":87,"linkquality":92}'
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":false}'
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'homeassistant/sensor/0x00158d000238abfc/click/config', payload '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","expire_after":1,"state_topic":"zigbee2mqtt/0x00158d000238abfc","json_attributes_topic":"zigbee2mqtt/0x00158d000238abfc","name":"0x00158d000238abfc_click","unique_id":"0x00158d000238abfc_click_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d000238abfc","name":"0x00158d000238abfc","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara double key wireless wall switch (WXKG02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'homeassistant/sensor/0x00158d000238abfc/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d000238abfc","json_attributes_topic":"zigbee2mqtt/0x00158d000238abfc","name":"0x00158d000238abfc_battery","unique_id":"0x00158d000238abfc_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d000238abfc","name":"0x00158d000238abfc","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara double key wireless wall switch (WXKG02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[... cut ...]
  zigbee2mqtt:info 3/15/2019, 4:46:15 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2ce7ca/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6ffffe2ce7ca","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2ce7ca","name":"0x000d6ffffe2ce7ca_linkquality","unique_id":"0x000d6ffffe2ce7ca_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2ce7ca","name":"0x000d6ffffe2ce7ca","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'

Koenkk, 2019-03-15:
You are running an old firmware, updating to `20190223` solves this issue. (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.2.0)

cedricous, 2019-03-16:
Awsome it works perfectly :-), it was the firmware, i'm really happy thx you Koenkk. 
I have a last question, I saw that the last version of zigbee2mqtt is 1.2, I use Hass.io with the edge version with the automatic update. And when I start zigbee2mqtt in the log it's indicate 1.1, do you know why the version is not updated automaticaly ? How can I do to have the last version (remove, and reinstall ?). Thx for your help.

/app/run.sh: line 10:    10 Segmentation fault      (core dumped) python3 set_config.py "$CONFIG_PATH" "$DATA_PATH"
2019-03-16T06:24:01: PM2 log: Launching in no daemon mode
2019-03-16T06:24:02: PM2 log: App [npm:0] starting in -fork mode-
2019-03-16T06:24:02: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /app
> node index.js
  zigbee2mqtt:info 3/16/2019, 6:24:06 AM Logging to directory: '/share/zigbee2mqtt/log/2019-03-16.06-24-06'
  zigbee2mqtt:debug 3/16/2019, 6:24:06 AM Removing old log directory '/share/zigbee2mqtt/log/2019-03-08.16-43-11'
  zigbee2mqtt:debug 3/16/2019, 6:24:07 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 3/16/2019, 6:24:07 AM Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:debug 3/16/2019, 6:24:07 AM Saving state to file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 3/16/2019, 6:24:07 AM Starting zigbee2mqtt version 1.1.1 (commit #unknown)
  zigbee2mqtt:info 3/16/2019, 6:24:07 AM Starting zigbee-shepherd
  zigbee2mqtt:info 3/16/2019, 6:24:08 AM zigbee-shepherd started
  zigbee2mqtt:info 3/16/2019, 6:24:08 AM Coordinator firmware version: '20190223'

Koenkk, 2019-03-16:
1.2.1 is available for hassio, if you somehow don't see the update, please ask it here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues

closed 2019-03-16 by Koenkk