Koenkk/zigbee2mqtt #1219: Cannot read property '100' of undefined

jeromelaban, 2019-03-09:
Using release 2.1.1, when doing some unknown operation (most probably a door sensor), zigbee2mqtt exits:

```
/app/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js:598
            return {contact: msg.data.data['65281']['100'] === 0};
                                                   ^

TypeError: Cannot read property '100' of undefined
    at Object.convert (/app/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js:598:52)
    at converters.forEach (/app/lib/extension/deviceReceive.js:157:41)
    at Array.forEach (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/app/lib/extension/deviceReceive.js:155:20)
    at extensions.filter.forEach (/app/lib/controller.js:139:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/app/lib/controller.js:139:14)
    at Zigbee.onMessage (/app/lib/zigbee.js:185:18)
    at ZShepherd.emit (events.js:182:13)
    at ZShepherd.<anonymous> (/app/node_modules/zigbee-shepherd/lib/shepherd.js:192:14)
```

Koenkk, 2019-03-09:
Can you post the debug log moments before this crash?

jeromelaban, 2019-03-10:
I spent a few hours trying to get it but could not. I'm wondering if it has to do with the fact that I used the new `zigbee2mqtt/bridge/config/reset` topic.

By the way, one of the reason I had to use it is because I was trying to re-pair a device, but I was not getting any message. Would there be a way (without having to resort to debug) to log when a device is trying to re-join ? I had to re-join because I changed the panID (before your today's new firware), but some devices did not rejoin automatically).

Koenkk, 2019-03-10:
- This crash shouldn't happen any (added an extra check).
- This is only possible when emptying `database.db` before doing this.

Should we close this?

jeromelaban, 2019-03-10:
It's also what I did. I emptied the database to try and change the encryption key. Ended up reverting as it's way too time consuming to re-pair devices (especially xiaomi ones, which don't respond to device description requests easily...)

closed 2019-03-10 by jeromelaban