Koenkk/zigbee2mqtt #1223: Pairing RTCGQ11LM: adding modelId to database fixes sensor, but why?

peterforeman, 2019-03-10:
Hi,

I'm unsuccessful in pairing my RTCGQ11LM sensor for days now. Trying over and over, the message stays "undefined device".

Please help me understand why, although the "attReport" sends the modelID, this does not get picked up and the device stays "undefined". Adding some info manually to the database.db gets the sensor working.

```
zigbee2mqtt:debug 3/10/2019, 10:01:44 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_motion.aq2"}}' of device 'undefined' (0x00158d0001dda9cd)
  zigbee2mqtt:warn 3/10/2019, 10:01:44 AM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 3/10/2019, 10:01:44 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

How I fixed this is as follows. I have another sensor which has been paired without problems. So I checked the database.db for the config of that device:

```
$ grep "0x00158d0001e16511" database.db
{"id":21,"type":"EndDevice","ieeeAddr":"0x00158d0001e16511","nwkAddr":57837,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"4jdf9U4dcu1yaWl5"}
{"id":21,"type":"EndDevice","ieeeAddr":"0x00158d0001e16511","nwkAddr":57837,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"4jdf9U4dcu1yaWl5"}
```

This is the config for the "undefined" sensor:

```
$ grep "0x00158d0001dda9cd" database.db
{"id":25,"type":"EndDevice","ieeeAddr":"0x00158d0001dda9cd","nwkAddr":16467,"manufId":4151,"epList":[1],"status":"online","joinTime":1552208270,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{}}},"_id":"ccdG66dIVV68ery9"}
{"id":25,"type":"EndDevice","ieeeAddr":"0x00158d0001dda9cd","nwkAddr":16467,"manufId":4151,"epList":[1],"status":"online","joinTime":1552208270,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"ccdG66dIVV68ery9"}
```

As you can see the working sensor has some extra manufName and modelId data. I then edit my database.db to add the following snippet to every line of the 'undefined' sensor:

```
"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2"
```

Result:
```
$ grep "0x00158d0001dda9cd" database.db
{"id":25,"type":"EndDevice","ieeeAddr":"0x00158d0001dda9cd","nwkAddr":16467,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"status":"online","joinTime":1552208270,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"ccdG66dIVV68ery9"}
{"id":25,"type":"EndDevice","ieeeAddr":"0x00158d0001dda9cd","nwkAddr":16467,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"ccdG66dIVV68ery9"}
```

Restarting zigbee2mqtt now results in a working sensor. Yay. But why isn't the pairing working in the first place? And why is it so easy to get it working by just copying some data?

Koenkk, 2019-03-10:
This means that the pairing procedure hasn't fully been completed.

Can you try to:

- Stop zigbee2mqtt
- Remove all entries of `0x00158d0001dda9cd ` from `database.db`
- Start zigbee2mqtt
- Re-pair device following http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#xiaomi

peterforeman, 2019-03-10:
Should've noted that I tried that as well, but no success. The device is working fine now, even though the pairing procedure apparently did not complete. If it's this easy to "fix" this, perhaps we can get something done to help the pairing process succeed even if the procedure isn't complete?

lolorc, 2019-03-10:
this happened to me a lot of times, editing database.db has always been enough to fix the issue.
I'd say it's not an uncompleted pairing but an issue with zigbee-sheperd

Koenkk, 2019-03-11:
From a zigbee perspective the pairing has been completed (the RTCGQ11LM) got the network key, otherwise it couldn't communicate.

For the zigbee-shepherd perspective it did also pair! But it didn't retrieve the modelID which is put in the `database.db`. Note that zigbee-shepherd uses the device descriptor request for this, this is a different thing than the `genBasic` `attReport` that you see. 

Perhaps zigbee-shepherd could be updated to update the modelID once it sees such a message.

bremy, 2019-03-13:
I can confirm that adding manually the following snippet as described here above is working like a charm

`"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2"
`

I installed yesterday a fresh config on my RPI2 using the lasted zigbee2mqtt master branch and that was impossible to get my RTCGQ11LM working. BTW all other sensors were perfectly paired (Aqara Temp/humidity/pressure, Aqara door contact, IKEA control outlet, IKEA on/off switch)

So seems that RTCGQ11LM remains really difficult to pair

suprimex, 2019-03-14:
I suppose I have very similar problem with just arrived Xiaomi MCCGQ01LM. 
Sure, I read "how to pair.." already,   and saw video on youtube. 
Tried gadzillion times to pair, every time getting as such:
`
3/14/2019, 5:50:36 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
3/14/2019, 5:50:36 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
3/14/2019, 5:51:40 PM - info: New device 'undefined' with address 0x00158d0002b82218 connected!
3/14/2019, 5:51:40 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002b82218","meta":{}}'
3/14/2019, 5:51:40 PM - warn: Device with modelID 'undefined' is not supported.
3/14/2019, 5:51:40 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
3/14/2019, 5:51:41 PM - warn: Device with modelID 'undefined' is not supported.
3/14/2019, 5:51:41 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
3/14/2019, 5:52:04 PM - warn: Message without device!
3/14/2019, 5:52:06 PM - info: Device incoming...
3/14/2019, 5:52:06 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
3/14/2019, 5:52:41 PM - warn: Device with modelID 'undefined' is not supported.
3/14/2019, 5:52:41 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
3/14/2019, 5:52:42 PM - warn: Device with modelID 'undefined' is not supported.
3/14/2019, 5:52:42 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
3/14/2019, 5:52:53 PM - warn: Message without device!
3/14/2019, 5:52:57 PM - info: Device incoming...
3/14/2019, 5:52:57 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
3/14/2019, 5:53:11 PM - warn: Device with modelID 'undefined' is not supported.
3/14/2019, 5:53:11 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
3/14/2019, 5:53:12 PM - warn: Device with modelID 'undefined' is not supported.
3/14/2019, 5:53:12 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
3/14/2019, 5:53:28 PM - warn: Message without device!
3/14/2019, 5:53:31 PM - info: Device incoming...
3/14/2019, 5:53:31 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
3/14/2019, 5:54:02 PM - error: Cannot get the Node Descriptor of the Device: 0x00158d0002b82218 (Error: Timed out after 10000 ms)
`

grep:
`pi@raspberrypi:/opt/zigbee2mqtt $ grep "0x00158d0002b82218" ./data/database.db
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b82218","nwkAddr":60482,"manufId":4151,"epList":[1],"status":"online","joinTime":1552578698,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{}}},"_id":"DO8c788m4NYMfmFf"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b82218","nwkAddr":60482,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"online","joinTime":1552578698,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{}}},"_id":"DO8c788m4NYMfmFf"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b82218","nwkAddr":22639,"manufId":4151,"epList":[1],"status":"online","joinTime":1552578757,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{}}},"_id":"AlJbvKkSYyicPnU4"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b82218","nwkAddr":22639,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"online","joinTime":1552578757,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{}}},"_id":"AlJbvKkSYyicPnU4"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b82218","nwkAddr":50833,"manufId":4151,"epList":[1],"status":"online","joinTime":1552578788,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{}}},"_id":"YUjvIg1lufODlysT"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b82218","nwkAddr":50833,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"online","joinTime":1552578788,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{}}},"_id":"YUjvIg1lufODlysT"}
`

peterforeman, 2019-03-16:
@Koenkk but, zigbee-shepherd is not part of zigbee2mqtt right? So that means that this issue cannot be solved from zigbee2mqtt? If so, we could update the docs with this workaround and perhaps file issue with zigbee-shepherd?

Koenkk, 2019-03-16:
@peterforeman yes, it's a zigbee-shepherd issue, but as zigbee-shepherd is not maintained anymore it basically comes down to us to fix this :)

Koenkk, 2019-05-16:
Should be improved in zigbee2mqtt 1.4

closed 2019-05-16 by Koenkk