Koenkk/zigbee2mqtt #1224: More or less consistently facing error code 183 or 205 issues with my bulbs

martinrosenauer, 2019-03-10:
@Koenkk, on Coordinator firmware version: '20190223', zigbee2mqtt version 1.2.1 (commit #18aee48) I'm having issues with two white A60 Hue bulbs (not entirely sure it's because they are Hue), which are right next to the CC2531 controller. They are on-off non-responsive and when being unresponsive I see

"message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x0017880110570b58","type":"device","friendlyName":"hue_bryggers_lampe"},"message":"{\"state\": \"ON\"}"}}'

or 

"message":"Error: AF data request fails, status code: 183. APS no ack.","meta":{"entity":{"ID":"0x0017880110570b58","type":"device","friendlyName":"hue_bryggers_lampe"},"message":"{\"state\": \"ON\"}"}}'

in the log.  According to the network map they are connected to other routers, which works fine. 

I have another CC2531 with coordinator firmware version: '20181024', and another folder with zigbee2mqtt version 1.0.1 (commit #e21bfc1). Switching back to this seems to switch back to a (at least more) stable version, but as I need the Trådfri E1524 support, I'd be really happy to stay on the latest firmware + dev.

CC2531 is with a 2m USB stable removed from the Tinkerboard, which is running zigbee2mqtt, and 2.4Ghz wifi should be on non-colliding channels. Zigbee network consists of a 25-30 different Xiaomi sensors/buttons, Trådfri bulbs, Hue bulbs and E1524s.

Thanks _alot_ for any advice!

Koenkk, 2019-03-10:
Can you try to re-pair the problematic devices?

martinrosenauer, 2019-03-10:
OK, removed the two Hue bulbs from configuration.yaml, database.db and state.json. 

As I don't have a Hue dimmer/bridge I'm not able to factory reset them and believe I will have to wait until I can get my hands on one next week. (or go to Ikea and get two Trådfri instead).

Please let me know if you have any other suggestion ?! :)

martinrosenauer, 2019-03-10:
@Koenkk,, are you in general having a perfectly stable zigbee network on a cc2531 with 20190223/z2m 1.2.1 ?

Koenkk, 2019-03-10:
@martinrosenauer yes, but initially not, I've documented my steps to improve the network here: http://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

DeliriousMetro, 2019-03-11:
+1 here too

After firmware update getting error 205, bulbs connect fine on bootup of zigbee2mqtt, states show ok, toggling brightness or switch causes this error.

Will be looking at getting an dimmer and probably returning it after repairing the bulbs

martinrosenauer, 2019-03-11:
@DeliriousMetro, are you only seeing this with Hue bulbs or also Trådfri or similar ? I plan to swing by Ikea today and buy Trådfri replacements for the two Hues where I see this and figure if 20190223/z2m 1.2.1 then becomes stable.

@Koenkk, thanks for the link - I have been doing 1 and 2, but 3 (the CC2530 as router) is pending. However, as I have 10 bulbs in the network (out of 34 devices) I would expect it to be adequate as routers ? Is your device-list far from that ?

salopette, 2019-03-11:
https://paste.ubuntu.com/p/2fhjTYTMZ8/

I also have intermissions more often

DeliriousMetro, 2019-03-11:
> @DeliriousMetro, are you only seeing this with Hue bulbs or also Trådfri or similar ? I plan to swing by Ikea today and buy Trådfri replacements for the two Hues where I see this and figure if 20190223/z2m 1.2.1 then becomes stable.
> 

I am pretty sure that the Zigbee2Mqtt software is not the cause, as it had worked previously on 1.2.1 without the coordinator update. Updating the coordinator CC2531 causes certain Hue bulbs to loose ability to pair and will have to be reset using the dimmer switch.

What is weird is other zigbee devices rejoin without any additional intervention. So probably a hue issue.

Anyway, I got my hue bulbs a couple of months ago £5 each for 6 so had it for a steal anyway. I did have an Ikea bulb but returned it due to it not pairing with zigbee as it had a very old firmware version, check the back of the box for the date!! Or else, you'll be stuck 30 mins in the Ikea returns...

Koenkk, 2019-03-12:
So for the people experiencing the 205 errors, I expect this to happen after a reflash of the stick. This error indicates that the coordinator (CC2531) has no route to the device. Once the coordinator has a route, this error shouldn't appear anymore.

You know that the coordinator has a route when it atleast has send once command (e.g. turn on/off) to a device.

To get a route in the coordinator:
- Wait a few minutes for the zigbee network to rebuild
- If that doesn't work, send some on/off commands to the device
- If that doesn't work, re-power the device
- If that doesn't work, re-pair the device.

DeliriousMetro, 2019-03-12:
Buying a dimmer and resetting it twice fixed it for the hue bulbs being non responsive. I am looking at moving away from these but problem is I have 6 left anyway lol.

Repairing did work but required to have the permit join on true whilst doing the pairing with the dimmer.

Thanks for the updates devs this open source project has been amazing.

FrankZZ, 2019-03-13:
> So for the people experiencing the 205 errors, I expect this to happen after a reflash of the stick. This error indicates that the coordinator (CC2531) has no route to the device. Once the [...]

I got 205's on all my IKEA TRADFRI bulbs aswell on 1.2.1 with 20190223
Tried your steps except re-pairing devices. I reflashed 20190223 and the 205's are gone

privatesam, 2019-03-14:
I've been experiencing this exact problem with the two Hue bulbs I bought last week. I was on an older firmware and could easily pair the bulbs if they were close to the coordinator but as soon as I moved them to the upstairs bedroom I got 205's despite other sensors happily talking to the coordinator from further away.  I tried resetting and re-pairing etc 

I flashed the latest firmware to the 2531 last night and it still didn't help the problem.  So I decided to reset and re-pair yet again but tried to do it from upstairs but unfortunately it didn't work - I'll try and move the bulb closer to the coordinator re-pair and then move back to the upstairs and see what the result is and report back.

FrankZZ, 2019-03-14:
> I've been experiencing this exact problem with the two Hue bulbs I bought last week. I was on an older firmware and could easily pair the bulbs if they were close to the coordinator but as soon as I [...]

do you have any routers between the coordinator and the place upstairs where you want to pair the hue bulbs? 
maybe this page can help you:
http://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

privatesam, 2019-03-14:
> 
> 
> > I've been experiencing this exact problem with the two Hue bulbs I bought last week. I was on an older firmware and could easily pair the bulbs if they were close to the coordinator but as soon as I moved them to the upstairs bedroom I got 205's despite other sensors happily talking to the coordinator from further away. I tried resetting and re-pairing etc
> > I flashed the latest firmware to the 2531 last night and it still didn't help the problem. So I decided to reset and re-pair yet again but tried to do it from upstairs but unfortunately it didn't work - I'll try and move the bulb closer to the coordinator re-pair and then move back to the upstairs and see what the result is and report back.
> 
> do you have any routers between the coordinator and the place upstairs where you want to pair the hue bulbs?
> maybe this page can help you:
> http://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html
> 
> perhaps there is something causing interference on the spot upstairs and not further away?

There is a router (a Innr bulb) upstairs but a map of my network showed it wasn't between the Hue and the coordinator.  The distance is not big between the Hue and the coordinator - Xioami sensors are further away and happily sending data to the coordinator.

SalkinDK, 2019-03-19:
Having the exact same problem here.
My setup is:
```
 System: HassOS 2.10  (aarch64 / raspberrypi3-64)
 Home Assistant version: 0.89.2
 Supervisor version: 148
 Add-on Version:  1.2.1
 Coordinator firmware version: CC2531ZNP-Prod_20190223
```
Network
```
Currently 20 devices are joined:
3/19/2019, 1:39:17 PM - info: Bedroom (0x00158d0001ff3fdb): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
3/19/2019, 1:39:17 PM - info: Garden_light (0x90fd9ffffedcb93f): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Router)
3/19/2019, 1:39:17 PM - info: Washing_Machine (0x00158d0002045647): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
3/19/2019, 1:39:17 PM - info: Garden_Door (0x00158d0002049a00): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
3/19/2019, 1:39:17 PM - info: Laundry_room (0x00158d0002320a11): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/19/2019, 1:39:17 PM - info: Bed_Light (0x000b57fffee3fd98): ICPSHC24-30EU-IL-1 - IKEA TRADFRI driver for wireless control (30 watt) (Router)
3/19/2019, 1:39:17 PM - info: Workshop_Light (0x000b57fffed6c624): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
3/19/2019, 1:39:17 PM - info: Laundry_Room_light (0x000b57fffed6b592): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
3/19/2019, 1:39:17 PM - info: Boiler_Room_light (0x000b57fffea01a74): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
3/19/2019, 1:39:17 PM - info: Basement_Bathroom_light (0x000b57fffedd5bde): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
3/19/2019, 1:39:17 PM - info: Basement_Room_light (0x000b57fffed5e4b5): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
3/19/2019, 1:39:17 PM - info: Basement_Living_light (0x000b57fffed4c433): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
3/19/2019, 1:39:17 PM - info: Basement (0x00158d000237547a): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
3/19/2019, 1:39:17 PM - info: Laundry_room_window (0x00158d00022f16f1): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
3/19/2019, 1:39:17 PM - info: Front_Door (0x00158d000232b12d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
3/19/2019, 1:39:17 PM - info: Beton_Lampe (0x000d6ffffe9c6032): E1603 - IKEA TRADFRI control outlet (Router)
3/19/2019, 1:39:17 PM - info: Mormor_light (0x00178801024a798e): 8718696449691 - Philips Hue White Single bulb B22 (Router)
3/19/2019, 1:39:17 PM - info: Sofa_light (0x0017880110582450): 9290011370 - Philips Hue white A60 bulb E27 (Router)
3/19/2019, 1:39:17 PM - info: Bedroom_light (0x00178801105827f7): 9290011370 - Philips Hue white A60 bulb E27 (Router)
3/19/2019, 1:39:17 PM - info: DoorBell (0x00158d0002134042): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)

```

If i use the firmware specified above, i get A LOT of problems error 205 many times a day, bulbs that stops working etc.
But if i go back using my old 2531 stick with Coordinator firmware version: 'undefined' (but one of the first one after the hassio add-on was released.
Everything is working without any errors at all, the only thing that stopped working is my doorbell (Xiaomi Aqara wireless switch) it isn't reporting presses, whenever i press the button it just reports {"battery":100,"voltage":3005,"linkquality":65}

So maybe it has something to do with the new firmware?

martinrosenauer, 2019-03-19:
@SalkinDK, try the updated firmware @Koenkk posted to me earlier. He wrote:

Please try with the following firmware: https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5
Relative to the 20190223 firmware, it has an increased memory heap at the cost of direct connected devices to the coordinator (15 -> 5).

I've had fairly good results with it so far, although I had to re-pair a lot of devices. Will report back in a few days in regards of stability.

SalkinDK, 2019-03-19:
> @SalkinDK, try the updated firmware @Koenkk posted to me earlier. He wrote:
>
> Please try with the following firmware: https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5
> Relative to [...]

Hmm to be honest i am not into repairing devices right now, the WAF is rather low because of the problems the last couple of days.
So i think i will just have to wait it out on the old firmware. i dont get groups reporting etc but i can turn the lights on and off.

Koenkk, 2019-03-19:
@SalkinDK re-pairing is not needed anymore http://www.zigbee2mqtt.io/information/what_does_and_doesnt_require_repairing.html

SalkinDK, 2019-03-19:
> @SalkinDK re-pairing is not needed anymore http://www.zigbee2mqtt.io/information/what_does_and_doesnt_require_repairing.html

i know, i can also change between my to sticks without repairing, it was because of @martinrosenauer answer about he had to repair when he used the experimental firmware you linked to earlier.

martinrosenauer, 2019-03-19:
@SalkinDK, I had to re-pair some Xiaomi sensors and 1 out of 3 E1524 Trådfri Remotes. The other ones worked - so somewhat 28 out of 35.

SalkinDK, 2019-03-20:
@martinrosenauer how is your status.?
I have been running the newest version of the add-on ( 1.2.1) with my old firmware stick and are seeing no errors at all. All bulbs working all the time, all in all a stable setup, but without all the fancy new things from the 20190223 firmware.

martinrosenauer, 2019-03-21:
@SalkinDK, it has been behaving quite stable - I checked the logs for errors and haven't seen any for the past 4 days.

Koenkk, 2019-03-21:
@SalkinDK @marcusaram could you also try with the max stability firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator

SalkinDK, 2019-03-26:
@Koenkk i installed that firmware rev yesterday morning and it has now been running for 24 hours, without any problems. i had to repair a single hue bulb but if that's the cave out for a stable network i am fine with that.
no 183 or 205 in the last 24 hours, and i  can even toggle HA groups without the network shitting itself.
So far so good.

martinrosenauer, 2019-03-31:
@Koenkk, I've now been running on the firmware you posted for approximately two weeks. It seems quite stable, but I still see devices which all of a sudden stops working and needs to be repaired. Specifically it has been Xiaomi Aqara Door and Window Sensors (MCCGQ11LM), where 2 out of 4 in the living room of our house just disconnected and had to be re-paired. Earlier it was another sensor of the same model. 

The zigbee network consists of 38 devices, a CC2531 as controller, a CC2531 as router and the rest being Trådfri bulbs and various Xiaomi battery powered devices.

You described the firmware as:

"https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5
Relative to the 20190223 firmware, it has an increased memory heap at the cost of direct connected devices to the coordinator (15 -> 5)".

Koenkk, 2019-04-02:
@martinrosenauer what is the linkquality of those sensors?

closed 2019-06-08 by stale[bot]