Koenkk/zigbee2mqtt #1225: Please commit your changes or stash them before you merge.

jossie67, 2019-03-10:
Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

jossie67, 2019-03-10:
When I try to update I get an error

From https://github.com/Koenkk/zigbee2mqtt
   18aee48..6e60063  dev        -> origin/dev
Updating 92d88b6..e5ca977
error: Your local changes to the following files would be overwritten by merge:
        package.json
Please commit your changes or stash them before you merge.
Aborting

I never changed any files inside the folder /opt/zigbee2mttq except configuration.yaml

Koenkk, 2019-03-10:
Execute `git checkout HEAD -- package.json` and try again.

jossie67, 2019-03-11:
thanks , fixed

Koenkk, 2019-03-11:
good!

closed 2019-03-11 by Koenkk