Koenkk/zigbee2mqtt #1227: CC2530 router update its state every 3 seconds instead of every minute

kennymc-c, 2019-03-10:
Today I switched to the Docker Image with 1.2.1 and I noticed that my CC2530 router now sends its status every 3 seconds which previously only happened every minute. Is this normal and intended? The logs become quite big and I don't want to change the log level because I still want to have the states of other devices in the log.

kennymc-c, 2019-03-10:
Ok, i solved it myself. I just updated my CC2531 to the latest 20190223 FW and rebooted the router. Now the router status updates are coming in 1 minute intervals again.

kennymc-c, 2019-03-15:
When I looked into the logs today, I noticed that the status update interval is again at 3 seconds.

Koenkk, 2019-03-15:
@ptvoinfo can you help?

bjobo, 2019-03-23:
Same for me. Every 3 seconds I get a report from my CC2530 router (latest standard firmware):
`Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.router' (0x00124b001bd12541)`
Would it be hepful to flash the 'diag' firmware and post the output?

kennymc-c, 2019-03-24:
I noticed that rebooting the router for 2 or 3 times can sometimes solve the problem. It is also strange that the state changes to false after the first reboot and only jumps back to true after another reboot.

SargonofAssyria, 2019-04-01:
I have exactly the same. After upgrading / restarting zigbee2mqtt the router connects, but start flooding the system with 5-6 messages every 3 seconds. After I pull the router from power and apply power to the router after a few seconds, every goes back to normal and messages are coming in very 60 seconds.
This happens every time when zigbee2mqtt is restarted.
Using version 1.2.1 (commit #76d7861)
Controller version 20190315 Max stability version.
Router version, not exactly know, but latest from ptvo.info It is a CC2531 stick but in the zigbee logging it states CC2530 router???

Koenkk, 2019-04-01:
Has this been fixed in the latest dev branch? (might require replugging once).

SargonofAssyria, 2019-04-01:
I am not a programmer, but it looks to me as if you disabled reporting when the CC2531 router present it self as CC2530.ROUTER. Is the network map still reported, or is all reporting stopped.
Like I wrote I am not a programmer, but still want the network amp.

Koenkk, 2019-04-01:
@SargonofAssyria that has nothing to do with reporting, the reporting feature is useless for device like the CC2530 router (http://www.zigbee2mqtt.io/information/report.html).

Does it work now?

kennymc-c, 2019-04-01:
With the latest arm32v6-dev image the problem is still present

Koenkk, 2019-04-02:
@kennymc-c did you re-plug the router after starting zigbee2mqtt?

SargonofAssyria, 2019-04-02:
I did not have to re-plug the router, everything went fine the initial startup.
But it could not hurt.

kennymc-c, 2019-04-02:
@Koenkk I think now it works for me, too. After two reboots of the router I could get back to a normal 1 min. interval/state true and also two restarts of zigbee2mqtt have not yet bring the problem back again. I will test this the next days and report if I notice anything unusual.

kennymc-c, 2019-04-05:
I think this issue can be closed as soon as the fix is also available in the master branch, because i had no more problems in the last days.

SargonofAssyria, 2019-04-06:
It is working fine here too.

Koenkk, 2019-04-07:
Great, thanks all!

closed 2019-04-07 by Koenkk

gabberhq, 2022-10-17:
Hi, I got a Weebee interface ( 2530 2591 ) as router with firmware **router-cc2530-cc2591-std.hex** 

every second, it sent this message to my logs : 

MQTT publish: topic 'zigbee2mqtt/ROUTER NAME', payload '{"led":false,"linkquality":116}'

how to avoid it ? I thought that the latest FW solved this issue ?

jonferreira, 2023-09-16:
Have the same issues, any way to to stop this?