Koenkk/zigbee2mqtt #1228: Amazon echo support?

echo4019, 2019-03-10:
Hi I've been using the zibbee to mqtt hub for a while now. I've recently added a philips hue bulb and was wondering if there is any way to control the bulb with my Amazon echo? 

Thanks.

nldroid, 2019-03-10:
My guess is that it depends on the home automation system you are using. If you use Home Assistant for example, you can.

**https://www.home-assistant.io/components/alexa/**

echo4019, 2019-03-10:
Yea it's home assistant that I'm using. What is the best solution I see there is a paid subscription service. Are there free alternatives? 

Thanks.

ik0adr, 2019-03-10:
[https://community.home-assistant.io/t/the-haaska-super-thread/76994](url)

dreimer1986, 2019-03-10:
There are many ways. I have experience with two of them. A simple way called Home Assistant Cloud and there is a more complex one called Alexa integration (https://www.home-assistant.io/components/alexa/) if you wanna write your own lil skill and are happy with always saying "Alexa tell Home Assistant to switch XY on", then the 2nd is for you. If you just want stuff to work and donate money to Home Assistant, choose the cloud. Never had any experience with Haaska though.

nldroid, 2019-03-10:
I think you can host the Alexa skill locally without the cloud option (at least the one for Google Home does).

This question is more for Home Assistant than it is for Zigbee2mqtt. Please look on the home assistant website or ask a question on their forum.

Koenkk, 2019-03-10:
Closing this as it isn't a zigbee2mqtt issue, feel free to continue discussion.

closed 2019-03-10 by Koenkk