Koenkk/zigbee2mqtt #1229: Control4/Elan integration?

waring192, 2019-03-10:
Just wondering if this could be integrated into Control4, Elan or RTI?

Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Koenkk, 2019-03-10:
Zigbee2mqtt can be integrated with any home automation software that supports MQTT.

If they don't support MQTT, zigbee2mqtt cannot integrate with it.

lsilva171, 2019-04-14:
Regarding Control4, you may have some luck with the URLs below
https://www.c4forums.com/topic/24818-control4-and-mqtt/
https://github.com/davidmassot/Berto

closed 2019-06-13 by Koenkk

zuhair68, 2021-03-25:
> Just wondering if this could be integrated into Control4, Elan or RTI?
>
> Did you read the FAQ?
> https://www.zigbee2mqtt.io/information/FAQ.md
>
> Unsupported device? [...]

There's a driver that connect Control4 to zigbee2mqtt.

https://www.berto.co.uk